mjs513/FreeIMU-Updates #17: Compile errors: FreeIMU_serial_ARM_9250_3115A2.ino

nandomtl, 2015-02-07:
I am trying to compile the module in the subject line, but am getting a pile of error msgs.
Probably a rookie mistake, but I can't figure what's the glitch... 
Below are the first few msgs - to make it more readable, I removed the prefix in all but the first line.

/Users/username/Documents/Arduino/libraries/FreeIMU/FreeIMU.h:436: error: 'HMC58X3' does not name a type
FreeIMU.h:436: error: 'HMC58X3' does not name a type
FreeIMU.h:472: error: 'BMP085' does not name a type
FreeIMU.h:497: error: ISO C++ forbids initialization of member 'zeroMotioncount'
FreeIMU.h:497: error: making 'zeroMotioncount' static
FreeIMU.h:497: error: ISO C++ forbids in-class initialization of non-const static member 'zeroMotioncount'
FreeIMU.h:512: error: a brace-enclosed initializer is not allowed here before '{' token
FreeIMU.h:512: error: ISO C++ forbids initialization of member 'sensor_order'
FreeIMU.h:512: error: making 'sensor_order' static
FreeIMU.h:512: error: invalid in-class initialization of static data member of non-integral type 'int [9]'
FreeIMU.h:513: error: a brace-enclosed initializer is not allowed here before '{' token
FreeIMU.h:513: error: ISO C++ forbids initialization of member 'sensor_sign'
FreeIMU.h:513: error: making 'sensor_sign' static
FreeIMU.h:513: error: invalid in-class initialization of static data member of non-integral type 'int [9]'
FreeIMU.h:545: error: ISO C++ forbids initialization of member 'lastUpdate1'
FreeIMU.h:545: error: making 'lastUpdate1' static

Below is a screenshot of the folder structure.

[image]

mjs513, 2015-02-07:
Hi - thanks for trying the library.  Lets see if I can help.

1. Did you remember to change the FreeIMU.h file to set the configuration for your board.  The sketch that you selected is for the MPU-9250 and the MPL3115 barometer.  You will need to uncomment the line for define Mario and comment any other board define that is currently defined.  Check out the wiki it should explain a little more on how to do this.

2. Some of the errors look like you are the Arduino 1.0.6 lDE.  You will need to use the Arduino 1.5.7 or greater IDE.

If this does not work let me know what IMU you are using and Arduino board.  I have been using primarily the Mega2560 and the Teensy 3.1.  If you are using a Uno it probably won't fit.

nandomtl, 2015-02-07:
Hello

1 - Yes, I did that. I am using an invensense MPU9250, but I should not be using "FreeIMU_serial_ARM_9250_3115A2.ino", as I have no other sensors on the board.
2 - I installed 1.5.8 - still had some errors about missing files, but nothing I needed (such as L3G.h), so I commented them out and the compile ran. But:
3 - You are correct - the compiled code (most sketches) does not fit in the Uno. 

I still am puzzled with the #include statement and the position of the libraries in the folder tree.
In one particular case (FreeIMU_quaternion.ino), the compiler dies because:
AK8963.h: No such file or directory
   #include "AK8963.h"
As you can see in my original post, the folder+file are there, and  "preferences" points to /Users/username/Documents/Arduino/. Also, I can see the folder name under the menu sketch/import libraries >, along with a couple dozen other folder names that are part of this package (screen shot below).
If you know of a good reference about this subject, I would appreciate if you could share it. (Already been to http://arduino.cc/en/Guide/Libraries.)

PS: is there a document in this package that describes what the several modules purpose is? Sometimes the name tells everything one needs to know, but at times I can't figure it.

Thanks for you help

mjs513, 2015-02-07:
Real simple fix here.  Add the following line to the sketch (FreeIMU_quaternion.ino) along with the other includes:
#include <AK8963.h>

It should now compile for you. Missed that adding that one.  Unfortunately I don't know of any good reference for this issue.  I learned from trial and error.

The sketch you should be using is: FreeIMU_serial_ARM_CPU.ino, this will give you everything you need from the 9250.  Just remember to comment out all the board defines except for the GEN_9250.

I have not got around to create a list and explanation of all the examples.  Like you said most were self explanatory.

Mike

nandomtl, 2015-02-07:
Hello

Thanks for the quick reply
Did you mean just #include with no parameters? The compiler does not like that...
FreeIMU_quaternion.ino:4:9: error: #include expects "FILENAME" or <FILENAME>

Cheers

Carlos

mjs513, 2015-02-08:
Hi Carlos,

My mistake - suppose to be #include <AK8963.h>

Sorry about that. 

Mike

mjs513, 2015-02-08:
Interesting, just checked my post and it didn't do the include correctly.  Lets try something different:
#include < AK8963.h >

Leave out the spaces before and after AK8963.h if the compiler complains.

Mike

nandomtl, 2015-02-08:
Mike, I did the tests  -  but...
#include <AK8963.h>
produces:
FreeIMU_quaternion_V2.ino:32:9: error: #include expects "FILENAME" or <FILENAME>
Error compiling.

...and
#include < AK8963.h >
gives:
FreeIMU_quaternion_V2.ino:4:22: fatal error:  AK8963.h : No such file or directory
(I also tried double quotes, no joy.)
To make sure other files in the libraries folder were being picked up, I rename one of them (ADXL345.h), and the compiler complained about it. 
FreeIMU_quaternion_V2.ino:2:21: fatal error: ADXL345.h: No such file or directory
So the compiler is looking at the right folder tree.
Weird.

Carlos

mjs513, 2015-02-09:
Carlos. This very strange. Have to think about this one. Did you ditch the
spaces after < and before the closing >. So it reads #include <AK8963.h>

Also just do a double check that the only board you have selected is the
9250. Does this happen on the FreeIMU_SERIAL_ARM_CPU.ino sketch?

Mike

nandomtl, 2015-02-09:
Hi, Mike
My previous post somehow got truncated: the first test was without the spaces, and the compiler gives a syntax error (... expects "FILENAME" or <FILENAME>). The weird thing is that other "#include" statements do NOT have the spaces - see below a few lines pasted from the file.
________________________________________
#include <ADXL345.h>
#include <bma180.h>
#include < AK8963.h >                  << with spaces
#include <HMC58X3.h>
#include <LSM303.h>
#include <ITG3200.h>
#include <MS561101BA.h>
#include <I2Cdev.h>
#include <MPU60X0.h>
#include <EEPROM.h>
//#include <L3G.h>                   <<  I commented out this line
#include <LPS331.h> 
#include <AP_Baro_MS5611.h>
#include "RunningAverage.h"
#include <iCompass.h>
________________________________________

The FreeIMU_SERIAL_ARM_CPU.ino compiles nicely but does not load, it is too big for the Uno. I bit less than 2 K bytes...
____________________
Sketch uses 34,026 bytes (105%) of program storage space. Maximum is 32,256 bytes.
Global variables use 1,926 bytes (94%) of dynamic memory, leaving 122 bytes for local variables. Maximum is 2,048 bytes.
processing.app.debug.RunnerException: Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
____________________

Pretty sure FreeIMU is ok, but below are the first 50 or so lines...
_________________________________________________

#ifndef FreeIMU_h
#define FreeIMU_h

// Uncomment the appropriated version of FreeIMU you are using
//#define FREEIMU_v01
//#define FREEIMU_v02
//#define FREEIMU_v03
//#define FREEIMU_v035
//#define FREEIMU_v035_MS
//#define FREEIMU_v035_BMP
//#define FREEIMU_v04

// 3rd party boards. Please consider donating or buying a FreeIMU board to support this library development.
//#define SEN_10121 //IMU Digital Combo Board - 6 Degrees of Freedom ITG3200/ADXL345 SEN-10121 http://www.sparkfun.com/products/10121
//#define SEN_10736 //9 Degrees of Freedom - Razor IMU SEN-10736 http://www.sparkfun.com/products/10736/
//#define SEN_10724 //9 Degrees of Freedom - Sensor Stick SEN-10724 http://www.sparkfun.com/products/10724
//#define SEN_10183 //9 Degrees of Freedom - Sensor Stick  SEN-10183 http://www.sparkfun.com/products/10183
//#define ARDUIMU_v3 //  DIYDrones ArduIMU+ V3 http://store.diydrones.com/ArduIMU_V3_p/kt-arduimu-30.htm or https://www.sparkfun.com/products/11055
//#define GEN_MPU6050 // Generic MPU6050 breakout board. Compatible with GY-521, SEN-11028 and other MPU6050 wich have the MPU6050 AD0 pin connected to GND.
//#define DFROBOT  //DFROBOT 10DOF SEN-1040 IMU
//#define MPU9250_5611  //MPU-9250 IMU with MS5611 Altimeter from eBay
//#define GEN_MPU9150
#define GEN_MPU9250  // Use for Invensense MPU-9250 breakout board
//#define Altimu10  // Pololu AltIMU v10 - 10 DOF IMU - http://www.pololu.com/product/1269
//#define GY_88  //GY-88 Sensor Board from eBay
//#define GY_87  //GY-87 Sensor Board from eBay, NOTE: Pressusre sensor is BMP180 but BMP085 library should work
//#define Mario   // MPU-9150 plus Altitude/Pressure Sensor Breakout - MPL3115A2  https://www.sparkfun.com/products/11084
//#define APM_2_5  //  APMM 2.5.2 (EBAY)
//#define Microduino    // 2015 02 07 CSa removed BMP085 in lie 224

//#define DISABLE_MAGN // Uncomment this line to disable the magnetometer in the sensor fusion algorithm

//Magnetic declination angle for iCompass
//#define MAG_DEC 4 //+4.0 degrees for Israel
//#define MAG_DEC -13.1603  //degrees for Flushing, NY
#define MAG_DEC -14.4833  //degrees for Montreal, QC

//Number of samples to average in iCompass
#define WINDOW_SIZE 1 //Set to 1 to turn off the Running Average

// Set filter type: 1 = Madgwick Gradient Descent, 0 - Madgwick implementation of Mahoney DCM
// in Quaternion form, 3 = Madwick Original Paper AHRS
#define MARG 3

// proportional gain governs rate of convergence to accelerometer/magnetometer
// integral gain governs rate of convergence of gyroscope biases
// set up defines for various boards in my inventory, DFROBOT and Freeimu have
// temperature calibration curves. (3.31.14)

_________________________________________________

Thanks a lot for trying to help, I appreciate it!

Carlos

mjs513, 2015-02-09:
> //#include << I commented out this line

Hi Carlos

Well the configuration looks correct. Since the other sketch compiles then
it is with the quaternion sketch.  Did you try to compile without the
spaces. If you didn't give it a try.

Mike

nandomtl, 2015-02-09:
Hi, Mike - I tried both ways, with and without spaces. The compiler issues
error messages in both cases.

On 8 February 2015 at 19:52, Mike S <notifications@github.com> wrote:

> Hi Carlos
>
> Well the configuration looks correct. Since the other sketch compiles then
> it is with the quaternion sketch. Did you try to compile without the
> spaces. If you didn't give it a try.
>
> Mike
>

mjs513, 2015-02-09:
When I get to my computer I will you the sketch that I have verified
compiles. You said you are using a UNO?

nandomtl, 2015-02-09:
Correct, Arduino Uno.

Thanks a lot!

Carlos

mjs513, 2015-02-09:
I have attached the sketch that compiles without issue for me.  Let me know if you have a problem.  Don’t use the one with matrix at the end – trying to convert quaternion to dcm. Still have to get the math right on that one J

 

Mike

nandomtl, 2015-02-09:
Mike, was there supposed to be a file attached??

On 9 February 2015 at 15:09, Mike S <notifications@github.com> wrote:

> I have attached the sketch that compiles without issue for me. Let me know
> if you have a problem. Don’t use the one with matrix at the end – trying to
> convert quaternion to dcm. Still have to get the math right on that one J
>
>
>
> Mike
>

mjs513, 2015-02-10:
Yes there was a file attached GITHUB must have blocked it.  Going to try and paste it below, just do a copy and paste into the ide (I deleted the lib:

```
#include <HMC58X3.h>
#include <I2Cdev.h>
#include <MPU60X0.h>
#include <AK8963.h>

#include "RunningAverage.h"
#include <iCompass.h>

#include <AP_Math_freeimu.h>
#include <Filter.h>    // Filter library

#include <EEPROM.h>
#include <Wire.h>
#include <SPI.h>

//#define DEBUG
#include "DebugUtils.h"
#include "CommunicationUtils.h"
#include "FreeIMU.h"
#include "FilteringScheme.h"
#include <MovingAvarageFilter.h>

float q[4], val[11];

// Set the FreeIMU object
FreeIMU my3IMU = FreeIMU();

void setup() {
  Serial.begin(115200);
  Wire.begin();
  
  delay(5);
  my3IMU.init();
  delay(5);
}

void loop() { 
  my3IMU.getQ(q,val);
  serialPrintFloatArr(q, 4);
  Serial.println("");
  delay(20);
}

```

nandomtl, 2015-02-10:
Mike, your code compiled fine (after I had a couple of library issues fixed).
But, to fix the library issues, I moved the source code a couple of levels higher (to /Users/carlossa/Documents/Arduino/FreeIMU_quaternion), so I was suspicious that the problem was the folder location. I then moved the original quaternion folder/source code to that position, and tried again.
The same old problem was there... phew!
But this time I decided to follow the compiler's suggestion:
____________________
In file included from FreeIMU_quaternion.ino:28:0:
/Users/carlossa/Documents/Arduino/libraries/FreeIMU/FreeIMU.h:299:22: fatal error: AK8963.h: No such file or directory
   #include "AK8963.h"
                      ^
compilation terminated.
Error compiling.
_____________________
I added the statement #include "AK8963.h"  to the file, and the compiler was happy!
I do not know enough about C to understand the difference between including with <> and "", but this fixed the issue.

Thanks a lot for your help!

Carlos

mjs513, 2015-02-10:
No problem Carlos. Let me know how it works out for. Are you willing to
what you are using it for? Just curious.

Mike

nandomtl, 2015-02-10:
It's the least I can do...

*http://tinyurl.com/ojenszs <http://tinyurl.com/ojenszs>*

Let me know what you think...

mjs513, 2015-02-10:
Just checked it out. Very nice. Becateful with using men's. Pretty
susceptible to interference from large iron like table legs, temperature,
etc. Will dig up some references for you and post them.

PS. I am a Aero engineer.

Mike

nandomtl, 2015-02-10:
The instruments driven by the 9250 won't be critical - the plane is going to be VFR only.
(Did you see the video?)

I'm an engineer too (electrical), but have spent all of my professional life in IT.

Cheers

Carlos

mjs513, 2015-02-10:
Just saw it. Had to view the link in desktop view to see and read
everything.

Like the instrument panel. Struggled to get it working in processing.

By the way booked marked the page so I can follow.

Will read when more awake. By the way if you need to reach me you can email
at FreeIMU@aol.com.

I have tested the code out on a uno, Mega, Teensy 3.1, Due, and even on the
Intel board. Had problems with Intel reading multiple I2c devices and
missing standard libraries. Haven't played with a pcduino, will have to
give it a try one of these days.

Been working on the library updates for over a year. So yep, slow going but
fun.

Good luck.
Mike

The instruments driven by the 9250 won't be critical - the plane is going
to be VFR only.
(Did you see the video?)

I'm an engineer too (electrical), but have spent all of my professional
life in IT.

Cheers

Carlos

closed 2015-02-16 by mjs513