mjs513/FreeIMU-Updates #18: compile error for mpu9250

rs132, 2015-02-16:
Hi guys,

In the FreeIMU.h file, I uncommented the following line:
 #define GEN_MPU9250  // Use for Invensense MPU-9250 breakout board 
All other lines for different IMUs are still commented out.

Howevre, while trying to compile, I get the following error:
error: 'mag' was not declared in this scope
   mag.getHeading(&raw_values[6], &raw_values[7], &raw_values[8]);   
   ^

Does anyone have any idea why this is happening? I am using arduino version 1.5.7 beta on windows 8.

mjs513, 2015-02-16:
Can you tell me what sketch you are using?

rs132, 2015-02-16:
FreeIMU_serial. When I tried running _9250_AK8963_demo.ino, that worked, so Im pretty sure my mpu9250 is working. Also, when I comment out that #define statemnet in the FreeIMU.h header file, then FreeIMU_serial works, except the magnetometer values are frozen. So, it compiles when all the #define statements for IMUs in the header file are commented out.

mjs513, 2015-02-16:
Interesting I just recompiled the sketch as is from GITHUB and it compiles without a problem for me using 1.5.7.  The defines in FreeIMU.h should look exactly like this:

```c++
// Uncomment the appropriated version of FreeIMU you are using
//#define FREEIMU_v01
//#define FREEIMU_v02
//#define FREEIMU_v03
//#define FREEIMU_v035
//#define FREEIMU_v035_MS
//#define FREEIMU_v035_BMP
//#define FREEIMU_v04

// 3rd party boards. Please consider donating or buying a FreeIMU board to support this library development.
//#define SEN_10121 //IMU Digital Combo Board - 6 Degrees of Freedom ITG3200/ADXL345 SEN-10121 http://www.sparkfun.com/products/10121
//#define SEN_10736 //9 Degrees of Freedom - Razor IMU SEN-10736 http://www.sparkfun.com/products/10736/
//#define SEN_10724 //9 Degrees of Freedom - Sensor Stick SEN-10724 http://www.sparkfun.com/products/10724
//#define SEN_10183 //9 Degrees of Freedom - Sensor Stick  SEN-10183 http://www.sparkfun.com/products/10183
//#define ARDUIMU_v3 //  DIYDrones ArduIMU+ V3 http://store.diydrones.com/ArduIMU_V3_p/kt-arduimu-30.htm or https://www.sparkfun.com/products/11055
//#define GEN_MPU6050 // Generic MPU6050 breakout board. Compatible with GY-521, SEN-11028 and other MPU6050 wich have the MPU6050 AD0 pin connected to GND.
//#define DFROBOT  //DFROBOT 10DOF SEN-1040 IMU
//#define MPU9250_5611  //MPU-9250 IMU with MS5611 Altimeter from eBay
//#define GEN_MPU9150
#define GEN_MPU9250  // Use for Invensense MPU-9250 breakout board
//#define Altimu10  // Pololu AltIMU v10 - 10 DOF IMU - http://www.pololu.com/product/1269
//#define GY_88  //GY-88 Sensor Board from eBay
//#define GY_87  //GY-87 Sensor Board from eBay, NOTE: Pressusre sensor is BMP180 but BMP085 library should work
//#define Mario   // MPU-9150 plus Altitude/Pressure Sensor Breakout - MPL3115A2  https://www.sparkfun.com/products/11084
//#define APM_2_5  //  APMM 2.5.2 (EBAY)
//#define Microduino

```
What board are you using?

Mike

rs132, 2015-02-16:
I'm using the mega2560. Could it perhaps be how my libraries are configured? So, I presently have Arduino/libraries/... then all the libraries folders copied from the master file from github, specifically from the FreeIMU-Updates-master/libraries folder.

Is this the same way that you added the libraries?

rs132, 2015-02-16:
Also, this only seems to happen for GEN_MPU9250. When I un-commented, for instance, #define GEN_MPU6050, then the FreeIMU_serial sketch compiled.

mjs513, 2015-02-16:
Strange behavior that I can not duplicate so this will be a problem. The libraries should be in the program folder under libraries.  See this post:  http://arduino.cc/en/Guide/Libraries

A couple of things to try.  (1) Run the FreeIMU_serial_ARM_CPU.ino sketch with the GEN_MPU9250 uncommented and all the other defines commented out (2) copy and paste the defines that I posted earlier over the ones that you currently have.

rs132, 2015-02-16:
Hmm, the same issue: error: 'mag' was not declared in this scope.

Are you also running arduino 1.5.7, or 1.6?

Also, your libraries are presently in Documents\Arduino\libraries, as opposed to C:\Program Files (x86)\Arduino\libraries. Do you think this makes any difference? 

Thanks a lot for trying to assist with this problem, it is very perplexing.

mjs513, 2015-02-16:
It might.  Mine are installed in the program directory. So move them to C:\Program Files (x86)\Arduino\libraries.  I am running 1.5.7.

rs132, 2015-02-16:
Well, it seems that I resolved the problem...very embarrassing for me haha! Inadvertently I had another #define uncommented, so I'm guessing it overrode the #define for the MPU9250 in that case.

However, for the next part of your tutorial, I'm unable to get the cube ode to work. I downloaded the necessary processing folder here https://github.com/mjs513/Processing-2.0b7, and ran the .exe, but nothing seemed to happen. Then, I opened the .pde file (I tried both Cube_Odo and Cube_Odo_Exp_v2), but got a ton of compilation errors. 

Would you happen to know what the cause of this error is?

mjs513, 2015-02-16:
Not a problem on the defines - easy to miss that is why I sent you a copy of what it should be.  It wlll get very confused if you have more than one selected. Ok, just tried it and had the same problem.  Lets me see what I can do.

mjs513, 2015-02-16:
Ok.  can't really figure out why it doesn't work.  Try downloading the latest version and give it a try.  You have a newer computer than I do (at least it sounds like it) so you should support the later versions.

rs132, 2015-02-16:
are you speaking about the processing step?

Perhaps I first need to configure it as such: http://playground.arduino.cc/Interfacing/Processing, before I can get the FreeIMU_cube_Odo.pde to compile?

mjs513, 2015-02-16:
No. I just compared the github directory to my processing directory after I downloaded and there appears to be a number of files that did not appear.  I am reuploading and will test it out and let you know.

rs132, 2015-02-16:
ah ok thanks so much! I really hope this is not too much of an inconvenience for you.

mjs513, 2015-02-16:
No not at all.  Hope it works for you.  Like I said you can always give the latest version of Processing a try as well.  Just remember you will have to set HAS_GPS to 0 and set your com port and baud in the GUI as well.  You may get drifting because you will have to do the calibration for your sensor.

mjs513, 2015-02-16:
In process of uploading a zip file instead of the individual files appears to be an issue with Github not uploading all the files to the server.  Should be done in about 10 minutes.

rs132, 2015-02-16:
Ok thanks, afterwards i'll let you know if any new issues arise, if not, then everything worked!

closed 2015-02-21 by mjs513