mjs513/FreeIMU-Updates #19: Arduino Due Compatibility

yigityildirim, 2015-02-16:
Hello, I have just bought an Arduino Due board, and I have been exploring your FreeIMU library to read the orientation data from a 9DOF sensor (Sparkfun Sensor Stick SEN-10724). However, I could not compile the examples for Arduino Due. I have been getting errors such as "EEPROM.h: No such file or directory compilation terminated." I have learned that the Arduino Due board does not include EEPROM, so is there any modifications we need to do, in order to compile the code? We are using Arduino IDE V.1.6.0. I have been working on a gimbal project, I would appreciate any help.

mjs513, 2015-02-16:
Try using the FreeIMU_serial_ARM_CPU.ino sketch.  There is added code that only enables the EEPROM for AVR boards like the UNO and Mega.  Let me know it works with the 10724 imu.

Mike

yigityildirim, 2015-02-16:
I have tried but I got these errors. I am new to Arduino coding, so I apologize if I am asking unnecessary questions.

In file included from FreeIMU_serial_ARM_CPU.ino:14:0:
/Users/mac/Documents/Arduino/libraries/itg3200filv05/ITG3200.h:46:28: error: expected identifier before numeric constant
 #define WHO_AM_I           0x00  // RW   SETUP: I2C address   
                            ^
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:20:8: note: in expansion of macro 'WHO_AM_I'
        WHO_AM_I       = 0x0F,
        ^
/Users/mac/Documents/Arduino/libraries/itg3200filv05/ITG3200.h:46:28: error: expected '}' before numeric constant
 #define WHO_AM_I           0x00  // RW   SETUP: I2C address   
                            ^
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:20:8: note: in expansion of macro 'WHO_AM_I'
        WHO_AM_I       = 0x0F,
        ^
/Users/mac/Documents/Arduino/libraries/itg3200filv05/ITG3200.h:46:28: error: expected unqualified-id before numeric constant
 #define WHO_AM_I           0x00  // RW   SETUP: I2C address   
                            ^
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:20:8: note: in expansion of macro 'WHO_AM_I'
        WHO_AM_I       = 0x0F,
        ^
In file included from FreeIMU_serial_ARM_CPU.ino:22:0:
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:71:5: error: 'vector' does not name a type
     vector<int16_t> g; // gyro angular velocity readings
     ^
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:75:9: error: expected unqualified-id before 'void'
     L3G(void);
         ^
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:75:9: error: expected ')' before 'void'
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:77:15: error: 'bool init' redeclared as different kind of symbol
     bool init(deviceType device = device_auto, sa0State sa0 = sa0_auto);
               ^
In file included from /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/sam/cores/arduino/Arduino.h:191:0,
                 from /Users/mac/Documents/Arduino/libraries/AP_Filter/RunningAverage.h:20,
                 from /Users/mac/Documents/Arduino/libraries/AP_Filter/Filter.h:15,
                 from FreeIMU_serial_ARM_CPU.ino:2:
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/sam/cores/arduino/wiring.h:31:13: error: previous declaration of 'void init()'
 extern void init( void ) ;
             ^
In file included from FreeIMU_serial_ARM_CPU.ino:22:0:
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:77:15: error: 'deviceType' was not declared in this scope
     bool init(deviceType device = device_auto, sa0State sa0 = sa0_auto);
               ^
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:77:48: error: 'sa0State' was not declared in this scope
     bool init(deviceType device = device_auto, sa0State sa0 = sa0_auto);
                                                ^
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:78:5: error: 'deviceType' does not name a type
     deviceType getDeviceType(void) { return _device; }
     ^
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:92:85: error: 'vector' does not name a type
     template <typename Ta, typename Tb, typename To> static void vector_cross(const vector<Ta> *a, const vector<Tb> *b, vector<To> *out);
                                                                                     ^
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:92:85: error: ISO C++ forbids declaration of 'parameter' with no type [-fpermissive]
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:92:91: error: expected ',' or '...' before '<' token
     template <typename Ta, typename Tb, typename To> static void vector_cross(const vector<Ta> *a, const vector<Tb> *b, vector<To> *out);
                                                                                           ^
In file included from FreeIMU_serial_ARM_CPU.ino:22:0:
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:93:71: error: 'vector' does not name a type
     template <typename Ta, typename Tb> static float vector_dot(const vector<Ta> *a, const vector<Tb> *b);
                                                                       ^
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:93:71: error: ISO C++ forbids declaration of 'parameter' with no type [-fpermissive]
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:93:77: error: expected ',' or '...' before '<' token
     template <typename Ta, typename Tb> static float vector_dot(const vector<Ta> *a, const vector<Tb> *b);
                                                                             ^
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:94:34: error: variable or field 'vector_normalize' declared void
     static void vector_normalize(vector<float> *a);
                                  ^
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:94:34: error: 'vector' was not declared in this scope
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:94:41: error: expected primary-expression before 'float'
     static void vector_normalize(vector<float> *a);
                                         ^
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:96:3: error: expected unqualified-id before 'private'
   private:
   ^
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:103:33: error: 'regAddr' has not been declared
       int testReg(byte address, regAddr reg);
                                 ^
/Users/mac/Documents/Arduino/libraries/L3G/L3G.h:104:1: error: expected declaration before '}' token
 };
 ^
Error compiling.

mjs513, 2015-02-16:
Sounds like you do not have the correct board configured in the FreeIMU.h file.  Did you edit the file so that it looks like the following:

```c++
// Uncomment the appropriated version of FreeIMU you are using
//#define FREEIMU_v01
//#define FREEIMU_v02
//#define FREEIMU_v03
//#define FREEIMU_v035
//#define FREEIMU_v035_MS
//#define FREEIMU_v035_BMP
//#define FREEIMU_v04

// 3rd party boards. Please consider donating or buying a FreeIMU board to support this library development.
//#define SEN_10121 //IMU Digital Combo Board - 6 Degrees of Freedom ITG3200/ADXL345 SEN-10121 http://www.sparkfun.com/products/10121
//#define SEN_10736 //9 Degrees of Freedom - Razor IMU SEN-10736 http://www.sparkfun.com/products/10736/
#define SEN_10724 //9 Degrees of Freedom - Sensor Stick SEN-10724 http://www.sparkfun.com/products/10724
//#define SEN_10183 //9 Degrees of Freedom - Sensor Stick  SEN-10183 http://www.sparkfun.com/products/10183
//#define ARDUIMU_v3 //  DIYDrones ArduIMU+ V3 http://store.diydrones.com/ArduIMU_V3_p/kt-arduimu-30.htm or https://www.sparkfun.com/products/11055
//#define GEN_MPU6050 // Generic MPU6050 breakout board. Compatible with GY-521, SEN-11028 and other MPU6050 wich have the MPU6050 AD0 pin connected to GND.
//#define DFROBOT  //DFROBOT 10DOF SEN-1040 IMU
//#define MPU9250_5611  //MPU-9250 IMU with MS5611 Altimeter from eBay
//#define GEN_MPU9150
//#define GEN_MPU9250  // Use for Invensense MPU-9250 breakout board
//#define Altimu10  // Pololu AltIMU v10 - 10 DOF IMU - http://www.pololu.com/product/1269
//#define GY_88  //GY-88 Sensor Board from eBay
//#define GY_87  //GY-87 Sensor Board from eBay, NOTE: Pressusre sensor is BMP180 but BMP085 library should work
//#define Mario   // MPU-9150 plus Altitude/Pressure Sensor Breakout - MPL3115A2  https://www.sparkfun.com/products/11084
//#define APM_2_5  //  APMM 2.5.2 (EBAY)
//#define Microduino

```

yigityildirim, 2015-02-16:
Yes, I have uncommented the correct line as you suggested. That doesn't seem to be the source of the problem.

mjs513, 2015-02-16:
Try deleting the L3G.h include from the sketch.  I do not have a mac so I can not you if is Mac specific or not.  Also, my libraries are in the program folder.

mjs513, 2015-02-16:
Also please make sure you only have one define uncommented.  Another user had a similar problem and it turned out that was the problem.

yigityildirim, 2015-02-16:
I have managed to compile the code by commenting the L3G.h. Thank you very much!

I tried the processing GUI to validate the sensor readings. There is rapid drifting in all directions and the rotations that I apply to the sensor are quiet higher than what I am seeing in the processing GUI. (It seems as if the cube is moving just a little).

I am guessing it may be due to the lack of calibration. I will see if it will solve the problem. I am open to any further suggestions. Thank you again!

mjs513, 2015-02-16:
Two things - (1) you will have to do a calibration on your sensor as the calibration file is not set up for your board (2) if you still get drifting you may need to tweak the free parameters but do the calibration first.

yigityildirim, 2015-02-16:
Is it possible to do the calibration without using the Eeprom library? Because as far as I understand Arduino Due does not include Eeprom.

mjs513, 2015-02-16:
Yes - you select the option to save to file.  Check the wiki out on how to do the calibration.  I do it all the time this way

closed 2015-02-21 by mjs513