mjs513/FreeIMU-Updates #20: MPU 9250 - Orientation incorrect on the Cube Processing Code

a2kram, 2015-03-03:
Hi mjs513!

Thanks a lot for your efforts. I had a project utilizing the mpu9250 and I was having issues with yaw drifts. I tried running your calibration gui then validating results on the Cube Processing Code, however, the cube was not responsive at all to my orientation changes. I noticed that on your calibration gui plots the scale of the magnetometer output was a lot larger than the scale I was getting. Also, I noticed that the calibration gui output was very different to the calibration9250 file you had for reference. I tried looking through the code but I wasnt able to understand the output of the writeArr function.

[image]

My calibration.h output:
const int acc_off_x = -581;
const int acc_off_y = 291;
const int acc_off_z = 1121;
const float acc_scale_x = 13774.645635;
const float acc_scale_y = 17193.711390;
const float acc_scale_z = 17176.305631;

const int magn_off_x = 43;
const int magn_off_y = 72;
const int magn_off_z = -210;
const float magn_scale_x = 252.625788;
const float magn_scale_y = 239.098613;
const float magn_scale_z = 263.636508;

Anyways, please let me know if the calibration gui is ready to be used for the mpu9250, if the only change required is to define GEN_MPU9250 in the FreeIMU.h file, and if you notice anything wrong with my calibration plots, or any pointers to look for when debugging.

Thanks again!

a2kram

mjs513, 2015-03-04:
Hi a2kram

You are correct. The only thing you have to change in freeimu.h is to uncomment the Gen_9250.  Just remember to comment out all other configurations.

As for your calibration it looks rather complete. Looks like you rotated the board very quickly and not on all axis so you wind up with an incomplre calibration. Suggest you look at wiki calibration page for the library.  Think you will find what you need.

Mike

a2kram, 2015-03-04:
Hi Mike,

Thanks for your quick reply. I fixed it! I found that the issue might have been with the compiler version with the Arduino IDE (I was using 1.05 and upgraded to 1.6). Before the upgrade, I had to make some changes to the FreeIMU code to make it compile (it was complaining that you cannot initialize fields within class definitions, so it made those fields static so they were not visible to the functions using them in FreeIMU.cpp, I just moved all the initializations to the class constuctor). When I upgraded and ran it with a fresh project without my changes, I found that the processing sketch was much more responsive and correct. The calibration gui also impoved my results.

I have a question though, you mentioned that my calibration results look incomplete, is it because of the red box not having a circle in the accelerometer plots? I tried redoing it much slower and following the instructions and videos in the wiki calibration page but I was never able to get the same plots you posted. I was wondering though, wouldnt the accelerometer results be indifferent to yaw about the z-axis since its orthogonal to the gravitational vector? 

Thanks again Mike,

Akram

mjs513, 2015-03-04:
Hi Akram,

Not sure if I mentioned the initialize fields within class definitions issue.  The code works fine with IDE versions 1.5.6 and above.  Think they updated the c compiler used with the GUI.  Thought I updated the code to avoid that - may have forgot to put it on Github will have to check.

You are probably correct about the accelerometer but you are  also calibrating the magnetometer.  Probably have some redundancy but better safe than sorry.  Not sure why you can not duplicate - suggest that you also rotate in a different directions than posted to points on the sphere.  What MCU are you using - if a 32 bit processor like the due you may have to make a change in the python code per the calibration wiki.

The thing that struck me was that  you did not have a circle around the cross in the accel plot and a cross in the other plots For the magnetometer the sphere is incomplete.  Did you look at the comments from Fabio at the bottom of the wiki page for someone who was having a similar problem? The rotations on the wiki page should give you enough points for a good calibration.  Try random rotations to see if you can get the other parts on the sphere to duplicate the charts on the wiki

A couple of other things for you. 1. If you hold control key down use the scroll on the mouse you can zoom out the accel sphere to see what you have. (2) when you first plug the 9250 in let it sit for a minute to let it warm up.

Mike

closed 2015-03-09 by mjs513