mjs513/FreeIMU-Updates #21: Teensy 3.1 and LSM9DS0

ginman, 2015-03-05:
Hi Mike,
I'm having a problem getting the LSM9DS0 working with FreeIMU_GUI. It seems to plot two or three values from only the accelerometer before the whole program hangs and then usually crashes if I try to disconnect or stop sampling. I wrote my own arduino sketch to output the sensor values, based on the sketch for the LSM303. Below is a screenshot where it only got 4 samples, along with my code.

[image]

```javascript
#include <Wire.h>
#include <SPI.h>
#include <Adafruit_LSM9DS0.h>
#include <Adafruit_Sensor.h>  // not used in this demo but required!

// i2c
Adafruit_LSM9DS0 lsm = Adafruit_LSM9DS0();

// You can also use software SPI
//Adafruit_LSM9DS0 lsm = Adafruit_LSM9DS0(13, 12, 11, 10, 9);
// Or hardware SPI! In this case, only CS pins are passed in
//Adafruit_LSM9DS0 lsm = Adafruit_LSM9DS0(10, 9);

char report[80];

void setupSensor()
{
  // 1.) Set the accelerometer range
  lsm.setupAccel(lsm.LSM9DS0_ACCELRANGE_2G);
  //lsm.setupAccel(lsm.LSM9DS0_ACCELRANGE_4G);
  //lsm.setupAccel(lsm.LSM9DS0_ACCELRANGE_6G);
  //lsm.setupAccel(lsm.LSM9DS0_ACCELRANGE_8G);
  //lsm.setupAccel(lsm.LSM9DS0_ACCELRANGE_16G);
  
  // 2.) Set the magnetometer sensitivity
  lsm.setupMag(lsm.LSM9DS0_MAGGAIN_2GAUSS);
  //lsm.setupMag(lsm.LSM9DS0_MAGGAIN_4GAUSS);
  //lsm.setupMag(lsm.LSM9DS0_MAGGAIN_8GAUSS);
  //lsm.setupMag(lsm.LSM9DS0_MAGGAIN_12GAUSS);

  // 3.) Setup the gyroscope
  lsm.setupGyro(lsm.LSM9DS0_GYROSCALE_245DPS);
  //lsm.setupGyro(lsm.LSM9DS0_GYROSCALE_500DPS);
  //lsm.setupGyro(lsm.LSM9DS0_GYROSCALE_2000DPS);
}

void setup() 
{
  while (!Serial); // flora & leonardo
  
  Serial.begin(57600);
  Serial.println("LSM9DS0");
  
  // Try to initialise and warn if we couldn't detect the chip
  if (!lsm.begin())
  {
    Serial.println("Oops ... unable to initialize the LSM9DS0. Check your wiring!");
    while (1);
  }
  //Serial.println("Found LSM9DS0 9DOF");
  //Serial.println("");
  //Serial.println("");
}

void loop() 
{
  lsm.read();
  
  snprintf(report, sizeof(report), "A: %6d %6d %6d    M: %6d %6d %6d",
    (int)lsm.accelData.x, (int)lsm.accelData.y, (int)lsm.accelData.z,
    (int)lsm.magData.x, (int)lsm.magData.y, (int)lsm.magData.z);
  Serial.println(report);
 
  delay(100);
}
```

mjs513, 2015-03-05:
Hi ginman

Guess you know that the adafruit LSM9DS0 isn't supported by the library at this time.  Primarily because I don't have one to test.  Anyway back to your problem, the GUI is looking for packed data that the freeimu_serial.ino sketch outputs when the GUI starts running.  This corresponds to the 'b' command in the sketch.  So would recommend you look at that sketch and copy the code you need to yours to make it compatible.  Here's a hint the line you need is writeArr(raw_values, 9, sizeof(int));  raw_values is the array of accel, gyro and mag values in ints. you will also need to use the Communications.h header which is where the write_arr function is.

Hope this helps.
Mike

ginman, 2015-03-07:
Thanks, I got it working! One question though, what is the idea behind having count = 100 in cal_gui.py?I had to change it to 1 and add the delay in my arduino script in order to get it to work, otherwise it seems like it will just wait for the 'b' command that never gets sent.

mjs513, 2015-03-09:
Hey ginman

Never really got into the GUI code except for that one fix for 32 vs 16bit.  The focus was on the FreeIMU library itself and make improvements and upgrades.  Just to let you know that I have used the GUI on multiple boards and never had a problem with it or it getting stuck. 

I looking at the code for count it looks like Fabio uses it limit the data stream to GUI so you don't run into the problem of having to put delays in the arduino code.  In other words at 115200 which was the original baud rate from Fabio's code it would probably overload the GUI.  What it is doing is transmitting data to the GUI every 100 readings.  I usually run it at 57600 just for info.

Based on your comments think I will tinker around with it a little when I get a chance and see what happens when I have some time.  Have a couple of other projects going right now.

sz830, 2015-03-21:
Hey @ginman 

I'm working on a project and landed at the same issue that you did in this issue.
Could you explain what you did to format your report information correctly?

I copied the write_arr and write_val functions verbatim
and used a modified version of getRawValues.

```
void loop() 
{
  lsm.read();
  getRawValues(raw_values);
  writeArr(raw_values, 9, sizeof(int));
  delay(100);
}

void writeArr(void * varr, uint8_t arr_length, uint8_t type_bytes) {
  byte * arr = (byte*) varr;
  for(uint8_t i=0; i<arr_length; i++) {
    writeVar(&arr[i * type_bytes], type_bytes);
  }
}

void writeVar(void * val, uint8_t type_bytes) {
  byte * addr=(byte *)(val);
  for(uint8_t i=0; i<type_bytes; i++) { 
    Serial.write(addr[i]);
  }
}

void getRawValues(int * raw_values) {
    raw_values[0] = 0;
    raw_values[1] = 0;
    raw_values[2] = 0;
    raw_values[3] = 0;
    raw_values[4] = 0;
    raw_values[5] = 0;
    raw_values[6] = 0;
    raw_values[7] = 0;
    raw_values[8] = 0;
    raw_values[9] = 0;
    raw_values[10] = 0;

      raw_values[0] = (int)lsm.accelData.x;
      raw_values[1] = (int)lsm.accelData.y;
      raw_values[2] = (int)lsm.accelData.z;
      
      raw_values[3] = (int)lsm.gyroData.x;
      raw_values[4] = (int)lsm.gyroData.y;
      raw_values[5] = (int)lsm.gyroData.z;
      
      raw_values[6] = (int)lsm.magData.x;
      raw_values[7] = (int)lsm.magData.y;
      raw_values[8] = (int)lsm.magData.z;

}
```

mjs513, 2015-03-22:
Hi sz830

ginman, 2015-03-23:
Your code looks ok. In cal_gui.py did you change it to count = 1 in the run loop? For reference here is my arduino code:

```javascript
#include <Wire.h>
#include <SPI.h>
#include <Adafruit_LSM9DS0.h>
#include <Adafruit_Sensor.h>  // not used in this demo but required!
#include "CommunicationUtils.h"
// i2c
Adafruit_LSM9DS0 lsm = Adafruit_LSM9DS0();

// You can also use software SPI
//Adafruit_LSM9DS0 lsm = Adafruit_LSM9DS0(13, 12, 11, 10, 9);
// Or hardware SPI! In this case, only CS pins are passed in
//Adafruit_LSM9DS0 lsm = Adafruit_LSM9DS0(10, 9);

char report[80];
int raw_values[9];

void setupSensor()
{
  // 1.) Set the accelerometer range
  lsm.setupAccel(lsm.LSM9DS0_ACCELRANGE_2G);
  //lsm.setupAccel(lsm.LSM9DS0_ACCELRANGE_4G);
  //lsm.setupAccel(lsm.LSM9DS0_ACCELRANGE_6G);
  //lsm.setupAccel(lsm.LSM9DS0_ACCELRANGE_8G);
  //lsm.setupAccel(lsm.LSM9DS0_ACCELRANGE_16G);
  
  // 2.) Set the magnetometer sensitivity
  lsm.setupMag(lsm.LSM9DS0_MAGGAIN_2GAUSS);
  //lsm.setupMag(lsm.LSM9DS0_MAGGAIN_4GAUSS);
  //lsm.setupMag(lsm.LSM9DS0_MAGGAIN_8GAUSS);
  //lsm.setupMag(lsm.LSM9DS0_MAGGAIN_12GAUSS);

  // 3.) Setup the gyroscope
  lsm.setupGyro(lsm.LSM9DS0_GYROSCALE_245DPS);
  //lsm.setupGyro(lsm.LSM9DS0_GYROSCALE_500DPS);
  //lsm.setupGyro(lsm.LSM9DS0_GYROSCALE_2000DPS);
}

void setup() 
{
  while (!Serial); // flora & leonardo
  
  Serial.begin(57600);
  Serial.println("LSM9DS0");
  
  // Try to initialise and warn if we couldn't detect the chip
  if (!lsm.begin())
  {
    Serial.println("Oops ... unable to initialize the LSM9DS0. Check your wiring!");
    while (1);
  }
  //Serial.println("Found LSM9DS0 9DOF");
  //Serial.println("");
  //Serial.println("");
}

void loop() 
{
  lsm.read();
  /*
  snprintf(report, sizeof(report), "A: %6d %6d %6d    M: %6d %6d %6d",
    (int)lsm.accelData.x, (int)lsm.accelData.y, (int)lsm.accelData.z,
    (int)lsm.magData.x, (int)lsm.magData.y, (int)lsm.magData.z);
  Serial.println(report);
 */
 // accel values
 raw_values[0] = (int)lsm.accelData.x;
 raw_values[1] = (int)lsm.accelData.y;
 raw_values[2] = (int)lsm.accelData.z;
 // gyro val
 raw_values[3] = (int)lsm.gyroData.x;
 raw_values[4] = (int)lsm.gyroData.y;
 raw_values[5] = (int)lsm.gyroData.z;
 // mag values
 raw_values[6] = (int)lsm.magData.x;
 raw_values[7] = (int)lsm.magData.y;
 raw_values[8] = (int)lsm.magData.z;
 writeArr(raw_values, 9, sizeof(int));
  delay(100);
}
```

sz830, 2015-03-23:
Could you tell me what the purpose of count is?
I changed the code but things still aren't working. I'll try again later today and try to post some more details.
Thanks for the response!

mjs513, 2015-03-23:
Hi

Am at work now but think I tried to answer that a couple of posts ago.

Fabio's serial code would output a fixed number of readings when a serial
command is sent. I didn't change this approach, at least yet. So the GUI
sends out a command to the serial sketch to get the data for the GUI.  The
sketch u have just sends out data so u need to make count 1.

Mike

closed 2015-04-15 by mjs513