mjs513/FreeIMU-Updates #23: can't compile

fightingsoul009, 2015-03-08:
I am using GY-88 module with arduino uno and ide is arduino 1.0.6 . 
I am relatively new to this any help will be greatly appreciated.
I get the following error.
  

In file included from FreeIMU_yaw_pitch_roll.ino:32:
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:496: error: ISO C++ forbids initialization of member 'zeroMotioncount'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:496: error: making 'zeroMotioncount' static
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:496: error: ISO C++ forbids in-class initialization of non-const static member 'zeroMotioncount'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:511: error: a brace-enclosed initializer is not allowed here before '{' token
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:511: error: ISO C++ forbids initialization of member 'sensor_order'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:511: error: making 'sensor_order' static
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:511: error: invalid in-class initialization of static data member of non-integral type 'int [9]'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:512: error: a brace-enclosed initializer is not allowed here before '{' token
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:512: error: ISO C++ forbids initialization of member 'sensor_sign'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:512: error: making 'sensor_sign' static
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:512: error: invalid in-class initialization of static data member of non-integral type 'int [9]'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:544: error: ISO C++ forbids initialization of member 'lastUpdate1'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:544: error: making 'lastUpdate1' static
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:544: error: ISO C++ forbids in-class initialization of non-const static member 'lastUpdate1'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:552: error: ISO C++ forbids initialization of member 'SEq_1'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:552: error: making 'SEq_1' static
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:552: error: ISO C++ forbids in-class initialization of non-const static member 'SEq_1'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:552: error: ISO C++ forbids initialization of member 'SEq_2'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:552: error: making 'SEq_2' static
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:552: error: ISO C++ forbids in-class initialization of non-const static member 'SEq_2'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:552: error: ISO C++ forbids initialization of member 'SEq_3'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:552: error: making 'SEq_3' static
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:552: error: ISO C++ forbids in-class initialization of non-const static member 'SEq_3'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:552: error: ISO C++ forbids initialization of member 'SEq_4'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:552: error: making 'SEq_4' static
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:552: error: ISO C++ forbids in-class initialization of non-const static member 'SEq_4'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:553: error: ISO C++ forbids initialization of member 'b_x'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:553: error: making 'b_x' static
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:553: error: ISO C++ forbids in-class initialization of non-const static member 'b_x'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:553: error: ISO C++ forbids initialization of member 'b_z'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:553: error: making 'b_z' static
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:553: error: ISO C++ forbids in-class initialization of non-const static member 'b_z'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:554: error: ISO C++ forbids initialization of member 'w_bx'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:554: error: making 'w_bx' static
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:554: error: ISO C++ forbids in-class initialization of non-const static member 'w_bx'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:554: error: ISO C++ forbids initialization of member 'w_by'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:554: error: making 'w_by' static
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:554: error: ISO C++ forbids in-class initialization of non-const static member 'w_by'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:554: error: ISO C++ forbids initialization of member 'w_bz'
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:554: error: making 'w_bz' static
C:\Users\Asad Ali\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:554: error: ISO C++ forbids in-class initialization of non-const static member 'w_bz'

mjs513, 2015-03-08:
Have to use Arduino IDE 1.5.6 or better.  Should check out some of the closed issues, other people have had the same problem.  Also check out the wiki of of the library.

Mjs

fightingsoul009, 2015-03-08:
after using arduino ide 1.6.0 , I am getting the following error.

In file included from FreeIMU_yaw_pitch_roll.ino:13:0:

C:\Users\Asad Ali\Documents\Arduino\libraries\itg3200filv05/ITG3200.h:46:28: error: expected identifier before numeric constant

 #define WHO_AM_I           0x00  // RW   SETUP: I2C address   

                            ^

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:20:8: note: in expansion of macro 'WHO_AM_I'

        WHO_AM_I       = 0x0F,

        ^

C:\Users\Asad Ali\Documents\Arduino\libraries\itg3200filv05/ITG3200.h:46:28: error: expected '}' before numeric constant

 #define WHO_AM_I           0x00  // RW   SETUP: I2C address   

                            ^

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:20:8: note: in expansion of macro 'WHO_AM_I'

        WHO_AM_I       = 0x0F,

        ^

C:\Users\Asad Ali\Documents\Arduino\libraries\itg3200filv05/ITG3200.h:46:28: error: expected unqualified-id before numeric constant

 #define WHO_AM_I           0x00  // RW   SETUP: I2C address   

                            ^

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:20:8: note: in expansion of macro 'WHO_AM_I'

        WHO_AM_I       = 0x0F,

        ^

In file included from FreeIMU_yaw_pitch_roll.ino:21:0:

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:71:5: error: 'vector' does not name a type

     vector<int16_t> g; // gyro angular velocity readings

     ^

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:75:9: error: expected unqualified-id before 'void'

     L3G(void);

         ^

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:75:9: error: expected ')' before 'void'

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:77:15: error: 'bool init' redeclared as different kind of symbol

     bool init(deviceType device = device_auto, sa0State sa0 = sa0_auto);

               ^

In file included from C:\Users\Asad Ali\Documents\Arduino\libraries\AP_Filter/RunningAverage.h:20:0,

                 from C:\Users\Asad Ali\Documents\Arduino\libraries\AP_Filter/Filter.h:15,

                 from FreeIMU_yaw_pitch_roll.ino:2:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:120:6: error: previous declaration of 'void init()'

 void init(void);

      ^

In file included from FreeIMU_yaw_pitch_roll.ino:21:0:

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:77:15: error: 'deviceType' was not declared in this scope

     bool init(deviceType device = device_auto, sa0State sa0 = sa0_auto);

               ^

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:77:48: error: 'sa0State' was not declared in this scope

     bool init(deviceType device = device_auto, sa0State sa0 = sa0_auto);

                                                ^

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:78:5: error: 'deviceType' does not name a type

     deviceType getDeviceType(void) { return _device; }

     ^

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:92:85: error: 'vector' does not name a type

     template <typename Ta, typename Tb, typename To> static void vector_cross(const vector<Ta> *a, const vector<Tb> *b, vector<To> *out);

                                                                                     ^

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:92:85: error: ISO C++ forbids declaration of 'parameter' with no type [-fpermissive]

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:92:91: error: expected ',' or '...' before '<' token

     template <typename Ta, typename Tb, typename To> static void vector_cross(const vector<Ta> *a, const vector<Tb> *b, vector<To> *out);

                                                                                           ^

In file included from FreeIMU_yaw_pitch_roll.ino:21:0:

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:93:71: error: 'vector' does not name a type

     template <typename Ta, typename Tb> static float vector_dot(const vector<Ta> *a, const vector<Tb> *b);

                                                                       ^

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:93:71: error: ISO C++ forbids declaration of 'parameter' with no type [-fpermissive]

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:93:77: error: expected ',' or '...' before '<' token

     template <typename Ta, typename Tb> static float vector_dot(const vector<Ta> *a, const vector<Tb> *b);

                                                                             ^

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:94:34: error: variable or field 'vector_normalize' declared void

     static void vector_normalize(vector<float> *a);

                                  ^

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:94:34: error: 'vector' was not declared in this scope

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:94:41: error: expected primary-expression before 'float'

     static void vector_normalize(vector<float> *a);

                                         ^

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:96:3: error: expected unqualified-id before 'private'

   private:

   ^

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:103:33: error: 'regAddr' has not been declared

       int testReg(byte address, regAddr reg);

                                 ^

C:\Users\Asad Ali\Documents\Arduino\libraries\L3G/L3G.h:104:1: error: expected declaration before '}' token

 };

 ^

Error compiling.

mjs513, 2015-03-08:
Did  you remember to configure Freeimu.h for the IMU that you are using.  What IMU/board are you using?

fightingsoul009, 2015-03-08:
Yes I did. I am using gy-88 .

mjs513, 2015-03-08:
Please double check.  You uncomment the GY-88 line and make sure there is no other board uncommented.  The error you are getting is typical when you have more than one board uncommented.  Check the Altimu10 board line and make sure it is commented.  I just tried it and it seems work for me.

Try this sketch:
```c++
#include <AP_Math_freeimu.h>
#include <Filter.h>    // Filter library
#include <Butter.h>    // Butterworth filter
#include <iCompass.h>

/**
 * FreeIMU library serial communication protocol
*/

#include <ADXL345.h>
#include <HMC58X3.h>
#include <LSM303.h>
#include <ITG3200.h>
#include <bma180.h>
#include <MS561101BA.h>
#include <BMP085.h>
#include <I2Cdev.h>
#include <MPU60X0.h>
#include <AK8975.h>
#include <AK8963.h>
#include <L3G.h>
#include <LPS331.h> 
#include <AP_Baro_MS5611.h>

#include <Wire.h>
#include <SPI.h>
#include <EEPROM.h>

//#define DEBUG
#include "DebugUtils.h"
#include "CommunicationUtils.h"
#include "FreeIMU.h"
#include "FilteringScheme.h"
#include "RunningAverage.h"
#include <MovingAvarageFilter.h>

int raw_values[11];
//char str[512];
float ypr[3]; // yaw pitch roll
float val[9];

// Set the FreeIMU object
FreeIMU my3IMU = FreeIMU();

void setup() {
  Serial.begin(115200);
  Wire.begin();
  
  delay(5);
  my3IMU.init(); // the parameter enable or disable fast mode
  delay(5);
}

void loop() { 
  
  my3IMU.getYawPitchRoll(ypr);
  Serial.print("Yaw: ");
  Serial.print(ypr[0]);
  Serial.print(" Pitch: ");
  Serial.print(ypr[1]);
  Serial.print(" Roll: ");
  Serial.print(ypr[2]);
  Serial.println("");
  
  delay(10);
}

```

fightingsoul009, 2015-03-08:
its like this in FreeIMU.h .I am still getting the same errors with the new sketch.

// Uncomment the appropriated version of FreeIMU you are using
//#define FREEIMU_v01
//#define FREEIMU_v02
//#define FREEIMU_v03
//#define FREEIMU_v035
//#define FREEIMU_v035_MS
//#define FREEIMU_v035_BMP
//#define FREEIMU_v04

// 3rd party boards. Please consider donating or buying a FreeIMU board to support this library development.
//#define SEN_10121 //IMU Digital Combo Board - 6 Degrees of Freedom ITG3200/ADXL345 SEN-10121 http://www.sparkfun.com/products/10121
//#define SEN_10736 //9 Degrees of Freedom - Razor IMU SEN-10736 http://www.sparkfun.com/products/10736/
//#define SEN_10724 //9 Degrees of Freedom - Sensor Stick SEN-10724 http://www.sparkfun.com/products/10724
//#define SEN_10183 //9 Degrees of Freedom - Sensor Stick  SEN-10183 http://www.sparkfun.com/products/10183
//#define ARDUIMU_v3 //  DIYDrones ArduIMU+ V3 http://store.diydrones.com/ArduIMU_V3_p/kt-arduimu-30.htm or https://www.sparkfun.com/products/11055
//#define GEN_MPU6050 // Generic MPU6050 breakout board. Compatible with GY-521, SEN-11028 and other MPU6050 wich have the MPU6050 AD0 pin connected to GND.
//#define DFROBOT  //DFROBOT 10DOF SEN-1040 IMU
//#define MPU9250_5611  //MPU-9250 IMU with MS5611 Altimeter from eBay
//#define GEN_MPU9150
//#define GEN_MPU9250  // Use for Invensense MPU-9250 breakout board
//#define Altimu10  // Pololu AltIMU v10 - 10 DOF IMU - http://www.pololu.com/product/1269
#define GY_88  //GY-88 Sensor Board from eBay
//#define GY_87  //GY-87 Sensor Board from eBay, NOTE: Pressusre sensor is BMP180 but BMP085 library should work
//#define Mario   // MPU-9150 plus Altitude/Pressure Sensor Breakout - MPL3115A2  https://www.sparkfun.com/products/11084
//#define APM_2_5  //  APMM 2.5.2 (EBAY)
//#define Microduino

mjs513, 2015-03-08:
Set the includes in the YPR like this:
```c++
#include <AP_Math_freeimu.h>
#include <Filter.h>    // Filter library
#include <Butter.h>    // Butterworth filter
#include <iCompass.h>

/**
 * FreeIMU library serial communication protocol
*/

#include <ADXL345.h>
#include <HMC58X3.h>
#include <I2Cdev.h>
#include <MPU60X0.h>

#include <Wire.h>
#include <SPI.h>
#include <EEPROM.h>

//#define DEBUG
#include "DebugUtils.h"
#include "CommunicationUtils.h"
#include "FreeIMU.h"
#include "FilteringScheme.h"
#include "RunningAverage.h"
#include <MovingAvarageFilter.h>
```
If doesn't work not sure what to tell you - your config of freeimu.h looks correct. Try checking out some of the closed issues to see if anything applies. If I get time I will download and 1.6 and give it a try.  I am currenrly using 1.5.6.

mjs513, 2015-03-08:
The only other thing that you can try is to move the libraries into the arduino library directory located in the directory where the arduino program is located on your computer.

mjs513, 2015-03-09:
I downloaded 1.6 and had no issues with compiling the sketch. Suggest that you move the libraries to the Arduino Library directory proper as opposed to your My Documents Arduino folder and see it works if the above sketch does not work for you. Hope you got it working

closed 2015-03-09 by mjs513

7hny, 2015-04-22:
I had the same problem, also with GY-88. Code worked after I reordered imports, `L3G.h` before `ITG3200.h`.

```
#include <L3G.h>
#include <ITG3200.h>
```

mjs513, 2015-04-22:
One of the things that's not documented, and probably really should make a note of it, is that you should delete the IMU libraries that you don't need.  So for the GY-88 you only need the libraries for the 5883L, BMP085 and the MPU6050.  

Mike