mjs513/FreeIMU-Updates #25: Unstable gravity vector

cleaf30, 2015-03-28:
I am using a freeIMU to capture dynamic acceleration but I getting some weird data. below is output with the IMU sitting still. The g(x,y,z) start off fine then all the sudden their is a drastic change(labeled below). The raw values all seem pretty consistent so I'd imagine it has something to do with the algorithm. The code I'm using to get g(x,y,z) is the same as whats given as a help function at the bottom of freeIMU.cpp

  g[0] = 2 * (q[1] * q[3] - q[0] * q[2]);
  g[1] = 2 * (q[0] * q[1] + q[2] * q[3]);
  g[2] = q[0] * q[0] - q[1] * q[1] - q[2] * q[2] + q[3] * q[3];
  G = sqrt(g[0] * g[0] + g[1] * g[1] + g[2] * g[2]);
                                                                                             
gx|gy|gz|G(magnitude)|raw[0] |raw[1]|raw[2]|raw[3]|raw[4]|raw[5]|raw[6]|raw[7]|raw[8]|temp|pressure
-----|-----------|----------|-----------------|-------|------|----------|-------|-----|------|--------|-----|--------|---------|------------
0.04|	0.00	|        1.00	|	1.00|		226|	116|	17304|	-44|	17|	-12|	-78|	123|	-606	|0.00,|0.00,
0.05|	-0.00|	1.00|		1.00	|	172|	134|	17312|	-45|	17|	-13|	-82|	125|	-607	|27.06,|0.00,
0.05	|-0.00|	1.00	|	1.00	|	190|	138|	17258|	-44|	18|	-13|	-78|	123|	-608	|27.06,|1008.03,
0.05|	-0.01|	1.00	|	1.00	|	206|	100|	17314|	-43|	18|	-13|	-79|	124|	-607	|27.08,|1008.08,
0.06|-0.01|	1.00	|	1.00	|	198|	118|	17296|	-44|   18|	-14|	-79|	123| -606	|27.08,|1008.04,
0.06|	-0.01|	1.00	|	1.00	|	202|	88	|17276|	-45|	18|	-13|	-78|	124|	-610	|27.09,|1008.05,
0.07|	-0.01|	1.00	|	1.00	|	210|	120	|17278|	-45|	18|	-12|	-79|	125|	-610	|27.09,|1008.03,
0.07|	-0.02|	1.00	|	1.00	|	188|	108|	17266|	-44|	17|	-13|	-78|	121|	-608	|27.08,|1008.02,
0.07|	-0.02|	1.00	|	1.00	|	222|	168|	17300|	-45|	18|	-13|	-80|	123|	-605	|27.08,|1008.02,
0.08|	-0.02|	1.00	|	1.00	|	190|	136|	17362|	-44|	17|	-12|	-79|	125|	-605	|27.09,|1008.04,
0.08	|-0.03|	1.00	|	1.00	|	218|	74	|17272|	-45|	18|	-13|	-76|	122|	-606	|27.09,|1008.05,
0.08	|-0.03|	1.00	|	1.00	|	202|	132|	17292|	-45|	19|	-12|	-79|	125|	-603	|27.09,|1008.04,
0.09	|-0.03|	1.00	|	1.00	|	166|	118|	17324|	-45|	18|	-13|	-77|	124|	-605|	27.09,|1008.01,
-0.90|	-0.01|	-0.44|		1.00	|	218|	144	|17328	|-44|	18|	-13|	-78|	121|	-604	|27.09,|1008.01,
-0.90|	-0.01|	-0.43|		1.00	|	208|	134	|17292	|-44|	18|	-12|	-79|	124|	-611	|27.09,|1008.00,
-0.90|	-0.01|	-0.44|		1.00	|	190|	144	|17292|	-44|	18|	-13|	-77|	123|	-603	|27.09,|1008.01,
-0.90|	-0.01|	-0.44|		1.00	|	212|	120	|17230|	-44|	18|	-13|	-78|	122|	-608	|27.09,|1008.07,
-0.90|	-0.01|	-0.44|		1.00	|	226|	104	|17322|	-45|	18|	-13|	-77|	124|	-607	|27.09,|1008.07,
-0.90|	-0.01|	-0.44|		1.00	|	196|	92|	17292|	-44|	18|	-13|	-78|	123|	-601	|27.09,|1008.04,
-0.89|	-0.01|	-0.45|		1.00	|	196|	104|	17266|	-45|	18|	-12|	-78|	122|	-605	|27.09,|1008.06,
-0.89|	-0.01|	-0.45|		1.00	|	190|	86|	17300|	-44|	18|	-13|	-80|	124|	-605	|27.09,|1008.01,
-0.90|	-0.01|	-0.44|		1.00	|	212|	114|	17254|	-45|	18|	-12|	-79|	123|	-608	|27.09,|1008.00,
-0.90|	-0.01|	-0.44|		1.00	|	212|	154|	17266|	-44|	18|	-12|	-79|	124|	-603	|27.09,|1007.99,
-0.90|	-0.01|	-0.44|		1.00	|	174|	116|	17314|	-44|	18|	-12|	-77|	123|	-608	|27.09,|1007.99,
-0.90|	-0.01|	-0.44|		1.00	|	184|	108|	17280|	-44|	18|	-12|	-81|	125|	-603	|27.09,|1008.04,
-0.90|	-0.01|	-0.44|		1.00	|	198|	102|	17318|	-44|	18|	-13|	-74|	121|	-607	|27.09,|1008.04,
-0.89|	-0.01|	-0.46|		1.00	|	196|	74|	17302|	-44|	18|	-12|	-79|	123|	-609	|27.09,|1008.07,
-0.90|	-0.01|	-0.44|		1.00	|	178	|100|	17320|	-45|	18|	-12|	-78|	125|	-606	|27.08,|1008.05,
-0.89|	-0.01|	-0.45|		1.00	|	200	|142	|17378|	-45|	18|	-12|	-77|	124|	-609	|27.08,|1008.04,
-0.90|	-0.01|	-0.44|		1.00	|	198	|62	|17306|	-44|	18|	-13|	-79|	125|	-607	|27.08,|1008.05,
-0.89|	-0.01|	-0.45|		1.00	|	186	|138	|17306|	-45|	17|	-13|	-78|	123|	-609	|27.08,|1008.03,
-0.90|	-0.01|	-0.44|		1.00	|	206	|110	|17304|	-44|	18|	-13|	-76|	126|	-604	|27.08,|1008.01,
-0.89|	-0.01|	-0.45|		1.00	|	148	|144	|17284|	-44|	17|	-13|	-78|	125|	-606	|27.08,|1007.99,

I noticed in the hookup guide it has the mega connected directly to the IMU. I've read about having to put in a level shifter to hook it up to the mega due to 5V pull ups. So I currently have a level shifter on the SDA, SCL lines set at 3.3V on the low side coming off the IMU's 3.3V pin and 5V on the high side coming off the mega. Is this necessary could it be causing me problems?
 
Also before experiencing this issue with my gravity vector I had a problem with the settling time for accelerations being way too long 10+ seconds. Any idea why this was happening or what I could do? And yes I have calibrated my IMU.

Thanks,
Cody

mjs513, 2015-03-28:
Hi Cody,

When you say you are using a freeimu are you actually using a FreeIMU v4 board or some other board? Also, which algorithm option are you using (set in FreeIMU.h): Marg = 0, 1,3,or 4?  The reason I ask is I can try and isolate the problem and hookup the equivalent board that I have.  I am also assuming you set the board configuration in the FreeIMU.h file as well.  One issue some folks had is they uncommented the board they have but forgot to comment out the one that is already uncommented. 

The data does look very strange when compared to the accelerations.  Look at those its telling me that you have the board steady but the vector directions changed.

As for the settling times you can tweak the gains that I have set as defaults for the different configurations to improve the converge time.

As for the level shifters that should not be an issue as I did that when hooking up to a Teensy.  Also, looking at some of Fabio's hookup guides he has it hooked directly to the +5v (vdd) and direct to the scl/scd lines.

Mike

mjs513, 2015-03-28:
Guess you resolved you problem - what was the issue.

cleaf30, 2015-03-28:
No I haven't resolved it I just tried to edit a sentence and ended up making a new issue. I don't know how to delete an issue so I closed it. Yes I have a v4r3. I'm currently using a marg = 3. I have the correct board uncommented.

mjs513, 2015-03-28:
Ok thanks this way I can hook mine up.  Can you post the sketch you used, will save me some time.  Probably will look at it tonight,  If you have the version with DCM incorporated give Marg = 4 a try and see if you get the same thing.  As for the tweaking of the gains, that should help with your other probably with settling time.  This was actually discussed in one of the other issues if you go rummaging around.

cleaf30, 2015-03-28:
I've just tried with marg = 0, 1, 3, and 4 and got similar results. Although I'm not sure if I have the DCM version, there wasn't anything in the comment about marg = 4 but it compiled. I downloaded the FreeIMU library a month or so ago. 

here's my sketch. thank for the help.

#include <HMC58X3.h>
#include <MS561101BA.h>
#include <I2Cdev.h>
#include <MPU60X0.h>
#include <EEPROM.h>
#include <AP_Baro_MS5611.h>
#include "RunningAverage.h"
#include <iCompass.h>

#include <BMP085.h>

#include <AP_Math_freeimu.h>
#include <Filter.h>    // Filter library
#include <Butter.h>    // Butterworth filter

#include <EEPROM.h>
#include <Wire.h>
#include <SPI.h>

//#define DEBUG
#include "DebugUtils.h"
#include "CommunicationUtils.h"
#include "FreeIMU.h"
#include "FilteringScheme.h"
#include <MovingAvarageFilter.h>

float q[4], val[11];
int raw_values[11];
char str[512];

float g[3],G;//gravity
int S;   //conversion factor to g's??
float rx, ry, rz, R;//raw acc
float cx, cy, cz, C; //dynamic acc
float dx, dy, dz, dynamAccelXY;//dynamic accleration in XY(horizontal) plane

// Set the FreeIMU object
FreeIMU my3IMU = FreeIMU();

//my3IMU.setTempCalib(0); // 0 - turn temp calibration off
void setup() {
  Serial.begin(115200);
  Wire.begin();
  
  delay(5);
  my3IMU.init();
  delay(5);
  
  S = 16384;
}

void loop() { 
  my3IMU.getQ(q,val);
  my3IMU.getRawValues(raw_values);
  
  //calibrated
  rx = val[0];
  ry = val[1];
  rz = val[2];
  
  R = sqrt(rx*rx + ry*ry + rz*rz);
  
  g[0] = 2 * (q[1] * q[3] - q[0] * q[2]);
  g[1] = 2 * (q[0] * q[1] + q[2] * q[3]);
  g[2] = q[0] * q[0] - q[1] * q[1] - q[2] * q[2] + q[3] * q[3];
  G = sqrt(g[0]*g[0] + g[1]*g[1] + g[2]*g[2]);
  
  //dymanic acceleration: accel values - gravity
  cx = rx - g[0];
  cy = ry - g[1];
  cz = rz - g[2];
  C = sqrt(cx*cx + cy*cy + cz*cz);
  
  //dynamic accel in xy plane (horizontal)
  //cross product of dynamic accel vector and orientation unit vector
   dx = (g[1]*cz - g[2]*cy);
   dy = -(g[2]*cx - g[0]*cz);
   dz = (g[0]*cy - g[1]*cx);
   dynamAccelXY = sqrt(dx*dx + dy*dy + dz*dz);
  
  Serial.print(g[0]);
  Serial.print("\t");
  Serial.print(g[1]);
  Serial.print("\t");
  Serial.print(g[2]);
  Serial.print("\t\t");
  Serial.print(G);
  Serial.print("\t\t");
//  //Raw
//  Serial.print(rx);
//  Serial.print("\t");
//  Serial.print(ry);
//  Serial.print("\t");
//  Serial.print(rz);
//  Serial.print("\t\t");
//  Serial.print(R);
//  Serial.print("\t\t");
//  //Dynamic  
//  Serial.print(cx);
//  Serial.print("\t");
//  Serial.print(cy);
//  Serial.print("\t");
//  Serial.print(cz);
//  Serial.print("\t\t");
//  Serial.print(C);
//  //Dynamic XY
//  Serial.print("\t\t");
//  Serial.print(dynamAccelXY);
//  Serial.print("\t");
  sprintf(str, "%d\t%d\t%d\t%d\t%d\t%d\t%d\t%d\t%d\t", raw_values[0], raw_values[1], raw_values[2], raw_values[3], raw_values[4], raw_values[5], raw_values[6], raw_values[7], raw_values[8]);
  Serial.print(str);
  Serial.print(my3IMU.getBaroTemperature()); Serial.print(",");
  Serial.print(my3IMU.getBaroPressure()); Serial.print(",");
  Serial.println("");
  delay(20);
  
}

mjs513, 2015-03-28:
Thanks.  The addition of the DCM for Marg = 4 was just posted Thursday.

 

#include 
#include 
#include 
#include 
#include 

#include 

#include 
#include // Filter library
#include // Butterworth filter

#include 
#include 
#include

cleaf30, 2015-03-28:
I just tried it with DCM and got similar results. It stabilizes after the drastic change. Could it be resetting itself?

gx|gy|gz|G|raw[0]|raw[1]|raw[2]|raw[3]|raw[4]|raw[5]|raw[6]|raw[7]|raw[8]|temp|pressure
----|----|---|----------|--------|-------|---------|---------|--------|--------|--------|--------|--------|-----|------
0.00|	0.00|	0.00|		0.00|		362|	354|	17412|	-42|	17|	-13|	-33|	20|	-605|	0.00,|0.00,
0.00|	-0.00|	0.00|		0.00|		368|	370|	17340|	-43|	17|	-13|	-31|	21|	-606|	25.10,|0.00,
0.00|	-0.00|	0.00|		0.00|		390|	342|	17342|	-42|	17|	-13|	-35|	20|	-603|	25.10,|1007.12,
0.00|	0.00|	1.01|		1.01|		394|	384|	17356|	-43|	17|	-13|	-37|	22|	-603|	25.15,|1007.24,
0.00|	0.00|	1.02|		1.02|		360|	358|	17356|	-42|	17|	-13|	-37|	23|	-602|	25.15,|1007.14,
0.00	|0.01|	1.03|		1.03|		364|	352|	17366|	-43|	17|	-13|	-32|	20|	-602|	25.18,|1007.23,
0.00|	0.01|	1.04|		1.04|		410|	358|	17332|	-42|	17|	-14|	-35|	23|	-608|	25.18,|1007.15,
0.00|	0.01|	1.05|		1.05|		370|	382|	17356|	-42|	17|	-14|	-34|	20|	-605|	25.21,|1007.20,
0.00|	0.01|	1.07|		1.07|		328|	360|	17330|	-42|	17|	-12|	-36|	20|	-605|	25.21,|1007.08,
0.00|	0.01|	1.09|		1.09|		360|	362|	17362|	-42|	17|	-13|	-35|	20|	-606|	25.22,|1007.12,
0.00|	0.01|	1.11|		1.11|		392|	370|	17342|	-42|	18|	-13|	-34|	20|	-602|	25.22,|1007.10,
0.00|	0.01|	1.13|		1.13|		396|	330|	17328|	-42|	17|	-12|	-36|	20|	-605|	25.24,|1007.14,
0.00|	0.01|	1.15|		1.15|		336|	344|	17350|	-42|	17|	-13|	-32|	20|	-598|	25.24,|1007.14,
-0.88|	-0.02|	-0.47|		1.00|		372|	346|	17388|	-42|	17|	-14|	-35|	22|	-606|	25.25,|1007.17,
-0.90|	-0.02|	-0.44|		1.00|		360	|370|	17384|	-42|	16|	-12|	-37|	19|	-606|	25.25,|1007.16,
-0.89|	-0.02|	-0.45|		1.00|		386|	370	|17342|	-42|	17|	-13|	-35|	21|	-605|	25.25,|1007.16,
-0.89|	-0.02|	-0.45|		1.00|		378|	308|	17408|	-43|	18|	-12|	-34|	19|	-608|	25.25,|1007.16,
-0.89|	-0.02|	-0.46|		1.00|		348|	338|	17330|	-42|	17|	-13|	-32|	20|	-604|	25.27,|1007.19,
-0.89|	-0.02|	-0.45|		1.00|		376|	340|	17342|	-42|	17|	-13|	-35|	20|	-604|	25.27,|1007.12,
-0.90|	-0.03|	-0.44|		1.00|		366|	350|	17364|	-43|	18|	-13|	-35|	20|	-604|	25.27,|1007.13,
-0.89|	-0.02|	-0.45|		1.00|		374|	380|	17310|	-42|	17|	-13|	-34|	20|	-600|	25.27,|1007.20,
-0.90|	-0.02|	-0.43|		1.00|		370|	354|	17374|	-42|	17|	-13|	-36|	22|	-600|	25.28,|1007.21,
-0.90|	-0.02|	-0.43|		1.00|		364|	372|	17382|	-42|	17|	-13|	-34|	20|	-604|	25.28,|1007.18,
-0.90|	-0.02|	-0.43|		1.00|		364|	350|	17366|	-42|	17|	-13|	-35|	22|	-603|	25.28,|1007.17,
-0.90|	-0.02|	-0.44|		1.00|		380|	380|	17356|	-42|	17|	-13|	-37|	20|	-606|	25.28,|1007.17,
-0.89|	-0.02|	-0.45|		1.00|		394|	384|	17318|	-41|	18|	-13|	-34|	20|	-605|	25.28,|1007.18,
-0.89|	-0.02|	-0.45|		1.00|		380|	380|	17378|	-42|	17|	-13|	-34|	23|	-607|	25.28,|1007.21,
-0.89| -0.02|	-0.46|		1.00|		404|	366|	17408|	-42|	17|	-13|	-34|	21|	-605|	25.28,|1007.20,
-0.89|	-0.02|	-0.45|		1.00|		422|	364|	17322|	-42|	17|	-13|	-33|	22|	-599|	25.28,|1007.18,
-0.90|	-0.02|	-0.43|		1.00|		368|	370|	17340|	-42|	17|	-13|	-34|	19|	-606|	25.29,|1007.19,
-0.89|	-0.02|	-0.46|		1.00|		380|	348|	17350|	-42|	17|	-13|	-33|	23|	-598|	25.29,|1007.11,
-0.89|	-0.02|	-0.45|		1.00|		400|	362|	17324|	-42|	16|	-14|	-35|	23|	-605|	25.29,|1007.12,
-0.90	|-0.02|	-0.44|		1.00|		390|	378|	17348|	-42|	17|	-13|	-33	|21	|-604|	25.29,|1007.19,
-0.90|	-0.02|	-0.44|		1.00|		356	|338|	17312|	-42|	17|	-13|	-35|	21|	-605|	25.29,|1007.19,
-0.88|	-0.02|	-0.47|		1.00|		346	|336	|17350|	-42	|17	|-13|	-37|	22	|-605|	25.29,|1007.14,

mjs513, 2015-03-28:
The answer to your question is yes.  It takes a minute or so to the MPU6050 to stabilize once you start it up.  If you read the stuff out there they all say you have to keep the imu steady during its startup period – this is pretty much everytime you rerun the sketch.  The algorithm also takes a little while to converge from their starting conditions.  

 

What I am curious about is the gravity vector itself why GZ isn’t larger than it is.  It looks like the board is titled.  Whaa

mjs513, 2015-03-28:
Here is what I am getting – by the way where you located (by any chance are you closer to the equator) – I am in NY:

 

0.00       -0.00     -0.00                    0.00                     624        78          16538   -39         8            -19         159        -258        -496      29.60,1009.74,

0.00       -0.00     -0.00                    0.00                     610        130        16506   -39         7            -16         160        -259        -497      29.60,1009.74,

0.00       -0.00     -0.00                    0.00                     618        110        16554   -40         7            -15         163        -259        -498      29.60,1009.73,

0.05       0.00       1.00                     1.00                     624        136        16540   -39         7            -15         161        -260        -497      29.60,1009.74,

0.05       0.00       1.00                     1.00                     600        128        16510   -40         8            -15         162        -256        -500      29.70,1009.73,

0.05       0.00       1.00                     1.00                     622        110        16510   -39         8            -15         162        -256        -499      29.70,1009.74,

0.05       0.01       1.00                     1.00                     602        124        16502   -39         8            -16         158        -256        -495      29.70,1009.77,

0.05       0.01       1.00                     1.00                     610        114        16540   -39         7            -16         159        -257        -496      29.70,1009.81,

0.05       0.01       1.00                     1.00                     618        86          16520   -39         7            -16         159        -256        -499      29.70,1009.77,

0.05       0.01       1.00                     1.00                     618        104        16518   -38         7            -16         157        -257        -496      29.70,1009.76,

0.05       0.01       1.00                     1.00                     602        70          16546   -39         7            -16         160        -259        -496      29.70,1009.84,

0.05       0.01       1.00                     1.00                     622        138        16538   -39         8            -15         159        -258        -499      29.70,1009.76,

0.05       0.01       1.00                     1.00                     622        106        16560   -39         7            -15         161        -260        -500      29.70,1009.78,

0.76       0.00       0.65                     1.00                     606        140        16546   -39         7            -16         158        -256        -498      29.70,1009.78,

0.76       0.00       0.65                     1.00                     592        114        16518   -39         7            -16         159        -255        -497      29.70,1009.84,

0.76       0.00       0.65                     1.00                     588        146        16538   -40         8            -15         163        -258        -494      29.70,1009.68,

0.76       0.00       0.64                     1.00                     586        136        16562   -40         8            -15         161        -256        -498      29.70,1009.77,

0.76       0.00       0.65                     1.00                     614        116        16504   -39         7            -15         159        -256        -498      29.70,1009.74,

0.77       0.00       0.64                     1.00                     612        88          16580   -39         8            -16         160        -259        -496      29.70,1009.78,

0.76       0.00       0.65                     1.00                     584        136        16548   -40         8            -15         163        -257        -497      29.70,1009.78,

0.76       0.00       0.65                     1.00                     588        166        16496   -40         7            -15         159        -260        -499      29.70,1009.76,

0.76       0.00       0.65                     1.00                     602        100        16510   -39         7            -16         160        -260        -498      29.70,1009.80,

0.76       0.00       0.65                     1.00                     626        122        16500   -40         7            -16         160        -256        -500      29.70,1009.81,

0.77       0.00       0.64                     1.00                     586        124        16530   -39         7            -16         158        -255        -498      29.70,1009.73,

0.77       0.00       0.64                     1.00                     618        108        16534   -39         7            -15         158        -256        -497      29.70,1009.83,

0.76       0.00       0.64                     1.00                     598        112        16600   -40         7            -16         164        -256        -494      29.70,1009.68,

0.75       0.00       0.66                     1.00                     630        142        16554   -39         7            -16         159        -256        -497      29.70,1009.74,

0.76       0.00       0.65                     1.00                     632        112        16548   -40         8            -16         161        -259        -499      29.70,1009.76,

[... cut ...]
0.76       0.00       0.64                     1.00                     616        100        16512   -40         7            -15         159        -254        -499      29.70,1009.81,

0.77       0.00       0.64                     1.00                     624        90          16530   -40         7            -16         159        -257        -494      29.70,1009.68,

0.76       0.00       0.65                     1.00                     616        124        16518   -39         8            -15         158        -254        -499      29.70,1009.74,

cleaf30, 2015-03-28:
No I'm in Vermont and I adjusted my declination angle.

mjs513, 2015-03-28:
OK just trying to see if the vector from x and z made sense. After I sent
you the I also rotated the imu to see what values I would get and the
vectors appeared correct for the rotation.

cleaf30, 2015-03-29:
So you didn't turn it from (0.05, 0.01, 1.00) to (0.76, 0.00, 0.65)? - meaning you are experiencing the same issue but once it gets to the new vector(0.76, 0.00, 0.65) it reacts appropriately? 

Also with DCM I've been experiencing G, the magnitude of the gravity vector, fluctuate from 1 to 2.

mjs513, 2015-03-29:
Exactly.  What is occurring is that the algorithm is converging from the starting values to the ones that are stabilized in the first few seconds as it expected the final vector is 0.76,0.00,0.65 .  After that once you start moving it reacts appropriately.

 

Was just curious about DCM to see what you would get.  Remember the angles are calculated based on the Direction Cosine Matrix and I am converting it to quaternions so the GUI will act appropriately.  Convergence on DCM would be different. Also the did not tune the gains to for the FreeIMU.  You would have to look up how to get the gravity vector direct from DCM to be accurate.

 

Mike

cleaf30, 2015-03-29:
Mine isn't stabilizing. I've tried putting resets in my code and the RESET_Q() function has no effect however RESET() will give me good data for a few lines. The only difference in those functions is resetting MPU6050. I had to change the address to my MPU6050 because I have a real time clock in my project that had the same address so I soldered ADO to VCC and commented/uncommented for the new address in FreeIMU.h and it seemed fine. I also changed the default address in the MPU60X0 library just in case. Could it be referring to the old address somewhere else possibly taking data from the RTC?

mjs513, 2015-03-29:
Don’t think so.  I am assuming you tested with RTC out of the loop and everything worked.  If not try it without the RTC.  Also make sure you stick with Marg = 0, 1 or 3.  Haven’t really played with DCM enough to be of any help.  

 

Look at the mag values with and without the RTC or disable the magnetometer as an experiment.  What RTC are you using?

cleaf30, 2015-03-29:
I tried it without the RTC and with magnetometer disabled neither had an effect. I'm using a DS1307 it's part of an adafruit data logging shield. I am also still getting the magnitude of the gravity vector to be around two with marg = 3. I though the getQ method would give me unit quaternion and after converting it to cartesian coordinates it would still be a unit vector. I'm currently seeing (1.73, -0.95, 0.04) which has a magnitude of 1.97

mjs513, 2015-03-29:
Not sure what the issue is.  Your original post showed that it was fairly stable after a few seconds which agreed with my tests.  Take a look at your accel, gyro and mag values compared to what you originally posted.  The only time you would see something like that occurring is if you moved the IMU quickly and then stopped moving.  You will see a some overshoot but it should settle down again to stable values.

cleaf30, 2015-03-29:
So would you suggest just doing a some math to scale the gravity vector to a unit vector. Here is some more data with marg = 3. where it is stable with a magnitude of 1 then I move it a bit and the magnitude drops to .17. 

gx|gy|gz|G|raw[0]|raw[1]|raw[2]|raw[3]|raw[4]|raw[5]|raw[6]|raw[7]|raw[8]|temp|press
-----|-------|------|-------|------|-----|---------|-------|----|-------|------|------|-------|-----|---------
0.27|-0.04|-0.96|1.00|816|646|17282|-46|18|-13|-138|-132|-522|28.45|1009.60
0.27|-0.04|-0.96|1.00|848|670|17238|-46|19|-12|-141|-132|-523|28.45|1009.59
0.28|-0.04|-0.96|1.00|842|660|17232|-46|18|-13|-140|-130|-520|28.45|1009.59
0.28|-0.04|-0.96|1.00|798|682|17274|-45|18|-13|-140|-134|-521|28.45|1009.57
0.27|-0.04|-0.96|1.00|790|678|17250|-46|18|-13|-140|-129|-522|28.44|1009.56
0.29|-0.04|-0.96|1.00|854|666|17290|-45|18|-13|-138|-130|-525|28.44|1009.61
0.28|-0.04|-0.96|1.00|878|672|17202|-45|17|-12|-139|-131|-523|28.45|1009.63
0.28|-0.04|-0.96|1.00|852|670|17286|-46|18|-12|-138|-129|-525|28.45|1009.66
0.26|-0.04|-0.97|1.00|828|664|17300|-45|19|-12|-137|-129|-521|28.45|1009.66
0.28|-0.04|-0.96|1.00|812|676|17188|-46|18|-12|-139|-130|-521|28.45|1009.63
0.26|-0.04|-0.96|1.00|844|694|17162|-46|19|-12|-139|-131|-525|28.45|1009.61
0.29|-0.05|-0.96|1.00|806|672|17252|-46|19|-13|-137|-131|-521|28.45|1009.61
0.29|-0.04|-0.96|1.00|834|674|17286|-46|19|-13|-138|-131|-522|28.45|1009.62
0.30|-0.04|-0.95|1.00|816|672|17252|-45|18|-13|-137|-129|-523|28.45|1009.58
0.29|-0.04|-0.96|1.00|886|688|17266|-46|18|-13|-139|-129|-519|28.45|1009.58
0.26|-0.04|-0.96|1.00|824|658|17216|-46|20|-12|-141|-131|-524|28.45|1009.63
0.29|-0.04|-0.96|1.00|850|662|17236|-46|18|-12|-140|-130|-524|28.45|1009.65
0.28|-0.04|-0.96|1.00|830|688|17264|-45|18|-12|-141|-129|-526|28.45|1009.62
0.26|-0.05|-0.96|1.00|842|660|17230|-46|18|-13|-140|-132|-524|28.45|1009.61
0.26|-0.04|-0.97|1.00|846|678|17244|-45|19|-13|-137|-129|-520|28.45|1009.55
0.30|-0.04|-0.95|1.00|826|652|17250|-46|18|-12|-138|-129|-522|28.45|1009.56
0.27|-0.05|-0.96|1.00|842|704|17240|-46|18|-13|-140|-129|-522|28.45|1009.61
0.27|-0.04|-0.96|1.00|866|664|17264|-46|18|-13|-137|-131|-522|28.45|1009.61
0.28|-0.04|-0.96|1.00|850|712|17252|-46|18|-12|-140|-129|-521|28.45|1009.62
0.28|-0.04|-0.96|1.00|842|686|17304|-45|19|-13|-139|-130|-526|28.46|1009.62
0.26|-0.04|-0.96|1.00|850|694|17226|-45|18|-12|-138|-129|-524|28.46|1009.64
0.28|-0.04|-0.96|1.00|880|678|17268|-46|18|-12|-139|-131|-516|28.45|1009.63
0.27|-0.04|-0.96|1.00|812|690|17230|-45|18|-12|-137|-132|-521|28.45|1009.61
0.29|-0.04|-0.96|1.00|804|660|17230|-46|19|-12|-137|-130|-522|28.45|1009.62
0.27|-0.04|-0.96|1.00|848|676|17270|-45|18|-13|-139|-133|-521|28.45|1009.57
0.29|-0.04|-0.96|1.00|820|696|17254|-45|19|-12|-139|-129|-525|28.46|1009.58
0.28|-0.04|-0.96|1.00|842|684|17230|-46|18|-12|-137|-131|-523|28.46|1009.66
0.26|-0.04|-0.96|1.00|832|686|17234|-45|19|-13|-142|-134|-521|28.46|1009.65
0.26|-0.04|-0.97|1.00|846|676|17236|-46|19|-13|-141|-130|-524|28.46|1009.61
0.27|-0.04|-0.96|1.00|884|658|17260|-45|19|-13|-139|-128|-523|28.45|1009.60
0.30|-0.04|-0.95|1.00|868|702|17278|-46|18|-12|-137|-131|-523|28.45|1009.59
0.28|-0.04|-0.96|1.00|838|686|17278|-45|19|-12|-138|-133|-523|28.46|1009.60
0.27|-0.04|-0.96|1.00|796|698|17284|-45|19|-12|-141|-129|-521|28.46|1009.59
0.28|-0.04|-0.96|1.00|848|686|17232|-46|19|-12|-139|-132|-518|28.45|1009.58
0.28|-0.04|-0.96|1.00|874|654|17246|-46|18|-12|-139|-133|-521|28.45|1009.59
0.28|-0.04|-0.96|1.00|840|686|17196|-45|19|-12|-137|-130|-524|28.45|1009.60
0.28|-0.04|-0.96|1.00|842|682|17270|-46|19|-12|-142|-133|-523|28.45|1009.56
0.30|-0.04|-0.95|1.00|826|670|17292|-46|19|-12|-136|-131|-527|28.45|1009.56
0.28|-0.04|-0.96|1.00|900|692|17262|-45|18|-12|-142|-132|-518|28.45|1009.58
0.29|-0.04|-0.96|1.00|884|686|17230|-46|19|-13|-135|-131|-523|28.46|1009.59
0.28|-0.04|-0.96|1.00|850|654|17216|-45|19|-13|-141|-129|-524|28.46|1009.57
0.28|-0.04|-0.96|1.00|844|668|17250|-45|18|-13|-140|-133|-521|28.46|1009.57
0.29|-0.04|-0.96|1.00|808|654|17230|-45|18|-12|-139|-133|-520|28.46|1009.58
0.28|-0.04|-0.96|1.00|860|662|17288|-46|18|-13|-139|-131|-523|28.46|1009.57
0.29|-0.04|-0.96|1.00|844|686|17232|-46|18|-12|-139|-132|-522|28.46|1009.63
0.42|-0.01|-0.91|1.00|-1756|1234|15202|98|472|-35|-131|-133|-524|28.46|1009.63
0.20|-0.05|-0.98|1.00|188|892|16518|-99|-147|5|-140|-131|-523|28.46|1009.61
0.04|-0.07|0.10|0.13|774|708|17152|-47|19|-42|-141|-130|-522|28.46|1009.62
0.26|0.05|1.09|1.13|1026|574|17402|-29|87|-121|-140|-133|-522|28.46|1009.61
0.26|0.05|1.09|1.13|1024|620|17376|-44|22|-16|-139|-131|-521|28.46|1009.61
0.26|0.05|1.10|1.13|932|632|17238|-46|16|-12|-142|-131|-525|28.46|1009.61
0.26|0.05|1.10|1.13|822|710|17230|-46|19|-12|-135|-132|-522|28.46|1009.61
0.26|0.05|1.10|1.13|870|710|17208|-46|19|-13|-139|-130|-522|28.46|1009.58
0.26|0.05|1.10|1.13|812|682|17258|-45|18|-12|-136|-134|-521|28.46|1009.59
0.26|0.05|1.10|1.13|772|666|17250|-46|19|-13|-141|-135|-519|28.46|1009.63
0.26|0.05|1.10|1.13|844|678|17218|-46|18|-13|-139|-133|-518|28.46|1009.63
0.26|0.05|1.10|1.13|826|636|17286|-46|19|-13|-139|-132|-517|28.46|1009.59
0.25|0.05|1.10|1.13|876|692|17212|-45|18|-12|-139|-131|-519|28.47|1009.60
0.25|0.05|1.10|1.13|862|692|17264|-45|19|-12|-142|-133|-525|28.47|1009.61
0.29|-0.05|-0.96|1.00|830|688|17226|-46|18|-13|-138|-133|-524|28.47|1009.62
0.27|-0.04|-0.96|1.00|804|660|17254|-46|19|-13|-139|-133|-522|28.47|1009.62
[... 88 lines cut ...]
[... cut ...]
0.05|-0.08|0.16|0.19|794|664|17286|-46|18|-12|-144|-118|-525|28.50|1009.62
0.05|-0.08|0.16|0.19|832|666|17250|-45|18|-13|-142|-120|-522|28.50|1009.63
0.05|-0.08|0.16|0.19|834|694|17274|-46|18|-12|-144|-122|-524|28.50|1009.59
0.05|-0.08|0.16|0.19|830|650|17240|-45|19|-12|-147|-118|-524|28.50|1009.59

It has been sitting for a while now and it stabilizes to a magnitude of 1 but the g vector is (0.54,-0.05,-0.84). I give it a 90 degree turn and to settles at (0.99,0.04,17). If I keep it the xy plane horizontal shouldn't the g vector be about (0,0,1)?

mjs513, 2015-03-29:
A couple of things. I am kind of curious that your accel and mag values are a bit different when at rest compared to your original post.  One of things you will read is that the there are variations in calibration values depending on temperature.  Also, I had to do the calibration a couple of times after first use.  So you might what to do another calibration. 

As for your question on gravity vector, not necessarily. Depends where you are on the earth,  Try printing ypr at the same time - you might see you are not perfectly level. Try leveling it out and see what you get for the gravity vector.

cleaf30, 2015-04-01:
You're right about the calibration. I've calibrated multiple times and have gotten consistent results however when I go back to the calibration.h file that comes with the FreeIMU library my IMU works much better. The main discrepancy is the z axis acceleration offset. ~700 in my calibration.h compared to ~1600 in the file that comes with the library. I've changed the z axis acc offset in my file and that works as well.

mjs513, 2015-04-01:
Hi Cody

 

Based on your data your z-axis scale should be around 17300 based on your raw data.  Check the GUI and see if you have word = 2.  Word =2 is for 16 bit processors like the mega.  Something funny is going with the cal file.  That is why you are probably getting strange values.

 

Mike

closed 2015-04-15 by mjs513