mjs513/FreeIMU-Updates #27: limited pitch and roll

nandomtl, 2015-04-06:
Hello,

I am testing "FreeIMU_yaw_pitch_roll.ino" with an MPU9250 on a Drotek board:
http://tinyurl.com/o85jeks
Arduino Uno R3 runs the sketch.
Yaw follows the board orientation quite well, but pitch and roll are limited to +-90 deg from the horizontal plane. In other words, the sketch does not see the board inverted. After +-90 degrees, if I continue the movement (roll or pitch), the reading starts moving back to zero.

Is there a parameter that controls this so roll and pitch are properly followed? If not, would you be able to suggest a spot where the code could be modified to do so? 

Thanks in advance

mjs513, 2015-04-07:
Actually it is behaving the exactly the way it should be for an aerospace oriented reference system.  If you want to change the range you would have to modify the ypr function.  Suggest that you check out this post from I2Cdevlib forum: http://www.i2cdevlib.com/forums/topic/24-roll-and-pitch-angles-ranges/ .  It might help you in figuring out how to change the function.  Let me know how it goes and if you send me a copy of the function I will see if I can incorporate it in the library.

Enjoy

nandomtl, 2015-04-07:
Thanks for the tip - the thread describes what I observe in my tests.

I'll try to modify the code in the next few days, probably the weekend. Will keep you posted.

I do have a question:
You say the code behaves as it should for an aerospace reference system. How can that be when, for instance, 60 deg pitch can mean one of two possibilities? There is no one-to-one correspondence...

Thanks for the tip, 'preciate it

nandomtl, 2015-04-09:
I was crawling through the FreeIMU.cpp code, and found this (starting at line 1573):

/**
 * Returns the yaw pitch and roll angles, respectively defined as the angles in degrees between
 * the Earth North and the IMU X axis (yaw), the Earth ground plane and the IMU X axis (pitch)
 * and the Earth ground plane and the IMU Y axis. 
 * 
 * Roll and Pitch angle ranges 0-360 degrees instead of 0-90 degrees
 * 
 * @note This is not an Euler representation: the rotations aren't consecutive rotations but only
 * angles from Earth and the IMU. For Euler representation Yaw, Pitch and Roll see FreeIMU::getEuler
 * 
 * @param ypr three floats array which will be populated by Yaw, Pitch and Roll angles in degrees
*/
void FreeIMU::getYawPitchRoll180(float * ypr)

I can't test this until the weekend, but it seems to do what I want... 
Comments?

mjs513, 2015-04-09:
To be honest I forgot that I added that function and uploaded it.  It was added for a comparison against a kalman filter implementation I was testing. When you rummage around the web you find that there are different axis ranges depending on the coordinate system. Had a good site that explained it been trying to find it to send to u.  Anyway give it a shot it should work

nandomtl, 2015-04-09:
Thanks - I'll test and let you know.

nandomtl, 2015-04-12:
Hello,

I tried to use the code in "getYawPitchRoll180" (FreeIMU March version), but the ranges it returns are a bit goofy...
Run 1 (826 samples)
Roll: -1.42 to 1.17
Pitch: -3.14 to +3.14
Yaw: -1.15 to +0.44

Run 2 (770 samples)
Roll: -1.55 to +1.2
Pitch: -3.14 to +3.14
Yaw: -1.96 to +2.04

Run 3 (782 samples)
Roll: -1.52 to +1.52
Pitch: -3.14 to +3.14
Yaw: -2.48 to +1.3

...so I decided to try the code from the forum you mentioned, as it seemed simple enough to adapt.
I had to add a translation so my instrumentation reacts properly:

Changed from:
Roll: 0 horizontal wings, 90 right wing pointing down, -90 right wing pointing up, +-180 inverted.
Pitch: 0 normal, 90 nose down, 180 horizontal inverted, 270 nose up.
...to:
Roll: normal is 0 deg, right wing down is 90 deg, right wing up is -90 deg  and inverted is +-179.9 deg
Pitch: horizontal (normal and inverted) is 0 deg, nose down is -90 deg, nose up +90 deg
___________________________________________
void FreeIMU::getYawPitchRoll180(float * ypr) {
	float q[4];				// quaternion
	float val[11];
	float gx, gy, gz;		// estimated gravity direction
	
	getQ(q, val);

	gx = 2 * (q[1]*q[3] - q[0]*q[2]);
	gy = 2 * (q[0]*q[1] + q[2]*q[3]);
	gz = q[0]*q[0] - q[1]*q[1] - q[2]*q[2] + q[3]*q[3];

//===================== the following code adapted from ====================
// http://www.i2cdevlib.com/forums/topic/24-roll-and-pitch-angles-ranges/
//==========================================================================

//---------- YAW ------------	  
  ypr[0] = atan2(2 * q[1] * q[2] - 2 * q[0] * q[3], 2 * q[0]*q[0] + 2 * q[1] * q[1] - 1);
//  Serial.println("-----*-----blip-----*-----");
//---------- PITCH -------------
    //ypr[1] = atan(gx / sqrt(gy*gy + gz*gz));
    if(gz>0) //range from 0 to 90 and from 0 to -90
            {
              ypr[1] = atan(gx / sqrt(gy*gy + gz*gz));
            }
    else if(gz<0) //to get range from 90 to 180 and ftom -90 to -180
                {
                  ypr[1] =  - (PI)-atan(gx / sqrt(gy*gy + gz*gz));
                } 
            
    if(ypr[1] > 0) //to get range from 0 to -360
         {
         ypr[1] = -2*PI + ypr[1];
         }

    ypr[1] = abs(ypr[1]);  
  
  
//---------- ROLL -------------  
  //ypr[2] = atan(gy / sqrt(gx*gx + gz*gz));

    if(gz>0) 
        {
         ypr[2] = atan(gy / sqrt(gx*gx + gz*gz));
        }
    else if(gz<0)
            {
              ypr[2] = -PI - atan(gy / sqrt(gx*gx + gz*gz));
            }
    if(ypr[2] > 0) 
         {
          ypr[2] = -2*PI + ypr[2];
        }
              
    ypr[2] = abs(ypr[2]);
    
//------------ convert to degrees -------------
  arr3_rad_to_deg(ypr);
 
//------------- translate pitch ------------
  if (ypr[1] < 90) 
      {
       ypr[1] = -ypr[1];
       }
       else
           if (ypr[1] < 270)
               {
               ypr[1] = ypr[1] - 180;
               }
               else ypr[1] = 360 - ypr[1];
//  ypr[1] -=180;                 //   pitch

//---------------- translate roll -------------
    if (ypr[2] < 180)
        {
        ypr[2] = -ypr[2];
        }
        else ypr[2] = 360 - ypr[2];
//  ypr[2] -=180;                 //   roll

}

___________________________________________
Things are working nicely now.

I uploaded a video (170 MB) to dropbox: 
https://www.dropbox.com/s/tdp95c0jvdmmsux/IMG_0274.MOV?dl=0

Thanks for the assistance.

mjs513, 2015-04-12:
Glad you got it working the way you want. By the way I checked out the
video and nice GUI.

Mike

nandomtl, 2015-04-13:
Thanks - the video will be my backup for an EAA presentation this month end.

I still do not know what data to use for the slip indicator. Any suggestions?

mjs513, 2015-04-13:
Check this video out.. https://m.youtube.com/watch?v=iK1kwSkREt4

The check out wiki on side slip for coordinated turns.

nandomtl, 2015-04-14:
Thanks much for the assistance.
I'm closing the issue now

Best

Carlos

closed 2015-04-14 by nandomtl