mjs513/FreeIMU-Updates #28: compile fails

nandomtl, 2015-04-10:
Hello,

I have been using the version published in February.
Today I downloaded the March version, and tried to compile FreeIMU_yaw_pitch_roll.ino
It failed because DCM.h could not be found (although the DCM folder shows in the library list).
To bypass the problem, I made MARG = 0 in the FreeIMU.h file.
Then I got a bunch or error msgs (below).
Question: Does FreeIMU requires a specific version of the Arduino IDE? 
I tried 1.6.1 and 1.6.3...
__________________________________

/Users/username/Documents/Arduino/libraries/FreeIMU/FreeIMU.cpp: In member function 'void FreeIMU::init()':
/Users/username/Documents/Arduino/libraries/FreeIMU/FreeIMU.cpp:457:7: error: 'FIMU_ACCGYRO_ADDR' was not declared in this scope
  init(FIMU_ACCGYRO_ADDR, false);
       ^
/Users/username/Documents/Arduino/libraries/FreeIMU/FreeIMU.cpp: In member function 'void FreeIMU::init(bool)':
/Users/username/Documents/Arduino/libraries/FreeIMU/FreeIMU.cpp:472:7: error: 'FIMU_ACCGYRO_ADDR' was not declared in this scope
  init(FIMU_ACCGYRO_ADDR, fastmode);
       ^
/Users/username/Documents/Arduino/libraries/FreeIMU/FreeIMU.cpp: In member function 'void FreeIMU::getValues(float*)':
/Users/username/Documents/Arduino/libraries/FreeIMU/FreeIMU.cpp:998:3: error: 'accgyro' was not declared in this scope
   accgyro.getMotion6(&accgyroval[0], &accgyroval[1], &accgyroval[2], 
   ^
/Users/username/Documents/Arduino/libraries/FreeIMU/FreeIMU.cpp:1070:14: error: 'sensor_sign' was not declared in this scope
  values[i] = sensor_sign[i] * values_cal[sensor_order[i]];
              ^
/Users/username/Documents/Arduino/libraries/FreeIMU/FreeIMU.cpp:1070:42: error: 'sensor_order' was not declared in this scope
  values[i] = sensor_sign[i] * values_cal[sensor_order[i]];
                                          ^
/Users/username/Documents/Arduino/libraries/FreeIMU/FreeIMU.cpp: In member function 'void FreeIMU::getQ(float*, float*)':
/Users/username/Documents/Arduino/libraries/FreeIMU/FreeIMU.cpp:1229:97: error: 'AHRSupdateIMU' was not declared in this scope
    AHRSupdateIMU(val[3] * M_PI/180, val[4] * M_PI/180, val[5] * M_PI/180, val[0], val[1], val[2]);
                                                                                                 ^
Multiple libraries were found for "I2Cdev.h"
 Used: /Users/username/Documents/Arduino/libraries/I2Cdev
 Not used: /Users/username/Documents/Arduino/libraries/MotionDriver
Error compiling.

mjs513, 2015-04-10:
Uses 1.5 or better. Based  on your errors looks like you need to update the
freeimu.h for the board you are using
I will see if I can duplicate your issue tomorrow. Time sleep

mjs513, 2015-04-10:
Hi nandomtl 
Now that I am a little more awake, delete the motiondriver directory.  You don't need anyway.  Since you put all the libraries in one spot there is a conflict because i2cdevlib is in two spots.  Also, in the YPR sketch there may be  a library conflict (only a guess for now) so delete all libraries you don't need (ie, the sensor libraries that are not on your IMU).  I will try both DCM and MARG=0 tonight just to be sure.

Mike

nandomtl, 2015-04-10:
Thanks, Mike

I'll test it tonight. Also, it's possible I forgot to un-comment the
#define 9250 statement (proving once again that there are more problems
between the screen and the chair than within the code...)

Cheers

nandomtl, 2015-04-10:
I removed the motiondriver directory and activated the #define 9250 statement, and it compiled without errors!

Thanks for the help!

closed 2015-04-10 by nandomtl