mjs513/FreeIMU-Updates #29: Sparkfun SEN- 10724 with Arduino Due - Can't read sensor data

yigityildirim, 2015-04-23:
Hello, 

2 months ago I opened an issue about Arduino Due Compatibility. I was getting strange behaviours from the processing GUI and you suggested to do the calibration first. Unfortunately I couldn't work on the project until now. I have Arduino Due and Sparkfun Sensor Stick SEN-10724. When I put "v" in the input field I get "FreeIMU library by Fabio Varesano - varesano.net, FREQ:84 MHz, LIB_VERSION: DEV, IMU: SparkFun 10724".When I tried to put "q" or "r" , I get nothing. But for example, when I enter "r" command twice, I can read the data from the sensor only for 2-3 seconds, then it stops. It's same for the "q" command, but when I continue to enter the command, I get "0000C07F,0000C07F,0000C07F,0000C07F,
". I tried the same sensor with Arduino UNO using the original FreeIMU library and I had no problem in reading the data from the serial monitor. The project has a deadline, so I would appreciate any help. Thank in advance.

mjs513, 2015-04-23:
Ok lets get to work.  For the issue with the const values I would first check that you have SDA/SCL hooked up correctly - remember on the Due its pins 20/21 as opposed to pins 4/5 on the Uno.  Second I would check that you only have the 10724 board uncommented and all other boards commented.  This is typically what I see when those either of those two things are incorrect.

As for the "r" and "q", its suppose to stop after about 30 reads.  The GUI continually sends out the read command after a certain burst count.

yigityildirim, 2015-04-23:
I think SDA/SCL connections are OK and I only have the 10724 board uncommented. I've just uploaded a video about the problem. In my opinion, something goes wrong other than calibration. 

https://www.youtube.com/watch?v=eB-_doSwIEQ

mjs513, 2015-04-23:
Thanks for the video it helped me figure out whats going on.  I hooked up my 10724 with the Due.  Issue is with Filter selection logic after I added the option for using a DCM filter.  If you want to check it out real quick set Marg = 4 in FreeIMU.h, it should work out of the box.  

I fixed the logic and uploaded the new files to Github.  You will need to replace your copies of FreeIMU.h and FreeIMU.cpp.  If you haven't downloaded it since incorporation of DCM you might what to replace the  all the files.

The only issue you will have is you may have to tweak the default setting for Kp, Ki, betadef, etc or set up you own code for a 10724 IMU.  I didn't set up default settings for the 10724 board as I don't really use it.  The default for DCM seem to work ok.

Thanks for letting me know about the issue.

yigityildirim, 2015-04-24:
Thanks a lot for your help. Unfortunately, I still have the same problem even when I set Marg = 4. Sorry for taking your time but I really don't understand what the problem is. By the way, I don't know which IMU is better than the other so it would be great if you suggest a different IMU that should work smoothly with the library.

mjs513, 2015-04-25:
Don't worry. I always found the 10724 a little strange to work with and if you read posts on the 10724 I am not the only one. My preference is the MPU6050 with the HMC5883 magnetometer. For pressure sensor I like the MS5611 I also like the mpu9250 as a second choice along with the Altimuv10.  So check out the GY88, drotek 10dof and pololu.

yigityildirim, 2015-04-26:
Thank you very much for the suggestions ! I'm going to purchase one of them as soon as possible to complete the project.

mjs513, 2015-04-26:
I am in the middle of something else right now but am going to continue to investigate the issue that you brought up with the 10724 IMU.  You got me curious.

yigityildirim, 2015-04-26:
Just one more quick question to make sure not to purchase the wrong sensor. Is this the one you suggested as your first choice : http://www.drotek.fr/shop/en/home/62-imu-10dof-mpu6050-hmc5883-ms5611.html
I also couldn't understand that If I use this one, which line should be uncommented in FreeIMU.h. Again, sorry for taking your time and thanks for your help.

mjs513, 2015-04-27:
Yes it is.  You would uncomment the following line: //#define FREEIMU_v04

yigityildirim, 2015-04-29:
Hi
I just got IMU 10dof from drotek. Unfortunately, I couldn't make it work. I tried both on Uno and Due using both your library and the original FreeIMU library of Varesano. I couldn't get any reading from the IMU. I'm pretty sure that the connections are solid and correct. I uncommented the line ://#define FREEIMU_v04 as you said. I couldn't find any reason of the problem other than the possibility of a broken IMU.

mjs513, 2015-04-30:
Well I will go back to basics. Make sure you only have the freeimu line
uncommented and all other boards commented. Make sure you have all
corrections correct. Doesn't hurt to double check.

Also check thus out to see if the i2c devices are recognized.
http://forum.arduino.cc/index.php?topic=197360

Try running example sketches for barkmter, imu and magazine separately.

yigityildirim, 2015-05-01:
By using i2c scanner, not sure but I found that the i2c address for Mpu-6050 is 0x69, not 0x68 by default. So I don't know if it's correct but I simply change the following lines in MPU60X0.h as : 
#define MPU60X0_ADDRESS_AD0_LOW     0x69
#define MPU60X0_ADDRESS_AD0_HIGH    0x70
Now I2c scanner founds 3 devices as 0x1E , 0x69 and 0x77.
Now, everything seems okay, I got fair results in GUI without calibration in Marg 3 settings. In Marg 4 yaw seems to work in reverse and in a strange way. Couple of things I notice are that the values freeze for a moment at the beginning of the GUI, cube doesn't seem to move fluently when I apply fast orientations and I found the response time a little bit high. Again, thank you very much for all of your help.

mjs513, 2015-05-01:
There is actually a option in FREEIMU.CPP to change the default 6050
address just for info.  I am assuming you are testing it on the Arduino
Due.  My guess is that you need to boost up the baud rate in the sketch to
115200 as well as in the processing sketch. You might see some
improvement.  The other thing is that you can change the burst constant so
that they Due is receiving more data for processing. I will have to look at
the DCM implementation to see the issue you are discussing.

yigityildirim, 2015-05-01:
Yes it is on the Due. I used 115200 baud rate and changed the burst constant but nothing changed. I found that it is nothing to do with how fast you rotate the IMU, but the fluency of the cube is interrupted by periodic momentary skipping. It's really obvious when you rotate the IMU fast. It's very fluent for a second, then it skips between angles just for a very little moment, then it's fluent again. It goes like that.

mjs513, 2015-05-02:
When I have time I will check it out.

closed 2015-05-27 by mjs513