mjs513/FreeIMU-Updates #30: Sketch too big

AndreaBruno, 2015-07-03:
Hi,

I've tried the FreeIMU_serial sketch on my IMUduino (Arduino Leonardo + freeIMU), but I got an error when compiling the code.
The Arduino IDE tell me that the sketch is too big, about 43.452 byte.
(I've already set my freeIMU version on FreeIMU.h file)

Thank you,

Andrea

mjs513, 2015-07-04:
The Leonardo only has 32k of seam you will need something with at least 64k which brings you to a Mega 2560, trendy 3.1 or equivalent.  The library is generic so by nature it has allot of extra code to cover the various imitation configurations.

mariocannistra, 2015-07-06:
Ciao Andrea, Mike.

@Mike: even if time allowed by my job/life is still reduced...I'm still around and continuing on my small project... I will have to catch up with a good number of changes/additions to the various sensors libs when I'll restart... :-)

@Andrea: I started my project using an arduino pro micro and found your same issue. When your sketch starts to grow with more sensors and/or a microSD to write to... the binary becomes too big for some micros unfortunately.  

I switched to Teensy 3.1 as Mike suggests and solved all my issues :-)   Very similar size, faster, you add a quartz and get also the RTC running.   With that I can use 9 axis inertial sensor (gyro, magn, accel) + baro,temp + gps + microsd . Just rebuilt the same sources with some minor changes.   

I still have to install the 1.6.x IDE and latest Visual Micro but I think that with those it will be even easier and possibly able to get a slightly smaller binary...

Enjoy the travel, it's funny and really engaging.

Bye,
Mario

dj-hedgehog, 2015-07-09:
I've got the very same problem using the ArduiMU v3+ so I will have to back to the original FreeIMU-version by Fabio Varesano.
I'd like to compare your version to it nonetheless. Are there some parts which we could easily eliminate?

AndreaBruno, 2015-07-13:
Ciao Mario, Mike.

@mjs513 Thank you for your reply.
I was looking for a library for the freeIMU (including IMUduino) that already include filter (like your Kalman implementation) because the data received from my IMUduino is really variable.

@mariocannistra Thank you also for the reply. 
Have you ever tried other library like this?
I've tried the [I2CDevLib](https://github.com/jrowberg/i2cdevlib) and this [Kalman Filter](https://github.com/TKJElectronics/KalmanFilter).

Also, as suggested by @dj-hedgehog, there are some part that we could easily eliminate to use this library on the classic atmega328 or atmega32u4?

Andrea

mjs513, 2015-07-13:
Ciao Andrea, Mario

All, sorry I did not reply sooner but been busy with a few other projects including work.  So anywhere here goes.

@AndreaBruno To try and reduce the size of the lib you would have to edit the FreeIMU.cpp and maybe the FreeIMU.h depending on what you deleted.  You could start by deleting all the code associated with pressure sensors that you are not using and go on to delete the code associated with IMU you are not using.  This may be a little more challenging.  In the Arduino serial sketch delete anything related to GPS.  You can also delete some of the output if you don't need it.

@mariocannistra Mario, I just uploaded two versions of the Kalman, one is the from the reference link you provide and the second is from Picopter.  Unfortunately the Picopter version is not compiling on the ARM boards yet - working on it now.  Currently using 1.6.1 but switching to 1.6.5.  Check the Readme and the video on youtube.

Mike

mariocannistra, 2015-07-14:
Ciao Mike, Andrea and dj-hedgehog

@Mike : i will certainly have a look at your new video and sw. Actually the kalman link has been provided by Andrea :-)  I'm also very interested in Kalman filters even if I moved that step of the processing off the board just because of my project requirements. More on this below in this answer.

@Andrea and @dj-hedgehog : the library you mentioned ( i2cdevlib ) is on my list of things to test. I've been following its development for months now and i think is a good one.  Anyway i see that fitting better others of my projects.  IMHO, FreeIMU is still the best one for certain project types.  I initially used Fabio's version then moved to this one.  I believe Mike's version is the only one around that is being kept updated and improved.

And now about Kalman filters: in my project I'm recording data from all sensors on a microSD (9 axis imu + baro + temp + gps + quaternions + time reference). I do this with the smallest possible time step in order to improve precision/reduce errors (this is one further motivation to use an higher clock mcu like the Teensy).   

When I finish the data logging sessions I download all the files from the microSD and post-process them with Kalman.

Initially I was doing that implementing Jacobians and it has been a long and difficult path with some result encouraging but hard to improve.  I continued my search on the topic and discovered that in those days Roger Labbe was updating his book (and Python lib) on github!  That has been a big discovery for me. We had some chats on the topic and he has been very kind answering my newbie questions :-)

His book (https://github.com/rlabbe/Kalman-and-Bayesian-Filters-in-Python) and sw (https://github.com/rlabbe/filterpy) are a must read/use on this topic.  The python lib he wrote abstracts you from the math difficulties of the traditional implementation approach ( use of Jacobians... ).  Have a look and see if there's a way for you to leverage it in your projects... :-)    I don't know if there's a way to run python on Teensy or similar MCUs but certainly you could give a try to Arietta G25.   I know.... I'm writing almost an heresy  :-)  Porting FreeIMU-Updates to Arietta could be a big job but... I2C is there just like SPI and UART, plus.... it runs Linux off a microSD and has python and other languages available.   Isn't this tempting? :-))))

I regret I did not share this before... unfortunately my time is shrinking instead of enlarging to fit all the topics i would like to follow... :-)

If you happen to have more time than please share your thoughts, trials, results. I'm always happy to chat about this stuff :-)

Ciao,
Mario

zrecommerce, 2015-08-26:
Hi everyone,

I am attempting to tackle this very ticket. I have a new branch named 'barebone' in my fork (femtoduino/FreeIMU-Updates)

The test sketch is available in femtoduino/imuduino-btle (Arduino/libraries/IMUduino/examples/IMUduino_Bluetooth_UART_YawPitchRoll_FreeIMU_Updates)

As of now, if i comment out the use of BLE in my sketch, and output to the Serial consol instead, the code now fits...

If I leave the BLE usage as is, my sketch is now at 108% in size (was 140% before I made.changes to the sketch, and FreeIMU-Updates libraries)

Note: the barebone branch of FreeIMU-Updates is meant to fit inside smaller AVRs. I am actively looking to shrink it another 8% or more.

dj-hedgehog, 2015-08-26:
@zrecommerce:
Thanks a lot. I'm really looking forward to try it once it's below 100%.
Please post here.

mjs513, 2015-08-26:
Hi Alex,

Glad you decided on taking on this issue.  Been looking at different options myself.  If you all are not hard over with using the MPU6050/HMC5883L combo you might want to take a look at the Adafruit LSM9DS0 IMU.  Doesn't have a pressure sensor but that should not be problem.  If you compile the YPR sketch (without) any changes to the lib this is what you get:

Sketch uses 24,138 bytes (74%) of program storage space. Maximum is 32,256 bytes.
Global variables use 1,670 bytes (81%) of dynamic memory, leaving 378 bytes for local variables. Maximum is 2,048 bytes.

What I was looking at was tackling the 6050 itself and the I2Cdevlib, i.e, get rid of everything that we don't really need but that's a lot bigger job than the approach you are using.  

Haven't gone through all your changes yet.  When I get some extra time I will do a fork and try and contribute.

Mike

mjs513, 2015-08-26:
PS Alex - just found your Tindie site - let me know when you get the Femduino-USB back in.  Think I will pick one up and give it shot and add it to my collection. Cheers.

zrecommerce, 2015-08-26:
I'm hoping to swap out the Femtoduino w/ USB for the ARM Cortex M0+ based
'Femto USB'

...fwiw, the dime sized FemtoBeacon (open source design) has an MPU-9250,
and MS561101-BA03.

I am splitting my time between order fulfillment for IMUduinos, and library
development. I want to add support for the SAM D21 and SAM R21 line of
Atmel chips to the main FreeIMU-Updates branch as soon as I can
stabilize/complete the barebone branch.

- Alex

mjs513, 2015-08-26:
Thanks for the infor. I saw the Femto USB but the only draw back to me was the lack of integration with the Arduino IDE - its just easier on me and laptop - don't have to install anything else.  Glad you are using the 9250 with the 5611, much more stable than the 9150.  I am impressed though with lsm9ds0, which was surprising to me.

Mike

zrecommerce, 2015-08-26:
The FemtoUSB is just an 'Arduino Zero' board. Instead of the Arduino Zero
bootloader (which expects the 32KHz crystal to tick before starting up), we
use the SAMBA bootloader, which works nicely with the Arduino IDE.

One of the things on my TODO list is to make a board package for it so
folks can use Arduino w/ it immediately.

:-)

Alex Albino
Sr. Fullstack Developer
Zend Certified Engineer
https://github.com/zrecommerce
https://github.com/femtoduino
https://github.com/femtoio

(707) 246-6864

mjs513, 2015-08-27:
Alex

I downloaded your example sketch and ran it after commenting out the motion detect routines in the MPU60X0 library with no other changes. Not sure what board you are using.  Using the Nano I get the following:

> Sketch uses 37,300 bytes (121%) of program storage space. Maximum is 30,720 bytes.
Global variables use 2,774 bytes (135%) of dynamic memory, leaving -726 bytes for local variables. Maximum is 2,048 bytes.

With the Uno:

>Sketch uses 37,300 bytes (115%) of program storage space. Maximum is 32,256 bytes.
Global variables use 2,774 bytes (135%) of dynamic memory, leaving -726 bytes for local variables. Maximum is 2,048 bytes

You might be able to squeeze a few more bytes by commenting those lines out.

Mike

zrecommerce, 2015-08-27:
I have my Arduino/library/ folder symlinked to a folder containing the
FreeIMU-Updates 'barebone' branch, and the Adafruit nRF8001 library, along
with a checkout of the IMUduino library (excluding the legacy FreeIMU libs)

I am using an IMUduino, which is just an Leonardo + FreeIMU + nRF8001

(Using Arduino 1.6.5)

- Alex

mjs513, 2015-08-27:
Hi All

Pretty much deleted almost everything from the FreeIMU lib and the best I could do was the following:
>Sketch uses 30,572 bytes (106%) of program storage space. Maximum is 28,672 bytes.
Global variables use 2,215 bytes (86%) of dynamic memory, leaving 345 bytes for local variables. Maximum is 2,560 bytes.

The Adafruit library pretty eats up about another 26% of storage space.  Maybe you can skinny the lib down or use another one.

Mike

mjs513, 2015-08-27:
Heres the link to the moded files:

https://drive.google.com/folderview?id=0BwzZjH9KYYMDfjJDc2Q3M29KYnlSdm4xeXhJTWZoVmxNLW00S04xUzdKQmhqTzN5azllWkE&usp=sharing

Cheers

AndreaBruno, 2015-08-28:
Hi All,

I've an IMUduino.
I've tried using different library and using [this kalman filter library](https://github.com/TKJElectronics/KalmanFilter) for the mpu, the code fit as well.

>Sketch uses 24,990 bytes (97%) of program storage space. Maximum is 28,672 bytes. Global variables use 1,818 bytes (71%) of dynamic memory, leaving 742 bytes for local variables. Maximum is 2,560 bytes.

mjs513, 2015-08-28:
Hi Andrea

I know that version of Kalman well.  I have included as an example with the FreeIMU lib.  I am able to get the lib to fit on a smaller board based on the what Alex and I worked on - know what to delete now.  There are a couple of things that would drive you over the limit from what I am seeing - incorporating the BLE library as well as incorporating the MS5611 lib for pressure and temperature. This is based on my testing and making changes to the lib to fit on the smaller boards. It looks like the nrf8001 adds about 6k and the 5611 4k to the sketch size. 

Also check out this version - may fit as well but again no ble or 5611 support:
 https://github.com/kriswiner/MPU6050HMC5883AHRS.git

regards
Mike

closed 2015-10-03 by mjs513