mjs513/FreeIMU-Updates #31: Various magnetometer issues

q009, 2015-07-22:
Hello.
I'm aware that this propably doesn't belong here but this is the only place I could seek help on that I know of...

I'm using a GY-86 sensor board, which consists of an MPU-6050 IMU and HMC5883L magnetometer.
I have calibrated the device. Now, it appears whenever I include the magnetometer in sensor fusion algorithms I get all kinds of odd behavior.
When I turn the magnetometer off everything works flawlessly.

When using MARG = 3 I'm getting strange drifts and flickering:
https://www.youtube.com/watch?v=lyzTcUa1fDI

When using MARG = 1 I'm getting even weirder behavior, difficult to describe:
I suggest you watch the whole video.
https://www.youtube.com/watch?v=S1MG4IdV9qo

This is my calibration:
http://i.imgur.com/GlYTFsX.png
I did the calibration twice:
Once for the accelerometer on a robot arm for best results
and second time for the magnetometer, done manually (I was unable to get the axes right otherwise).
After that I combined the results into a single file.
The calibration shows the second part - the magnetometer.

If anybody knows what could be wrong I'd appreciate any sort of help.
Thank you

q009, 2015-07-23:
Found the source of my problem

#if IS_9DOM() && not defined(DISABLE_MAGN)
	if(val[11] - motiondetect_old < 0) {  
		getQ_simple(q, val);
	}
#endif

Commenting this out fixes everything

mjs513, 2015-08-16:
Hi q009

Sorry for not getting back to you sooner but been busy at work and frankly had me stumped too. By the way I commented that line out of the cpp file as well.  Always meant to do it since it was an experiment in odometry.

Mike

closed 2015-08-16 by mjs513