mjs513/FreeIMU-Updates #33: Teensy    #include <EEPROM.h>

grezco, 2015-08-17:
Hello Mike,

This CPU definition, on FreeIMU serial (for example) seems not to be full because EEPROM.h is necessary for Teensy 3.1.

#if defined(__AVR__)
  #include <EEPROM.h>
#endif

mjs513, 2015-08-24:
Hi grezco

I will have to do a little more research on this one.  Originally that line was limited because of the Due and Galileo running under early versions of the Arduino IDE.  Under the newest version of the IDE I think the Teensy is using the AVR gcc chain for compile as opposed to the SAM chain.  I will let you know what happens when I get a chance to hit this issue.  Working on a couple of other things right now.  

Thanks for pointing this out, I completely forgot about it.  Also the code that Fabio included was centered around the Arduino Uno - never tried it out - I just load from the calibration file.

Mike

AMSDesha, 2015-11-16:
Hi 
i am working with MPU 9150 and 9250 raw data can be obtained. but quarternion values are wrong with the existance sample ? did any one face such a issue ?

mjs513, 2015-11-16:
Not once configured properly. Would suggest that you make sure you have the
correct board selected in freeimu.h.  akso make sure u did the calibration.

Mike

AMSDesha, 2015-11-19:
Thank you for replying
Actually it is not a configuration problem." FreeIMU_raw" example is working fine but "FreeIMU_quaternion" example is not working well. First component of quaternion is not changing with the time middle two are working fine and the last component is again wrong.   

i checked this with some other PCs and also checked for all the filters.
It is look like some thing is wrong with code

mjs513, 2015-11-19:
When I get I chance I will check it out but so far no one has reported this as a problem.

AMSDesha, 2015-12-04:
Did you get a chance to see the problem i mentioned above ?

dethegeek, 2016-01-15:
Hi 

Having the same issue about EEPROM.h

Found this post with a working solution for me :+1: 

https://forum.pjrc.com/threads/24420-Teensy-3-0-and-Adafruit-TFT-Library-Compile-Errors?p=36779&viewfull=1#post36779

mjs513, 2016-01-16:
Hi dethegeek - just checked the eeprom page at PRJC and looks like you don't even need to check for the eeprom.  However, with that said will need to change code to update max size of eeprom in the ino.  In the middle of a robotics project right now.  will let you know when i commit the changes.  It also appears my teensy 3.1 has died so....

By the way thanks for the link

dethegeek, 2016-01-16:
I agree : the page says EEPROM.h is provided by Arduino IDE. 

However, I don't know if the examples are intended to be compiled for platforms unknown to me, some with EEPROM and other without. That's why I kept the conditional inclusion of this library.

mjs513, 2016-01-16:
I will be posting updates to all sketches in the next couple of days the removal of the AVR test for eeprom.

Mike

mjs513, 2016-01-20:
Just updated sketches for fixes to make eeprom available for boards with eeprom capability - address AVE and Intel boards.

Please post with a new issue if there is a problem.

Thanks
Mile

closed 2016-01-20 by mjs513