mjs513/FreeIMU-Updates #35: Is the library based on Fabio Varesano's FreeIMU?

rei-vilo, 2015-08-30:
Is the library based on Fabio Varesano's FreeIMU?

If so, it would be nice to mention his name, as the library was developed as his MoS thesis **Using Arduino for Tangible Human Computer Interaction**. 

* http://www.varesano.net/projects/hardware/FreeIMU#library
* http://www.varesano.net/files/FreeIMU-20121122_1126.zip
* http://www.varesano.net/files/MoS_thesis/thesis.pdf worth reading!

As you may know, Unfortunately, Fabio died in Dec 23, 2012 http://www.varesano.net/blog/fabio/ciao-fabio.

barzilaydn, 2015-08-30:
It is, and he is credited - see on top of FreeIMU.h:
`Copyright (C) 2011 Fabio Varesano <fabio at varesano dot net>`
[https://github.com/mjs513/FreeIMU-Updates/blob/master/libraries/FreeIMU/FreeIMU.h]

rei-vilo, 2015-08-30:
Thanks! I mean in the [ReadMe.md](https://github.com/mjs513/FreeIMU-Updates/blob/master/README.md) file.

closed 2015-08-30 by rei-vilo

mjs513, 2015-08-30:
Fabio's original headers are retained in all files that he created for the FreeIMU library including the examples and the processing even though major changes were made.  In addition the Introduction page of the Wiki fully documents this fact.  The ReadMe.md file was meant as a change log of the changes that were incorporated into Fabio's original library.  I started out using the FreeIMU lib just before Fabio left us and I could not bear to see his work just stagnate so I stuck with it and made some rather major modifications.