mjs513/FreeIMU-Updates #36: Problems with IMUduino

GalDalal, 2015-09-27:
Hi,
I'm working with the IMUDUINO BTLE, using the FreeIMU library from here : 
https://github.com/femtoduino/imuduino-btle
and I have two issues regarding the board:

1. Yaw, pitch and roll- each of them works well in separate, and the values are stable and reasonable. But, while looking at all of them together, the yaw is affected from the pitch and the roll in an unexpected way (In that case, the data we get is not reliable).

2. Data stabilizing duration- it takes too long for the data to stabilize and get ready for sampling. The stabilizing duration is also inconsistent.

Do you have any recommendations regarding which methods we should use in order to solve those issues?Or any other solution to these problems
The methods in which we use now at the setup() are:
init (true), zeroGyro, calLoad 
(we are working with the most updated libraries). 
I'd love to For help,
Gal.

mjs513, 2015-09-29:
Good Morning Gal

Sorry for the delay in getting back to you but lets see if I can help.

1a.  Usually what I tell people is first make sure you have the correct board configuration identified in the FreeIMU.h file - you can use the Freeimu v04 board in this case since the hardware is the same.
1b.  Make sure you have gone through the calibration and saved it back to the Freeimu directory.  Can not tell you how important this is.
1c.  Once you have checked 1a and 1b.  Try disabling the magnetometer in the Freeimu.h file.
1d.  There is a coupling when your pitch gets too high - yaw direction will flip.  I am not sure what you mean by unexpected.

2.  Stabilization is dependent on a few factors.  It is based on the algorithm you select in the freeimu.h,  - madgwick, dcm etc.  These are in turn tuned by the Kp, Ki etc factors in the freeimu.cpp file.  I set some up for the boards I have but you may need to tweak it for your board.  With the MPU-6050 you need to make sure when you first turn it on to keep it steady and level and let the program work, i.e. , go  through its initializations.  Also, are you using a custom sketch? if you are have you tested it with one of the canned examples?

Hope this helps.
Mike

GalDalal, 2015-09-30:
Hello Mike, and thanks for the help.
like i said i'm using the FreeIMU library **from here** : 
https://github.com/femtoduino/imuduino-btle
and not  from : https://github.com/mjs513/FreeIMU-Updates.
Therefore Section 1c and 1(Stabilization) do not exist in libraries we work with.
So I can not try out your recommendations on the subject.

* Section 1a is ok, and i'm using Freeimu v04 board configuration.
* Section 1b I did not know what I should do, and what it means "gone through the calibration and saved it back to the Freeimu directory".
* Setion 1d, I understand it would be coupling when the pitch gets too high, But what happens is that when I change **only!** the Roll or Pitch, the Yaw is also affected, and his values are changing and not remain constant as it should be.

Thanks,
Gal.

mjs513, 2015-09-30:
Hi Gal

Not sure what I can tell you but based on another issue that I left open the FreeIMU version that is in the femtoduino library is a skinny downed version of the FreeIMU library so I would have to do some tests to see what the impact is. Not sure when I could get to this, busy at work right now. But I will take a look.

But anyway back to the calibration.  In the FreeIMU directory that you downloaded is a calibration.h file.  This holds the adjustments for the magnetometer and accelerometer (scale and offset factors).  Each board should be calibrated before you start.  You would run the calibration GUI (which is not part of the lib that I can see) but in the FreeIMU directory, this is a python program so you would need it installed.  Check out the Wiki associated with this library there is a page on how to do the calibration.

Mike

GalDalal, 2015-10-06:
Hi mike,
I wanted to know if you have the most updated file of calibration.h
Such is compatible with the new hardware card Imuduino-Btle from here:
http://femto.io/products/imuduino
Thanks.

mjs513, 2015-10-07:
As I said before you need to do your own calibration for the board you have.  To get you close however, you can try using the calibration-freeimu2h. or calibration-apm.h

You just need to delete the existing calibration.h and rename one of the ones above.  Again you need really need to do your own calibration otherwise you will start seeing drifting in the YPR.

closed 2015-10-20 by mjs513