mjs513/FreeIMU-Updates #37: Compile error

Zieju, 2015-09-30:
Hi
I have a problem with compiling the examples included in FreeIMU libraries. I've copied libraries to the right folder of course. Could you help me?

Opcje projektu zmienione, przeładuj całość
Użycie biblioteki adxl345driver z folderu C:\Users\Hp\Documents\Arduino\libraries\adxl345driver (legacy)
Użycie biblioteki HMC58X3 z folderu C:\Users\Hp\Documents\Arduino\libraries\HMC58X3 (legacy)
Użycie biblioteki LSM303 z folderu C:\Users\Hp\Documents\Arduino\libraries\LSM303 (legacy)
Użycie biblioteki itg3200filv05 z folderu C:\Users\Hp\Documents\Arduino\libraries\itg3200filv05 (legacy)
Użycie biblioteki bma180 z folderu C:\Users\Hp\Documents\Arduino\libraries\bma180 (legacy)
Użycie biblioteki MS561101BA z folderu C:\Users\Hp\Documents\Arduino\libraries\MS561101BA (legacy)
Użycie biblioteki bmp085driver z folderu C:\Users\Hp\Documents\Arduino\libraries\bmp085driver (legacy)
Użycie biblioteki I2Cdev z folderu C:\Users\Hp\Documents\Arduino\libraries\I2Cdev (legacy)
Użycie biblioteki MPU60X0 z folderu C:\Users\Hp\Documents\Arduino\libraries\MPU60X0 (legacy)
Użycie biblioteki AK8975 z folderu C:\Users\Hp\Documents\Arduino\libraries\AK8975 (legacy)
Użycie biblioteki AK8963 z folderu C:\Users\Hp\Documents\Arduino\libraries\AK8963 (legacy)
Użycie biblioteki L3G z folderu C:\Users\Hp\Documents\Arduino\libraries\L3G (legacy)
Użycie biblioteki LPS331 z folderu C:\Users\Hp\Documents\Arduino\libraries\LPS331 (legacy)
Użycie biblioteki SparkFun LSM9DS0 Breakout z folderu C:\Users\Hp\Documents\Arduino\libraries\LSM9DS0 
Użycie biblioteki MS5637-1 z folderu C:\Users\Hp\Documents\Arduino\libraries\MS5637-1 (legacy)
Użycie biblioteki AP_Math_freeimu z folderu C:\Users\Hp\Documents\Arduino\libraries\AP_Math_freeimu (legacy)
Użycie biblioteki AP_Filter z folderu C:\Users\Hp\Documents\Arduino\libraries\AP_Filter (legacy)
Użycie biblioteki iCompass z folderu C:\Users\Hp\Documents\Arduino\libraries\iCompass (legacy)
Użycie biblioteki Wire z folderu C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire 
Użycie biblioteki SPI z folderu C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 
Użycie biblioteki EEPROM z folderu C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM 
Użycie biblioteki DebugUtils z folderu C:\Users\Hp\Documents\Arduino\libraries\DebugUtils (legacy)
Użycie biblioteki FreeIMU z folderu C:\Users\Hp\Documents\Arduino\libraries\FreeIMU (legacy)
Użycie biblioteki DCM z folderu C:\Users\Hp\Documents\Arduino\libraries\DCM (legacy)
Użycie biblioteki Kalman z folderu C:\Users\Hp\Documents\Arduino\libraries\Kalman (legacy)

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_ADK -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega -IC:\Users\Hp\Documents\Arduino\libraries\adxl345driver -IC:\Users\Hp\Documents\Arduino\libraries\HMC58X3 -IC:\Users\Hp\Documents\Arduino\libraries\LSM303 -IC:\Users\Hp\Documents\Arduino\libraries\itg3200filv05 -IC:\Users\Hp\Documents\Arduino\libraries\bma180 -IC:\Users\Hp\Documents\Arduino\libraries\MS561101BA -IC:\Users\Hp\Documents\Arduino\libraries\bmp085driver -IC:\Users\Hp\Documents\Arduino\libraries\I2Cdev -IC:\Users\Hp\Documents\Arduino\libraries\MPU60X0 -IC:\Users\Hp\Documents\Arduino\libraries\AK8975 -IC:\Users\Hp\Documents\Arduino\libraries\AK8963 -IC:\Users\Hp\Documents\Arduino\libraries\L3G -IC:\Users\Hp\Documents\Arduino\libraries\LPS331 -IC:\Users\Hp\Documents\Arduino\libraries\LSM9DS0\src -IC:\Users\Hp\Documents\Arduino\libraries\MS5637-1 -IC:\Users\Hp\Documents\Arduino\libraries\AP_Math_freeimu -IC:\Users\Hp\Documents\Arduino\libraries\AP_Filter -IC:\Users\Hp\Documents\Arduino\libraries\iCompass -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM -IC:\Users\Hp\Documents\Arduino\libraries\DebugUtils -IC:\Users\Hp\Documents\Arduino\libraries\FreeIMU -IC:\Users\Hp\Documents\Arduino\libraries\DCM -IC:\Users\Hp\Documents\Arduino\libraries\Kalman C:\Users\Hp\AppData\Local\Temp\build5007050788206217377.tmp\FreeIMU_yaw_pitch_roll.cpp -o C:\Users\Hp\AppData\Local\Temp\build5007050788206217377.tmp\FreeIMU_yaw_pitch_roll.cpp.o 
In file included from FreeIMU_yaw_pitch_roll.ino:9:0:
C:\Users\Hp\Documents\Arduino\libraries\itg3200filv05/ITG3200.h:46:28: error: expected identifier before numeric constant
 #define WHO_AM_I           0x00  // RW   SETUP: I2C address   
                            ^
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:20:8: note: in expansion of macro 'WHO_AM_I'
        WHO_AM_I       = 0x0F,
        ^
C:\Users\Hp\Documents\Arduino\libraries\itg3200filv05/ITG3200.h:46:28: error: expected '}' before numeric constant
 #define WHO_AM_I           0x00  // RW   SETUP: I2C address   
                            ^
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:20:8: note: in expansion of macro 'WHO_AM_I'
        WHO_AM_I       = 0x0F,
        ^
C:\Users\Hp\Documents\Arduino\libraries\itg3200filv05/ITG3200.h:46:28: error: expected unqualified-id before numeric constant
 #define WHO_AM_I           0x00  // RW   SETUP: I2C address   
                            ^
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:20:8: note: in expansion of macro 'WHO_AM_I'
        WHO_AM_I       = 0x0F,
        ^
In file included from FreeIMU_yaw_pitch_roll.ino:17:0:
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:71:5: error: 'vector' does not name a type
     vector<int16_t> g; // gyro angular velocity readings
     ^
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:75:9: error: expected unqualified-id before 'void'
     L3G(void);
         ^
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:75:9: error: expected ')' before 'void'
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:77:15: error: 'bool init' redeclared as different kind of symbol
     bool init(deviceType device = device_auto, sa0State sa0 = sa0_auto);
               ^
In file included from C:\Users\Hp\Documents\Arduino\libraries\adxl345driver/ADXL345.h:15:0,
                 from FreeIMU_yaw_pitch_roll.ino:6:
C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:120:6: error: previous declaration of 'void init()'
 void init(void);
      ^
In file included from FreeIMU_yaw_pitch_roll.ino:17:0:
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:77:15: error: 'deviceType' was not declared in this scope
     bool init(deviceType device = device_auto, sa0State sa0 = sa0_auto);
               ^
[... 16 lines cut ...]
                                                                       ^
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:93:71: error: ISO C++ forbids declaration of 'parameter' with no type [-fpermissive]
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:93:77: error: expected ',' or '...' before '<' token
     template <typename Ta, typename Tb> static float vector_dot(const vector<Ta> *a, const vector<Tb> *b);
                                                                             ^
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:94:34: error: variable or field 'vector_normalize' declared void
     static void vector_normalize(vector<float> *a);
                                  ^
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:94:34: error: 'vector' was not declared in this scope
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:94:41: error: expected primary-expression before 'float'
     static void vector_normalize(vector<float> *a);
                                         ^
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:96:3: error: expected unqualified-id before 'private'
   private:
   ^
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:103:33: error: 'regAddr' has not been declared
       int testReg(byte address, regAddr reg);
                                 ^
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:104:1: error: expected declaration before '}' token
 };
 ^
Multiple libraries were found for "I2Cdev.h"
 Used: C:\Users\Hp\Documents\Arduino\libraries\I2Cdev
 Not used: C:\Users\Hp\Documents\Arduino\libraries\MotionDriver
Błąd kompilacji.

mjs513, 2015-10-03:
Hi Blad

Sorry for the delay in response.  First try and delete the MotionDriver directory - you don't really need it. Give it another try.  Also, make sure you have the correct board selected in the freeimu.h file.  They wiki should be of help.

Mike

Zieju, 2015-10-05:
Hi
Thank you very much for your response. I deleted Motion Driver directory and uncommented Pololu Altimu v10 like on attached screenshot (I'm not sure if I done it properly). The problem still occurs. It seems it is due some error in LG3 library. In downloaded zip there it's not present (beside of "l3g-arduino" empty directory), so i downloaded it from pololu site.
Thanks in advance and sorry for my terrible english and noob questions :) Regards Kamil

[image]

Using library adxl345driver in folder: C:\Users\Hp\Documents\Arduino\libraries\adxl345driver (legacy)
Using library HMC58X3 in folder: C:\Users\Hp\Documents\Arduino\libraries\HMC58X3 (legacy)
Using library LSM303 in folder: C:\Users\Hp\Documents\Arduino\libraries\LSM303 (legacy)
Using library itg3200filv05 in folder: C:\Users\Hp\Documents\Arduino\libraries\itg3200filv05 (legacy)
Using library bma180 in folder: C:\Users\Hp\Documents\Arduino\libraries\bma180 (legacy)
Using library MS561101BA in folder: C:\Users\Hp\Documents\Arduino\libraries\MS561101BA (legacy)
Using library bmp085driver in folder: C:\Users\Hp\Documents\Arduino\libraries\bmp085driver (legacy)
Using library I2Cdev in folder: C:\Users\Hp\Documents\Arduino\libraries\I2Cdev (legacy)
Using library MPU60X0 in folder: C:\Users\Hp\Documents\Arduino\libraries\MPU60X0 (legacy)
Using library AK8975 in folder: C:\Users\Hp\Documents\Arduino\libraries\AK8975 (legacy)
Using library AK8963 in folder: C:\Users\Hp\Documents\Arduino\libraries\AK8963 (legacy)
Using library L3G in folder: C:\Users\Hp\Documents\Arduino\libraries\L3G (legacy)
Using library LPS331 in folder: C:\Users\Hp\Documents\Arduino\libraries\LPS331 (legacy)
Using library SparkFun LSM9DS0 Breakout in folder: C:\Users\Hp\Documents\Arduino\libraries\LSM9DS0 
Using library MS5637-1 in folder: C:\Users\Hp\Documents\Arduino\libraries\MS5637-1 (legacy)
Using library AP_Math_freeimu in folder: C:\Users\Hp\Documents\Arduino\libraries\AP_Math_freeimu (legacy)
Using library AP_Filter in folder: C:\Users\Hp\Documents\Arduino\libraries\AP_Filter (legacy)
Using library iCompass in folder: C:\Users\Hp\Documents\Arduino\libraries\iCompass (legacy)
Using library Wire in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire 
Using library SPI in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 
Using library EEPROM in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM 
Using library DebugUtils in folder: C:\Users\Hp\Documents\Arduino\libraries\DebugUtils (legacy)
Using library FreeIMU in folder: C:\Users\Hp\Documents\Arduino\libraries\FreeIMU (legacy)
Using library DCM in folder: C:\Users\Hp\Documents\Arduino\libraries\DCM (legacy)
Using library Kalman in folder: C:\Users\Hp\Documents\Arduino\libraries\Kalman (legacy)
Using library AltSoftSerial in folder: C:\Users\Hp\Documents\Arduino\libraries\AltSoftSerial (legacy)
Using library TinyGPSPlus in folder: C:\Users\Hp\Documents\Arduino\libraries\TinyGPSPlus (legacy)

C:\Program Files (x86)\Arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10707 -DARDUINO_AVR_ADK -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega -IC:\Users\Hp\Documents\Arduino\libraries\adxl345driver -IC:\Users\Hp\Documents\Arduino\libraries\HMC58X3 -IC:\Users\Hp\Documents\Arduino\libraries\LSM303 -IC:\Users\Hp\Documents\Arduino\libraries\itg3200filv05 -IC:\Users\Hp\Documents\Arduino\libraries\bma180 -IC:\Users\Hp\Documents\Arduino\libraries\MS561101BA -IC:\Users\Hp\Documents\Arduino\libraries\bmp085driver -IC:\Users\Hp\Documents\Arduino\libraries\I2Cdev -IC:\Users\Hp\Documents\Arduino\libraries\MPU60X0 -IC:\Users\Hp\Documents\Arduino\libraries\AK8975 -IC:\Users\Hp\Documents\Arduino\libraries\AK8963 -IC:\Users\Hp\Documents\Arduino\libraries\L3G -IC:\Users\Hp\Documents\Arduino\libraries\LPS331 -IC:\Users\Hp\Documents\Arduino\libraries\LSM9DS0\src -IC:\Users\Hp\Documents\Arduino\libraries\MS5637-1 -IC:\Users\Hp\Documents\Arduino\libraries\AP_Math_freeimu -IC:\Users\Hp\Documents\Arduino\libraries\AP_Filter -IC:\Users\Hp\Documents\Arduino\libraries\iCompass -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM -IC:\Users\Hp\Documents\Arduino\libraries\DebugUtils -IC:\Users\Hp\Documents\Arduino\libraries\FreeIMU -IC:\Users\Hp\Documents\Arduino\libraries\DCM -IC:\Users\Hp\Documents\Arduino\libraries\Kalman -IC:\Users\Hp\Documents\Arduino\libraries\AltSoftSerial -IC:\Users\Hp\Documents\Arduino\libraries\TinyGPSPlus C:\Users\Hp\AppData\Local\Temp\build6460723416148179469.tmp\FreeIMU_serial.cpp -o C:\Users\Hp\AppData\Local\Temp\build6460723416148179469.tmp\FreeIMU_serial.cpp.o 
In file included from FreeIMU_serial.ino:9:0:
C:\Users\Hp\Documents\Arduino\libraries\itg3200filv05/ITG3200.h:46:28: error: expected identifier before numeric constant
 #define WHO_AM_I           0x00  // RW   SETUP: I2C address   
                            ^
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:20:8: note: in expansion of macro 'WHO_AM_I'
        WHO_AM_I       = 0x0F,
        ^
C:\Users\Hp\Documents\Arduino\libraries\itg3200filv05/ITG3200.h:46:28: error: expected '}' before numeric constant
[... cut ...]
     static void vector_normalize(vector<float> *a);
                                         ^
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:96:3: error: expected unqualified-id before 'private'
   private:
   ^
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:103:33: error: 'regAddr' has not been declared
       int testReg(byte address, regAddr reg);
                                 ^
C:\Users\Hp\Documents\Arduino\libraries\L3G/L3G.h:104:1: error: expected declaration before '}' token
 };
 ^
Error compiling.

mjs513, 2015-10-05:
Kamil

Its very hard to read your attached file.  Can you use wordpad.  My suspicion is that you forgot to comment out one of the other boards but I would need to see the code in readable format to verify.  I just tried a compile it works or though there is typo in the FreeIMU.cpp file that will throw an error.  line 729 currently reads gyro_sensitivity = 70.0f and should read gyro_sensitivity = 70.0f;  (have to add a semicolon.

Mike

Zieju, 2015-10-05:
I appreciate that you want to help me. It seems that there is a semicolon at the place. If you would like see my libraries, please look at this zip:
https://drive.google.com/open?id=0ByxnsOtQcfLnWndjRWcxWFYwWEE
Kamil

mjs513, 2015-10-05:
Kamil

Try commenting out this line in the FreeIMU.h file:
#define ST_LSM9DS0   should be //#define ST_LSM9DS0   

as a first step

Mike

Zieju, 2015-10-05:
Done, still same errors.

mjs513, 2015-10-05:
Blad

I replaced your versions with the ones on my desktop and I got the same errors.  I just re-downloaded the files from the repository and it compiles with out a problem. Try downloading the repository and replace your FreeIMU.h and FreeIMU.cpp files with the version you just downloaded.  Just remember to comment out the line above and uncomment the altimu10 board

Zieju, 2015-10-08:
Mike
Sorry for the delay, but I had a lot of work. I've just redownloaded the libraries and it didn't help. I got same errors. But when I commented the line including ITG3200 gyro, the Freeimu serial sketch compiles well. I set everything as you wrote in a wiki, even performed a calibration. But the gyro doesn't work :) I've got AltIMU v.4 and it works pretty good for example with that sketch https://github.com/pololu/minimu-9-ahrs-arduino . It is pretty strange for me and I can't find an error in freeimu.h. I reuploaded my libraries if you would like to look at it one more time ( https://drive.google.com/file/d/0ByxnsOtQcfLnS0FYLS1CU3ZuV0k/view?usp=sharing ).

Cheers, Kamil

mjs513, 2015-10-08:
Kamil

One problem that you are going to have is that the LPS25H (baro) is not supported in the Freeimu library.  The AltIMU10-v3 which I have is supported, which has a different baro.  I did notice that you tried to define a AltIMU10 board but you didn't go far enough.  You would need to add a || defined(AltIMU) to the appropriate gyro, mag and accel as well as to the 9dof and axis aligned lines in freeimu.h.

Mike

Zieju, 2015-10-14:
Since I don't need barometer in my project I ignored the difference between v3 and v4.
I defined it simply as AltIMU v10. So I have to define in as the completely new IMU board? Gyro, accelerometer and magnetometer seems to be defined well in sections that you mentioned and I don't know what I can change there.

Kamil

mjs513, 2015-10-14:
Kamil

Relatively easy to modify the existing Altimuv10 configuration.  Just look a the defines for the pressure sensor and Has_Press.  Then delete the Altimuv10.  that will tell the program not to look for the pressure sensor.  What you really have to do is to define the AltIMUv4 then in later defines tell add that board to the define list for the gyro, accel and magnetomer.  Just look for where the Altimu10v3 board is and add the your board to that list.

Glad you got the gyro readings again.

Mike

mjs513, 2015-10-20:
Going to go ahead and close this issue since I think you are well on your way and the original issue has been resolved. 

Thanks for using the library. --- Mike

closed 2015-10-20 by mjs513