mjs513/FreeIMU-Updates #38: Compile error : accgyro_addr

sbishop61, 2015-10-04:
I am using a GY-86 equivalent to a FREEIMU_v04 (which is what I have set in FreeIMU.h)
Using an earlier version of the libraries and 1.5.7 all good.
With 1.6.5 and the latest version I get compile errors even after deleting the MotionDriver directory.
Any ideas?
Thanks

C:\Users\Stephen\Documents\Arduino 1.6.5 FreeIMU\libraries\FreeIMU\FreeIMU.cpp: In member function 'void FreeIMU::init0(bool)':
C:\Users\Stephen\Documents\Arduino 1.6.5 FreeIMU\libraries\FreeIMU\FreeIMU.cpp:635:27: error: 'accgyro_addr' was not declared in this scope
  accgyro = MPU60X0(false, accgyro_addr);
                           ^
C:\Users\Stephen\Documents\Arduino 1.6.5 FreeIMU\libraries\FreeIMU\FreeIMU.cpp:694:20: error: 'lsm' was not declared in this scope
  uint16_t status = lsm.begin(lsm.G_SCALE_2000DPS, lsm.A_SCALE_2G );
                    ^
C:\Users\Stephen\Documents\Arduino 1.6.5 FreeIMU\libraries\FreeIMU\FreeIMU.cpp: In member function 'void FreeIMU::getRawValues(int*)':
C:\Users\Stephen\Documents\Arduino 1.6.5 FreeIMU\libraries\FreeIMU\FreeIMU.cpp:971:2: error: 'lsm' was not declared in this scope
  lsm.readAccel();
  ^
C:\Users\Stephen\Documents\Arduino 1.6.5 FreeIMU\libraries\FreeIMU\FreeIMU.cpp: In member function 'void FreeIMU::getValues(float*)':
C:\Users\Stephen\Documents\Arduino 1.6.5 FreeIMU\libraries\FreeIMU\FreeIMU.cpp:1049:2: error: 'lsm' was not declared in this scope
  lsm.readAccel();

mjs513, 2015-10-05:
lsm is defined for the lsm9ds0 imu.  If you downloaded the latest library that is what it is currently defaulted to.  Make sure that the define for the st_lsm9dso is commented out and the FREEIMU_v04 define is uncommented.  I just tried it and it compiled without an issue.

sbishop61, 2015-10-05:
Mike,
 thanks - it was the lsm9ds0 define that I had missed.
It compiled as soon as I commented out that define.
Thanks again.

closed 2015-10-05 by sbishop61

mjs513, 2015-10-05:
Not a problem.  Enjoy.  Let me know if you run across any issues.

sbishop61, 2015-10-05:
Just need to slim it down for Uno/Pro as FreeIMU_altitude compiles but while it seems to fit Flash and Ram (just) it doesn't run - too big in reality I suspect.

mjs513, 2015-10-05:
That's on my todo list - actually next believe it or not.  Since most people are using the 6050 going to create a version without all the other options to see if I can slim it down.

sbishop61, 2015-10-05:
Mike, what good news. I most interested in getting compass and accelerometer smoothed altitude from a GY-86 (for a paragliding instrument) - but I realise that might too much on a Arduino Pro.

mjs513, 2015-10-05:
You might want to check out the Teensy 3.1 if you want a small form factor board with a lot of power. :)

mjs513, 2015-10-10:
Added a FreeIMU60XX library.  This is essentially the same as the FreeIMU library tailored to fit on the Arduino Uno and supporting only the Invensese 6000, 6050, 9150 and the 9250 MPU.  It also now only supports the MS5611 pressure sensor.  Had to delete the Motion Detection and Temp Calibration routines to make it fit.  Only includes two libraries.