mjs513/FreeIMU-Updates #39: YPR MPU 9250 Example

tunococeht, 2015-10-11:
Good afternoon,

Great libraries! I was checking out your FreeIMU_yaw_pitch_roll example with MPU 9250. YPR and quaternions are being calculated and Serial.printed but notchanging with motion. I believe it is not recognizing the IMU as 9250. I get accel, gyro and mag from this set up with other code (I2C and SPI (not mag on SPI)). Any tips on getting 9250 working with FreeIMU? I really like what you are doing!

Thanks!
-Tuno

mjs513, 2015-10-12:
Hi Tuno

Did you remember to select the correct IMU in the FreeIMU.h file.  You have to uncomment the Gen_9250 and comment out the current board selected.

Mike

tunococeht, 2015-10-12:
Thanks for the reply Mike! I moved the entire FreeIMU-Updates-master.zip
into the library folder and uncommented Gen_9250 and commented out the
current.

By moving the entire zip contents into it caused multiple debug and
communication files.

L3G.h does not exist in the zip and had to be commented out of
FreeIMU/Example\FreeIMU_yaw_pitch_roll.ino. Got 9250 working. One note is
that the code is hard coded to read 9250 with the address pin pulled to
GND. If the address pin is pulled high, it will still read but the data
will be wrong.

Any pro tips on improving yaw? :) I haven't tried much other than 9250. Do
you have a preference for a 9dof imu setup?

Thanks again!
-Tuno

mjs513, 2015-10-12:
The debug errors and compile issues have been resolved in the latest version of the lib which I just posted yesterday and this morning.  I also updated the ypr example with a new function call to getEuler360degAttidude which returns, YPR, altitude and mag heading in one call.

The l3g lib is a linked lib so if you just do a download you won't get the l3g library.  What you have to do is to click on the l3g -arduino directory on github.  Doing this will bring you to the Pololu page where you can download the l3g library.

Check around line 360 which shows you how to change the default address for the MPU :)

As for pro tips to improve yaw stability.  Take your time in doing the calibration.  You may have to tweak the kp's and Ki;s factors.  CHeck the wiki pages and they will provide you some good info on both topics.

As for favorite boards - the 9250 is good, as well as the 6050.  I also like the LSM9DS0 which I picked up from Adafruit.  So far my favorite pressure sensor is the MS5611.

Hope this helps 
Mike

mjs513, 2015-10-20:
Tuno - I am going ahead and close this issue.  If there is anything else you can always re-open this.

Thanks for using the library.
Mike

closed 2015-10-20 by mjs513