mjs513/FreeIMU-Updates #41: Altimu 10 (version  4) and Adafruit 10-DOF

grezco, 2015-11-29:
Hello,
To use Altimu10 v4 breakout wth FreeIMU, I replaced LPS331 Library by LPS pololu Library that is compatible with v3 and v4 ( https://github.com/pololu/lps-arduino ) , and changed the name library in FreeIMU.h
I also modified FreeIMU.cpp and FreeIMU.h for support Adafruit 10-DOF.

All seem to be correct...

[FreeIMU.cpp.txt](https://github.com/mjs513/FreeIMU-Updates/files/46445/FreeIMU.cpp.txt)
[FreeIMU.h.txt](https://github.com/mjs513/FreeIMU-Updates/files/46446/FreeIMU.h.txt)

mjs513, 2015-11-30:
Morning

I will take a look and will update the files if there are no other gotchas.

Thanks for the update- Mike

mjs513, 2015-12-01:
Looked through the changes and they should work.  Have you tested the code changes on the v4 and the adafruit 10dof IMUs?  - mike

grezco, 2015-12-01:
Hello,

 

Yes Mike, I have tested both.

mjs513, 2015-12-01:
Just made the push and updated the example files to support the new library.  Made a couple of other minor fixes to the examples as well.  

Thanks Again for making the updates.

Mike

closed 2016-01-20 by mjs513