mjs513/FreeIMU-Updates #42: compilation for Teensy 3.2

dethegeek, 2016-01-16:
Hi

I'm investifgating about comilation issues for Teensy 3.2

I noticed several type mismatch, in FreeIMU.h and FreeIMU.cpp 

int is not the same as int16_t.

Compilation passed after changinf int to int16_t. I have to check this chang does not breaks compilation for regular Arduino boards.

An other thing

mjs513, 2016-01-16:
To keep it consistent with other sketches change the  int16 to int for raw_values in the calib sketch and it works fine.  Becareful with changing it in the library itself.  You will probably find that it breaks some of the sensor.

Mike

dethegeek, 2016-01-16:
OK, I'l try this way. Thanks

mjs513, 2016-01-20:
Fixed with latest updates.

Mike

closed 2016-01-20 by mjs513