mjs513/FreeIMU-Updates #43: Bad Preprocessor directive with Adafruit 10DOF

dethegeek, 2016-01-16:
Hi

FreeIMU_serial_calib.ino cannot compile with a Adafruit 10DOF : FreeIMU object does not contains and expose the public magn attribute. 

I need to do some tests to ensure my fix is goot (and probably a pull request)

mjs513, 2016-01-16:
Hi there

I just compiled it without a problem.  Can not seem to duplicate.

Mike

dethegeek, 2016-01-16:
Sorry

I forgot to mention : the error is in the example sketch FreeIMU_serial_calib.ino. 

I disabled all hardware related #include and enabled only LSM303, L3G and BMP085.

I also commented out : 
```
#define Has_LSM303 0
#define HAS_GPS 0
```

By the way, should I use this sketch to calibrate my sensors ?

mjs513, 2016-01-20:
Sorry it took me so long to answer this one but been busy with converting to IDE 1.6.7 and work with Intel boards.  Anyway I traced the error, change the line:

    #if IS_9DOM() && (!HAS_MPU9150()  && !HAS_MPU9250() && !HAS_ALTIMU10() && !HAS_LSM9DS0())

to
        #if IS_9DOM() && (!HAS_MPU9150()  && !HAS_MPU9250() && !HAS_ALTIMU10() && !HAS_ADA_10_DOF() && !HAS_LSM9DS0())

Let me know if this works.  It compiles but I have not actually tested it.  As for your question on what sketch to use.  I would recommend going with FreeIMU_serial_ARM_CPU.ino. 

Mike

mjs513, 2016-01-21:
I posted updates to the library with the above change.  I am going to go ahead and close this issue.

Mike

dethegeek, 2016-01-22:
Just tested. The code compiles, but when I send a 'b' command, I only get plenty of empty lines.

If I check the code, I notice the bunck of #if has no entry for a Adafruit 10DOF IMU.

Why the code for 'r' and 'b' comands are so different ? I thought r is for human readable output, and 'b' is for binary output. 

After making this sketch work, I'll switch to FreeIMU_serial_ARM_CPU.ino.

mjs513, 2016-01-22:
If you look at the serial.ino file you are correct - you need to add HAS_ADA_10_DOF() to line 166.  So it should read:
```
#elif HAS_ALTIMU10() || HAS_ADA_10_DOF() 
```
I am going to have to make this change in several other sketches as well.  As to the reason why the code is so different now entirely sure - this is a legacy from Fabio's original code base that I just carried over.

PS. You could also just use the altimu10 config setting as well, if you want to give it a quick try.

dethegeek, 2016-01-22:
As the changes I suggested are similar to those you do, I'll open tickets for issues I find, and submit pull requests when possible. 

When it is possible, people should help you, then I'll do.

mjs513, 2016-01-22:
Thanks appreciate it.  Two questions - (1) did the change work for you and (2) where do you live.  I am in NY

dethegeek, 2016-01-22:
(1) : yes it works
(2) : I live in France

closed 2016-01-23 by dethegeek