mjs513/FreeIMU-Updates #48: Teensy 3.2 + Adafruit 10DOF : weird raw value

dethegeek, 2016-01-23:
Hi

I'm one step further now. The calibration tool shows my IMU sends wrong data. 

Before trying FreeIMU, I was orking on a sketch found elsewhere on internet, much more simple. I managed to make it work with FreeIMU calibration tool by implementing the needed commands of the FreeIMU protocol.

After a first investigation, I *guess* there is something wrong in FreeIMU::init() our init0() and the Teensy; again with preprocessor directives found here.

My chip was programmed to run @ 72MHz, I tried @ 48MHz without better results.

I'll try to compare with a Arduino Mini Pro soon.

mjs513, 2016-01-24:
One bit of warning when using a 16bot processor like the teensy 3.2.  In the calculated.Ph file you have to change the word value to 4 instead of the preset of 2.  Its right at the beginning of the file.  This might be causing  you problems.  do you get reasonable raw values directly from the FreeIMU sketch
  If so the this is probably the

dethegeek, 2016-01-24:
Hi

You're right, it works better with a word size set to 4. 

Implementing a new command in  the protocol, the IMU may send its word size to the GUI.

mjs513, 2016-01-24:
If you do have it set up as a optional command.  Suggest you use sizeof. I
have used it before.

dethegeek, 2016-01-28:
Hi

I was also thinking about sizeof, and I managed to make something working here https://github.com/dethegeek/FreeIMU-Updates/tree/master%2Bautodetect_word

This needs some testing on various hardware I don't have before making a pull request. Hope you may find some time soon to check it.

btry, 2016-01-30:
Hi 

It appears my code worked with luck. Tried to calibrate my sensors and it failed after word size detection. The trick seems to be with Serial.send_now(), this is specific to the teensy implementation.

mjs513, 2016-01-31:
Just finished a round of changes to compile FreeIMU under 1.6.7.   As soon as I get some times I will give it a try.  What ink file did you modify.

dethegeek, 2016-06-19:
Hi

I'm back working on my IMU. I updated my local copy with your latest changes, and saw the calibration GUI works without my changes to detect word size. 

Anyway, my change did not actually worked because some strange bekavior related to serial communication.

Then I think we may close the issue.

closed 2016-06-19 by dethegeek

mjs513, 2016-06-19:
ok thanks.  good luck.