mjs513/FreeIMU-Updates #51: MPU9150 Erratic Magnetometer Readings

martinferris, 2016-02-25:
Having uploaded and implemented the MPU9150_AHRS.ino example, I'm getting strange and erratic mag readings and was hoping you could help identify the problem. I have edited the code to only display the mag serial prints and this is a series of outputs when the arduino is stationary:

mx = -12521 my = 9042 mz = -31466 mG
mx = -12521 my = 9042 mz = -31466 mG
mx = -9448 my = 1084 mz = -29930 mG
mx = -10985 my = -20424 mz = -27625 mG
mx = -10985 my = -20424 mz = -27625 mG
mx = -11753 my = -15047 mz = -29930 mG
mx = -11753 my = -15047 mz = -29930 mG
mx = -7912 my = -22728 mz = -26089 mG
mx = -14825 my = 24404 mz = -26857 mG
mx = -14825 my = 24404 mz = -26857 mG
mx = -16361 my = -19655 mz = -27625 mG
mx = -13289 my = -27337 mz = -24553 mG
mx = -13289 my = -27337 mz = -24553 mG
mx = -8680 my = -28873 mz = -29162 mG
mx = -8680 my = -28873 mz = -29162 mG

Is this a problem with my setup? I am using an arduino nano328 with an MPU9150 IMU and bluetooth HC-05 module. The test code above was over read USB COM port. I'd appreciate any help, thanks!

mjs513, 2016-02-25:
Hi Martin

A couple of things: 1) make sure you have go into the FreeIMU.h file and uncomment the 9150 imu and make sure all the other boards are commented out.  It sounds like the wrong board is selected and 2) use the serial arm CPU sketch. The one you are using was a special case if the mpl3115 pressure sensor is used.

Hope this helps

Mike

martinferris, 2016-02-25:
Hi Mike,

I turns out I had forgotten to add this line of code in the setup:
`    Wire.begin();`

I can't believe it was so trivial! I'm still new to this whole Arduino world. Thank you for the quick response and feedback, it was greatly appreciated.

All the best,
Martin

closed 2016-03-20 by mjs513