mjs513/FreeIMU-Updates #52: FreeIMU GUI

martinferris, 2016-04-10:
I'm trying to use the FreeIMU GUI but whenever I connect to the IMU's serial port the application freezes. The cmd window's last line indicates that "Arduino serial port opened correctly" and nothing further. I noticed this comment in the cal_gui.py file "# TODO: hangs if a wrong serial protocol has been loaded. To be fixed.". However, the correct serial port is COM3, and that is what I have entered into the input text field inside the GUI.

I want to use the GUI to get some test data to try and understand why there is a lot of drift before finding the IMU's absolute position between movements. 

I'm using an MPU9150 on a Nano Atmega328 and am pretty new to Arduino stuff so any help would be greatly appreciated, thanks!

mjs513, 2016-04-13:
Hi Martin

Sorry for the delay in getting back to you - I haven't used a Nano before so I assume it works the same as all the other Arduino boards.  Make sure you connect the Nano to usb port, get the COM# and use that in the GUI.  Just remember you have to send the data to the serial port and the Bluetooth - not sure how you have that set up - usually its through telling the Arduino to send to SerialX instead of Serial.
As for your drift issue make sure you have the 9150 board selected FreeIMU.h file - uncomment board and make sure you comment out all others. See what you get.
If you are still getting a lot of drift try using Marg = 1 or Marg =4.  If still you may have to tweak the tuning factors in the FreeIMU.h file.  Make sure you call both the accel and mag.

Good Luck
Mike

martinferris, 2016-04-14:
Hi Mike,

No problem at all, thanks for getting back to me. I'll give all those a shot, but I'm in the process of trying to get the GUI working.

Unfortunately I only have 30kb of memory on the Nano Atmega328 given to me on this project so I'm trying to slim down the FreeIMU_serial_ARM_9150_3115A2 sketch. I'm looking through the cal_gui.py to see what functions are called from the IMU to see how much of the sketch i can cut out.

martinferris, 2016-04-14:
Hi Mike,

I've managed to cut down the FreeIMU_serial_ARM_9150_3115A2, however global variables use 1,902 bytes (92%) of dynamic memory, leaving 146 bytes for local variables. When I run it, the nano seems to get stuck in the init() function. I have investigated the FreeIMU.cpp file, and everything seems to make sense but I can't figure out why it's getting stuck in that loop.

Any ideas?

martinferris, 2016-04-14:
Must've been the lack of memory! Switched the Marg to 3 and the resulting dynamic memory used was only 67%. 

Thanks again for your help Mike, it's greatly appreciated :)

closed 2016-04-14 by martinferris

mjs513, 2016-04-15:
Yeah thats the way it sounds. You can try 4 as well which is a dcm implementation. Good luck