mjs513/FreeIMU-Updates #53: LSM9DS1 Settings inconsistency in freeIMU.cpp

twmeares, 2016-04-11:
In the freeimu.cpp file you have `lsm.settings.gyro.scale = 245;` and a few lines later you have  
```
    //Angular rate FS = ±245 dps, gyro_sensitivity = 8.75
	//Angular rate FS = ±500 dps, gyro_sensitivity =  17.50
	//Angular rate FS = ±2000 dps, gyro_sensitivity =  70
	gyro_sensitivity = 70.0f; 
```
Don't these values conflict? With a scale of 245 the sensitivity should be 8.75 or for a scale of 2000 sensitivity should be 70.

mjs513, 2016-04-12:
You are correct. The scale and sensitivity should match per the comments.
Will update the lib so that the gyro.scale=2000. I am on my phone right
now.  Thanks for pointing it out.

Mike

twmeares, 2016-04-12:
Interesting enough, after changing the values to gyro.scale = 2000; and sensitivity = 70.0; and re-calibrating the IMU, I get really odd performance. physically moving the IMU 90 degree will produce only about a 4 or 5 degree change in yaw. Conversely using gyro.scale = 245 and sensitivity = 8.75 the IMU will move more than a full 360 rotation if I physically move the board about 90 degrees. I am quite confused by this behavior because the mismatch gyro.scale = 245 and sensitivity = 70 seems to produce the best results.

mjs513, 2016-04-13:
There is something strange going on with my gyro right now and sure why. If
you rotate the board do you see any real changes in your gyro readings.
Have to go back to specs.

mjs513, 2016-04-13:
Hi twmeares ,
Tracked down what could be issue.  I added test for data availability, adjusted setup values and tweaked the filter factors.  It seems to be responsive (don't use Marg = 3 for now).  Gyro is at 245 and accel at 2.  Only files changed were FeeIMU.h and FreeIMU.cpp.

Cheers
Mike

twmeares, 2016-04-13:
Hi Mike,

I believe the problem lies in the implementation of converting the raw gyro values to into dps. If you take a look at the data sheet you'll notice it says that for scale 2000 the sensitivity is 70 mdps/LSB. In your implementation you're using values_cal[3] = (values_cal[3] - gyro_off_x) / gyro_sensitivity; So gyro_sensitivity should be in terms of LSB/dps. Which would make 70 mdps/LSB = 14.285 LSB/dps. However, I went further to inspect the sparkfun libraries implementation to find that they were using 32768/2000 to get the sensitivity which equates to 16.384 LSB/dps. The numbers sparkfun uses come from 32768 being half of the range of a 16 bit number and 2000 being half of range of the gyro. 

I'm not sure which value is better to use. visually it seems that the 16.384 is producing better results for me. Thanks again for adding the LSM9DS1 to the freeIMU library. 

Regards,
Taylor

mjs513, 2016-06-06:
There is an update correcting a situation where Freeimu will be getting confused between the 0 and 1 version.

closed 2016-06-06 by mjs513