mjs513/FreeIMU-Updates #54: FreeIMU GUI Slow Read

martinferris, 2016-04-19:
Hi Mike,

I've managed to fit the serial sketch onto my Nano, but when I run the calibration sampling it seems to be only sampling at a rate of once every 1.5 or so seconds. From watching people's calibration videos, it should be considerably faster. 

Any ideas as to why? I've tested sending the 'b' character in the serial terminal (which is what the GUI sends), and it only returns values when i send 'b' twice. Here is the sampling section of code for your convenience: 

```
def run(self):
    print "sampling start.."
    self.acc_file = open(acc_file_name, 'w')
    self.magn_file = open(magn_file_name, 'w')
    count = 100
    in_values = 9
    reading = [0.0 for i in range(in_values)]
    # read data for calibration    
    while not self.exiting:
      # determine word size   
      self.ser.write('b')
      self.ser.write(chr(count))
      for j in range(count):
        for i in range(in_values):
          if word == 4:
            reading[i] = unpack('hh', self.ser.read(4))[0]
          if word == 2:
            reading[i] = unpack('h', self.ser.read(2))[0]            
        self.ser.read(2) # consumes remaining '\r\n'
        if reading[8] == 0:
          reading[6] = 1
          reading[7] = 1
          reading[8] = 1      
        # prepare readings to store on file
        acc_readings_line = "%d %d %d\r\n" % (reading[0], reading[1], reading[2])
        self.acc_file.write(acc_readings_line)
        magn_readings_line = "%d %d %d\r\n" % (reading[6], reading[7], reading[8])
        self.magn_file.write(magn_readings_line)
      # every count times we pass some data to the GUI
      self.emit(SIGNAL("new_data(PyQt_PyObject)"), reading)
      print ".",
    # closing acc and magn files
    self.acc_file.close()
    self.magn_file.close()
    return 
```

 I don't think its my memory usage causing the problem like before, but here is it anyway:
```
Sketch uses 13,182 bytes (42%) of program storage space. Maximum is 30,720 bytes.
Global variables use 1,306 bytes (63%) of dynamic memory, leaving 742 bytes for local variables. Maximum is 2,048 bytes.
```

mjs513, 2016-04-19:
Try changing the baud rate in the sketch and python to 115200

martinferris, 2016-04-19:
Doubled checked, they are both set to 115200 already

mjs513, 2016-05-19:
Believe there is some code in the file that skips some data.  Can't remember exactly the lines but look around where it does the serial reads

Mine

1oginov, 2016-05-25:
Got the same problem, check line 398 in cal_gui.py:

mjs513, 2016-05-28:
@1oginov has got it right. Just one bit of warning.  There is no such thing as a stock calibration.h file.  You need to do your own calibration for your IMU. Temperature, soft/hard iron effects can affect the calibration.  Let me know if this works out or there is another issue.

closed 2016-06-06 by mjs513