mjs513/FreeIMU-Updates #55: No accelerometer data in GUI

xepheria, 2016-05-06:
I'm using an Adafruit 10 DOF chip with an Arduino Uno board and the FreeIMU60XX sketch. I'm able to view magnetometer data in the GUI just fine, but I can seem to get any accelerometer output. When I run the Processing application I'm also not getting any change in the 3d cube. Am I missing a dependency or something? Thanks for any assistance.

mjs513, 2016-05-06:
Make sure you have this line uncommented in FreeIMU.h:

#define ADA_10_DOF

 

All other boards should be commented out.

xepheria, 2016-05-06:
Hi,

Thanks for the quick response. I do have that board uncommented only, and the others are all commented out. But that board isn't included in the list of boards in the FreeIMU_6050.h file. I have to use that one since I'm using a 16-bit board.

mjs513, 2016-05-19:
You should be using the FreeIMU serial arm CPU.ino sketch. The 6050.h file is for a mpu6050 board not the adafeuit board.

mjs513, 2016-06-06:
Closing issue.  If problem persists please just reopen.

closed 2016-06-06 by mjs513