mjs513/FreeIMU-Updates #56: Any instructions for calibration? (I'm happy to write them with a little help)

scornflake, 2016-05-18:
Hi!

I've been playing with a Teensy3.2, Adafruit 10DOFv4 and also LSM9DS0.

I've some success with the Adav4 and it's own example, but little with FreeIMU.

I'm located in NZ (southern hemisphere), and wondering if I need to adjust sensor signs?
Overall, I'd like to develop a readme/getting_started.txt that shows how to calibrate boards, how to work out what the sensor sign should be, and how to check ranges given by FreeIMU (i,e: accell should have range -1..1 approx).

What's stopping me is I'm not exactly sure on the following:

- Whens stationary, should accell (getValues()) Z be +ve or -ve 1g?
- When rolling right to 90 deg, what is the correct value/sign for accell X and Mag X?
- When pitching up to 90g, what is the correct value/sign for accell Y, Mag Y?
- Does the MagZ sign need to be reversed for southern hemisphere?
- Does calibration.h need to be deleted and the code recompiled before using cal_gui.py? 

I guess what I am trying to achieve is a known correct baseline, before I begin experimenting with the code further. With my current attempts at processing sketches, I'm not getting sensible rotations on the device. This leads me to think I have some initial setup incorrect, hence wanting to write up / contribute the above.

If this is not the correct forum for this question, let me know a more appropriate place.

scornflake, 2016-05-18:
OK. I found this:

https://github.com/mjs513/FreeIMU-Updates/wiki/07.-FreeIMU---Axis-Alignment---Adding-a-New-IMU

Which has helped.

So maybe something else is wrong.  I've reverted all the code to HEAD, and reset the sensor_sign to 1,1,1 1,1,1 1,1,1. (Altimu-v4, axis aligned).

It works well (sort of, but slow in response) on MARG=0, but other filters cause either 1) jitter, 3) lack of resolution, with 90 deg showing up as about 20 deg, or 4) strange rotations around Z when both pitch and roll in effect... that it, without me rotating the device.

I'm not sure if any adjustments to sign need to be made for southern hemisphere (only asking because read that it might be necessary in a Tilt Compensation PDF from STI, for the L303).

So perhaps a rephrased question is:
- How do I verify that my sensors are aligned and outputting data that FreeIMU expects?
- If I were to invert the IMU on the breadboard, is it possible to adjust sensor_sign for this (I was never able to, and feel I'm missing an understanding of how the 3 sets of readings interact with reality)?

mjs513, 2016-05-19:
Hi scornflake

I agree something does not sound rignt.

In the wiki you will find a couple of other pages on calibration and setting up sensor signs and alignment.  Accuracy sounds like you don't have a good calibration on your sensors. As for jitter that could be that you need to adjust the tuning factors.  That is addressed on one of the wiki pages as well.

One question I can answer right away is you news to run Cal GUI.py for your board and location.  Yes delete the calibration.h file and generate your own.  As for slow response I usually have to slow the sketch up when using the trendy.

I will check it out tomorrow with the adafruit 10dof.  As for the southern hemishphere I will have check it out.

Mike

mjs513, 2016-05-19:
Ok.  I had to fix the code for the lsm9ds0 in freeimu.cpp file.  I am attaching it.  So just download and replace the existing file.  It should work much better now.

[FreeIMU.zip](https://github.com/mjs513/FreeIMU-Updates/files/272587/FreeIMU.zip)

scornflake, 2016-05-20:
Thanks, what did you change in your .zipped version (I've not diff’ed it yet).

I had a push almost ready to fix a couple of minor naming issues so that it’d compile… 

Re setup/behaviour: I’ll take another look and maybe do a video as well, so you can get a better feel for setup.  Funny you mention MinIMU as that's the other main sketch I've been playing with. In fact I'll get that on video as well, and you'll see just how much faster it is in response (it's quicker, by far).

I'll go from scratch again, with a fresh calibration.

scornflake, 2016-05-20:
Here's how I'm doing setup:

https://dl.dropboxusercontent.com/u/421935/FreeIMU/FreeIMU%20Setup.m4v

I'll do another video to show the results in both MinIMU and FreeIMU. Out tonight so will prob do it tomorrow.

mjs513, 2016-05-20:
I had to fix usage for the LSM9DS0 vs LSM9DS1.  In some cases I had used a common code which was incorrect and would not compile.  So for the LSM9DS0 it should be much better.  One note that I just remembered is that the orientation of the Altimu-10v3 is where the chips are facing down.  If you do it the other way you would probably have to adjust the sensor signs which I think you did based on the quick look in the video.  Marg = 4 is a dcm implementation which would give you a comparison to the MiniIMU.

Not sure why it is so slow for you.  There is a timing sketch that you can run for raw data to give you an idea.  Over the years I have pretty much looked at most of the implementations out there including Kris's, pololu's, apm, etc. to try and keep yaw drift to a minimum. :)

scornflake, 2016-05-21:
Question: for an axis, if I point that axis towards earth, should I be reading +1g/ + ~gauss, or -ve?

mjs513, 2016-05-21:
I get az negative and mz positive when pointing the z axis up or down respectively.

I am attaching two files, a zip with a new Arduino sketch and a new processing sketch.  I added a burst variable to the mix to allow the you to change the amount of data that comes across in one shot.  Right now its set to 5096. You can still use this for calibration and looking at raw data. Still playing with them.  For the lsm9ds0 with Teensy 3.2 its pretty fast about 0.0035 seconds between calculations.

[FreeIMUcubeOdoExpv5.zip](https://github.com/mjs513/FreeIMU-Updates/files/275810/FreeIMUcubeOdoExpv5.zip)

[FreeIMUserialprocessing.zip](https://github.com/mjs513/FreeIMU-Updates/files/275809/FreeIMUserialprocessing.zip)

scornflake, 2016-05-22:
Oh, I have a v4, if I said v3 it was only because there was a drawing of it. Sorry, didn't mean to mislead there.

mjs513, 2016-05-22:
Thats ok i ordered a v4 as well they were on clearanc

scornflake, 2016-05-22:
I have made changes to FreeIMU so that it continuously streams data at a fixed rate to the client. I didn't include any of that because I didn't want to muddy the issues I'm having (and I resort to the git FreeIMU when I'm testing stuff anyway).  I *thought* I had a really good method of calibrating last week, but I'm now going around in circles again. Hence asking about the signs on axis.  So should an accelerometer read -ve if it's axis is pointing upwards?  And a mag read +ve  in the same condition?  Or are the signs also entirely dependent upon what the algorithm expects?

mjs513, 2016-05-22:
I am in the er with my 100yr old mother right now. When i get a chance i will post something for you to use as ref.

scornflake, 2016-05-22:
Take your time. Don't worry about me! Go spend time with your mother.

mjs513, 2016-05-22:
I am sitting by her now. She is sound asleep. At least this is giving me
something to do until they get her admitted.

scornflake, 2016-05-22:
I understand.  My thoughts are with you (I've been in similar situation with my own mum).  Any time you want a distraction I'll be happy to confuse us all with my own confusedness.

mjs513, 2016-05-22:
Always new NZers were good people. Thanks all confusions welcome

Mike

scornflake, 2016-05-22:
Oh, I'm not sure you want to be stepping into my head. I'm not even sure what's going on in there myself sometimes! 

Last weeks *genius* thought was that if I aligned my axis such that they lined up with what the algorithm wanted (X right, Y up, Z down, clockwise....) then all would be good. I had hoped I could ignore the orientation on the board, and simply stare at the sensor output, swapping signs as required until each axis had its correct sign.  However; either that plan is flawed or my execution sucks. Hence Q re axis. I'm now questioning if I even have the axis in the right direction, and what the signs *should* be.  Is my approach valid?

mjs513, 2016-05-28:
Hi scornflake.  Didn't forget you.  Anyway on the LSM9Ds0 board I am using z-axis points up, y-axis points to the left and the x-axis points forward.  
[image]

[image]

The two screenshots should answer your questions.  Remember the magnetometers will probably be different since you are in the southern hemisphere.  So don't panic on the sign.  I think there are a couple of links on the wiki that will lead you to a web site that you can put your location in and determine what they should be.

Cheers
Mike

scornflake, 2016-05-29:
Awesome thanks!

I got tied up doing “work” last week, so haven’t had another chance to sit down with this.
Hopefully this week!

—
Neil Clayton
neil@cloudnine.net.nz

mjs513, 2016-05-29:
Hey Neil
Just to save you some time here is the alignment that I am using in the freeimu.h file

```
int sensor_order[9] = {0,1,2,3,4,5,6,7,8};
int sensor_sign[9] = {1,1,1,1,1,1,1,1,-1};	
```
Good luck.
Mike

closed 2016-10-27 by mjs513