mjs513/FreeIMU-Updates #57: freeimu update doesn't work

belagio, 2016-05-26:
Hi
I upload code to an arduino. my code has problems because L3G library has problem .
after I delet L3G from code the code is compiled ,but when I upload it on my arduino board and test it with processing just show zero. why my freeimu doesn't work?
thank you

mjs513, 2016-05-26:
Need a little more info. What imu are using. What arduino are u using.

belagio, 2016-05-28:
I use GY-87 and try both arduino Uno and Mega 2560.

mjs513, 2016-05-28:
Ok.  The GY-87 is a MPU6050+HMC5883L+BMP180 so it does not need to have the l3d library.  That's number 1.  Second you need to make sure you have the correct configuration set up in the FreeIMU.h file for the IMU that you are using.  It is explained in the wiki.  Third to test whether the sketch is working you can issue a 'v' command in the Arduino serial monitor to see if you can a response.  You can also check the raw data by using the 'r' command.  Just remember when not to use the single quotes as part of the command.  If all is ok then make sure that you have the correct serial port selected and speeds setup in the processing sketch.  If that is all ok just close the processing sketch and try reopening it a few time.  Sugggest that you read the wiki.

closed 2016-06-06 by mjs513