mjs513/FreeIMU-Updates #58: can't compile

yigityildirim, 2016-05-27:
Hi,

I have Arduino due and MPU6050 with the HMC5883 magnetometer. One year ago, I managed to have fair results from the sensor, now I have updated the library, Arduino IDE and i got these kind of compiling errors in FreeIMU_serial_ARM_CPU:

In file included from FreeIMU_serial_ARM_CPU.ino:27:0:
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:29:10: error: 'constexpr' does not name a type
   static constexpr float Coef1 = 1.9777864838f;
          ^
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:29:10: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:30:10: error: 'constexpr' does not name a type
   static constexpr float Coef2 = -0.9780305085f;
          ^
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:30:10: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:31:10: error: 'constexpr' does not name a type
   static constexpr float GAIN = 1.639178228e+04f;
          ^
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:31:10: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:46:10: error: 'constexpr' does not name a type
   static constexpr float Coef1 = 1.9555782403f;
          ^
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:46:10: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:47:10: error: 'constexpr' does not name a type
   static constexpr float Coef2 = -0.9565436765f;
          ^
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:47:10: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:48:10: error: 'constexpr' does not name a type
   static constexpr float GAIN = 4.143204922e+03f;
          ^
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:48:10: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:62:10: error: 'constexpr' does not name a type
   static constexpr float Coef1 = 1.9111970674f;
          ^
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:62:10: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:63:10: error: 'constexpr' does not name a type
   static constexpr float Coef2 = -0.9149758348f;
          ^
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:63:10: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:64:10: error: 'constexpr' does not name a type
   static constexpr float GAIN = 1.058546241e+03f;
          ^
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:64:10: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:78:10: error: 'constexpr' does not name a type
   static constexpr float Coef1 = 1.8668922797f;
          ^
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:78:10: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:79:10: error: 'constexpr' does not name a type
   static constexpr float Coef2 = -0.8752145483f;
          ^
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:79:10: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:80:10: error: 'constexpr' does not name a type
   static constexpr float GAIN = 4.806381793e+02f;
          ^
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:80:10: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
/Users/mac/Documents/Arduino/libraries/AP_Filter/Butter.h:94:10: error: 'constexpr' does not name a type
   static constexpr float Coef1 = 1.8226949252f;

----------------------------------------------------------------------------------------------------------------------------

In addition, I couldn't disable the magnetometer just by uncommenting the disable_magn line with the old settings I used a year ago. Thanks in advance.

mjs513, 2016-05-28:
Yes I know I had to rework some of the libraries and the freeimu proper to get around the changes in the ide.  Try downloading the latest version.  I mention some of the changes I had to do as a result of the ide change in the readme.

mjs513, 2016-06-06:
Closing issue since I have not heard back.  Can only assume it is now working.

closed 2016-06-06 by mjs513