mjs513/FreeIMU-Updates #60: problem with arduino 101 test

ase1930, 2016-06-10:
while compiling  CurieFreeIMU_serial_ARM_CPU sketch with arduino 101 
it gives an error says 
C:\Users\ahmed\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:499:8: error: 'void FreeIMU::init0(bool)' cannot be overloaded

   void init0(bool fastmode);

mjs513, 2016-06-11:
It appears that you are using MARG filter set to 4 in the freeimu.h file.  MARG=4 selects the DCM filter which requires a magnetometer to be attached.  Change this value to 1 so it should read MARG=1 and it will compile fine.  I just checked it.  Make you do a calibration first.  See the wiki.

ase1930, 2016-06-12:
hello sir, I have changed MARG=1  but another problem occurred 

it says 

In file included from C:\Users\ahmed\Documents\Arduino\libraries\CurieImu\examples\CurieFreeIMU_serial_ARM_CPU\CurieFreeIMU_serial_ARM_CPU.ino:26:0:

C:\Users\ahmed\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:499:8: error: 'void FreeIMU::init0(bool)' cannot be overloaded

   void init0(bool fastmode);

        ^

C:\Users\ahmed\Documents\Arduino\libraries\FreeIMU/FreeIMU.h:486:7: error: with 'void FreeIMU::init0(bool)'

  void init0(bool fastmode);

       ^

From: Mike S
Sent: ‎Saturday‎, ‎June‎ ‎11‎, ‎2016 ‎11‎:‎19‎ ‎PM
To: mjs513/FreeIMU-Updates
Cc: ase1930, Author

It appears that you are using MARG filter set to 4 in the freeimu.h file. MARG=4 selects the DCM filter which requires a magnetometer to be attached. Change this value to 1 so it should read MARG=1 and it will compile fine. I just checked it. Make you do a calibration first. See the wiki.

ase1930, 2016-06-12:
this is an image from my  freeimu.h file

 

Sent from Windows Mail

From: Mike S
Sent: ‎Saturday‎, ‎June‎ ‎11‎, ‎2016 ‎11‎:‎19‎ ‎PM
To: mjs513/FreeIMU-Updates
Cc: ase1930, Author

It appears that you are using MARG filter set to 4 in the freeimu.h file. MARG=4 selects the DCM filter which requires a magnetometer to be attached. Change this value to 1 so it should read MARG=1 and it will compile fine. I just checked it. Make you do a calibration first. See the wiki.

mjs513, 2016-06-12:
I compiled it today as a test and had no issues. So here goes a few questions.  I am using 1.6.7 idea with the lastest curie board - 1.0.6.  When did you download the library?

ase1930, 2016-06-12:
I am also using  1.6.7 idea with the lastest curie board - 1.0.6 and I have downloaded freeimu library 2 days ago.

Sent from Windows Mail

From: Mike S
Sent: ‎Sunday‎, ‎June‎ ‎12‎, ‎2016 ‎5‎:‎02‎ ‎AM
To: mjs513/FreeIMU-Updates
Cc: ase1930, Author

I compiled it today as a test and had no issues. So here goes a few questions. I am using 1.6.7 idea with the lastest curie board - 1.0.6. When did you download the library?

mjs513, 2016-06-12:
Are you sure you have the board configuration set to just CurieIMU.  In other words in the FreeIMU.h file the CurieIMU board is uncommented and all other boards are commented in that section.  I can not seem to be able to duplicate your issue and for some reason I still can not see the pasted image.

ase1930, 2016-06-12:
this is my freeimu.h and freeimu.cpp files

Sent from Windows Mail

From: Mike S
Sent: ‎Sunday‎, ‎June‎ ‎12‎, ‎2016 ‎1‎:‎54‎ ‎PM
To: mjs513/FreeIMU-Updates
Cc: ase1930, Author

Are you sure you have the board configuration set to just CurieIMU. In other words in the FreeIMU.h file the CurieIMU board is uncommented and all other boards are commented in that section. I can not seem to be able to duplicate your issue and for some reason I still can not see the pasted image.

mjs513, 2016-06-13:
Still no files attached or images. Sguggest you zip the files and upload it as an attachment.  PS. Have you gone through hthe wiki on setting configurations.

ase1930, 2016-06-15:
this is my freeimu.h and freeimu.cpp files 

Sent from Windows Mail

From: Mike S
Sent: ‎Monday‎, ‎June‎ ‎13‎, ‎2016 ‎2‎:‎25‎ ‎AM
To: mjs513/FreeIMU-Updates
Cc: ase1930, Author

Still no files attached or images. Sguggest you zip the files and upload it as an attachment. PS. Have you gone through hthe wiki on setting configurations.

mjs513, 2016-06-30:
Ok.  I have not heard back so I am going to go ahead and close this

closed 2016-06-30 by mjs513