mjs513/FreeIMU-Updates #63: Can't build to board

dahburj, 2016-06-25:
Hello, 

I'm using a GY-85 IMU sensor and an arduino mega 2560 and I can't build this thing at, here are the error messages I'm getting:

Arduino: 1.6.9 (Windows 8.1), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from C:\Users\Jamal\Documents\Arduino\libraries\FreeIMU\examples\FreeIMU_quaternion\FreeIMU_quaternion.ino:30:0:

C:\Users\Jamal\Documents\Arduino\libraries\AP_Math_freeimu/AP_Math_freeimu.h:31:0: warning: "DEG_TO_RAD" redefined [enabled by default]

 #define DEG_TO_RAD 0.017453292519943295769236907684886f

 ^

In file included from sketch\FreeIMU_quaternion.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:50:0: note: this is the location of the previous definition

 #define DEG_TO_RAD 0.017453292519943295769236907684886

 ^

In file included from C:\Users\Jamal\Documents\Arduino\libraries\FreeIMU\examples\FreeIMU_quaternion\FreeIMU_quaternion.ino:30:0:

C:\Users\Jamal\Documents\Arduino\libraries\AP_Math_freeimu/AP_Math_freeimu.h:32:0: warning: "RAD_TO_DEG" redefined [enabled by default]

 #define RAD_TO_DEG 57.295779513082320876798154814105f

 ^

In file included from sketch\FreeIMU_quaternion.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:51:0: note: this is the location of the previous definition

 #define RAD_TO_DEG 57.295779513082320876798154814105

 ^

In file included from C:\Users\Jamal\Documents\Arduino\libraries\AP_Filter/LowPassFilter.h:26:0,

                 from C:\Users\Jamal\Documents\Arduino\libraries\AP_Filter/Filter.h:11,

                 from C:\Users\Jamal\Documents\Arduino\libraries\FreeIMU\FreeIMU.cpp:300:

C:\Users\Jamal\Documents\Arduino\libraries\AP_Math_freeimu/AP_Math_freeimu.h:31:0: warning: "DEG_TO_RAD" redefined [enabled by default]

 #define DEG_TO_RAD 0.017453292519943295769236907684886f

 ^

In file included from C:\Users\Jamal\Documents\Arduino\libraries\FreeIMU\FreeIMU.cpp:292:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:50:0: note: this is the location of the previous definition

 #define DEG_TO_RAD 0.017453292519943295769236907684886

 ^

In file included from C:\Users\Jamal\Documents\Arduino\libraries\AP_Filter/LowPassFilter.h:26:0,

                 from C:\Users\Jamal\Documents\Arduino\libraries\AP_Filter/Filter.h:11,

                 from C:\Users\Jamal\Documents\Arduino\libraries\FreeIMU\FreeIMU.cpp:300:

C:\Users\Jamal\Documents\Arduino\libraries\AP_Math_freeimu/AP_Math_freeimu.h:32:0: warning: "RAD_TO_DEG" redefined [enabled by default]

 #define RAD_TO_DEG 57.295779513082320876798154814105f

 ^

In file included from C:\Users\Jamal\Documents\Arduino\libraries\FreeIMU\FreeIMU.cpp:292:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:51:0: note: this is the location of the previous definition

 #define RAD_TO_DEG 57.295779513082320876798154814105

 ^

C:\Users\Jamal\Documents\Arduino\libraries\FreeIMU\FreeIMU.cpp:372:0: warning: "M_PI" redefined [enabled by default]

 #define M_PI 3.14159265359

 ^

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:26:0,

                 from C:\Users\Jamal\Documents\Arduino\libraries\FreeIMU\FreeIMU.cpp:292:

c:\program files (x86)\arduino\hardware\tools\avr\avr\include\math.h:86:0: note: this is the location of the previous definition

 #define M_PI  3.14159265358979323846 /* pi */

 ^

C:\Users\Jamal\Documents\Arduino\libraries\FreeIMU\FreeIMU.cpp:1478:4: warning: #warning Accelerometer calibration active: have you calibrated your device? [-Wcpp]

   #warning Accelerometer calibration active: have you calibrated your device?

    ^

C:\Users\Jamal\Documents\Arduino\libraries\FreeIMU\FreeIMU.cpp:1497:4: warning: #warning Magnetometer calibration active: have you calibrated your device? [-Wcpp]

   #warning Magnetometer calibration active: have you calibrated your device?

    ^

C:\Users\Jamal\Documents\Arduino\libraries\FreeIMU\FreeIMU.cpp: In member function 'void FreeIMU::init(int, int, bool)':

C:\Users\Jamal\Documents\Arduino\libraries\FreeIMU\FreeIMU.cpp:670:30: error: 'accgyro_addr' was not declared in this scope

     accgyro = MPU60X0(false, accgyro_addr);

                              ^

Multiple libraries were found for "I2Cdev.h"
 Used: C:\Users\Jamal\Documents\Arduino\libraries\I2Cdev
 Not used: C:\Users\Jamal\Documents\Arduino\libraries\MotionDriver
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

I deleted all my libraries and then imported only the ones from this master folder, and I commented out version 3 in the header file and comment Curie and uncommented gy-88 (I assumed that was the closest to mine) and alas to no avail.

mjs513, 2016-06-25:
wrong board selected for your setup.  The GY-85 consist of a ITG3205 + ADXL345 + HMC5883L while the GY-88 consists of a MPU-6050 HMC5883L.  For the GY-85 you should uncomment the following:

`#define FREEIMU_v03`

and make sure all other boards are commented out.  I have not tested your configuration so not sure if the sensor axes are aligned or not.

dahburj, 2016-06-25:
Thank you, now it worked. However I feel the sensors aren't aligned, with the other version, I would build the Quaternion script and import those values into unit and they would rotate my object, however, drifted unfortunately. With this it's all over the place, and constantly shaking, do you know how I can fix this issue?

mjs513, 2016-06-25:
Suggest that you read the wiki.  But to make a long story short you need to do the calibration for the sensor per the wiki and you need to try different MARG filter - one may be better than the other.  The other thing is you may need to adjust the factors used by the MARG filter used to stabilize the drifting and shaking.  Try MARG=1 or MARG = 4 to start.  As I said I really have not played with the GY-85 so little testing has been done.  Suggest you also read up on the filters themseleves so you can understand what you are dealing with.

dahburj, 2016-06-25:
Sounds good, I think I might as well buy another IMU while I'm at it that will work with your code, any recommendations on what I should buy that will work best with this?

mjs513, 2016-06-30:
I always liked the 6050 and 5883 combo.  But there are several others out there that work well.  The altimu10 from Pololu or a board with the MPU950 imu.

closed 2016-06-30 by mjs513