mjs513/FreeIMU-Updates #65: wireless connection

belagio, 2016-07-11:
Hi
I want to use freeimu with wireless connection. I have HM-TRP modules for wireless connection. After I assemble this module to free imu and test it the processing shows zero.
Please help me how can I make freeimu with wireless connection.
Thank you.

mjs513, 2016-07-11:
It looks like the module you are using requires a separate arduino library.
Also since I do not have these modules and I don't know what is required on
pc side with processing to implement correctly.  Right now I am only
supporting xbee since that is the most common.  Sorry I couldn't be of more
help.

belagio, 2016-07-13:
Thank you
If I want to use xbee, what kind of this I can use?
If I want to use xbee, I should use freeimu serial for arduino and freeimu cube xbee for processing.
Thank you

closed 2016-10-27 by mjs513

belagio, 2017-01-30:
Hello MJ
Thank you for helping me.
Bye