mjs513/FreeIMU-Updates #67: Sparkfun LSM9DS1

dahburj, 2016-10-25:
Hi, 

I'm having alot of issues with the sparkfun LSM9DS1 and this library, I've tried calibrating it multiple times, and I get the circles and upload the calibration file, have an issue building if the #include <L3G.h> isn't commented out, so I comment it out. However, if I try to use the processing cube, it takes a long time for the information to go out to the cube (>30 seconds) and when it finally does, any little movement sends the cube into a frenzy. Please help, I'm so looking forward to using the AHRS capabilities of this library. Is there something I'm missing? Is the calibration process different then just using the calibration GUI provided?

mjs513, 2016-10-27:
First thing I would try are some of the other filters available in the lib.  Check line 75 in FreeIMU.h.  If that does not work you may need to adjust the tuning factors around line 167 in FreeIMU.h.  Check the wiki on how to do that.

dahburj, 2016-10-28:
I tried the different filters that didn't work, I also tried messing around with those tuning factors (I couldnt find out how on the wiki though) by making the numbers smaller in hoping it will work but that didn't help either. I did notice one thing while testing, the gyroscope is def the issue. The numbers in processing cube sketch that refer to the gyroscope go >+-50.00, I think those numbers are huge and that's whats causing this crazy movement at the slightest touch. How can I adjust the gyroscope sensitivity for the lsm9ds1? If that is the problem, if not, what do you think it could be?

dahburj, 2016-10-28:
I'm trying to figure how to from the gyro sensitivity section on the wiki, I just don't know which one applies to my chip *hitting forehead on table for being so ignorant*

mjs513, 2016-11-24:
The sensitivity is set in the FreeIMU.cpp file.  Look for lsm9ds1.

mjs513, 2016-12-01:
By the way, leave the l3g uncommented and comment out the itg3200.h and you won't have a problem.  I getting ready to upload some changes to incorparte the new libs from pololu.

closed 2016-12-01 by mjs513