mjs513/FreeIMU-Updates #70: Problem compiling FreeIMU_serial.ino

jose-naharro, 2016-12-09:
Hi, I'm trying to compile and upload to an Arduino Mega 2560 with a MPU6560+HMC5883L+BMP180 10Dof board.
Compilation is aborted. This is the error message:
...\Arduino\libraries\CurieImu\src\CurieImu.cpp:22:23: fatal error: interrupt.h: No such file or directory

 #include "interrupt.h"

                       ^

compilation terminated.

I'm not working with a Curie board, but I've all the libraries that you have in the libraries folder.

Regards.

mjs513, 2016-12-09:
Hi Jose

Suggest that you read the wiki to understand how everything works.  You need to select the proper board configuration in the FreeIMU.h file.  To make it easier you need to comment out the line for the CurieIMU and uncomment the following line //#define FREEIMU_v035_BMP.  That should get you started.  You will need to do a calibration.  Check the wiki.

Mike

jose-naharro, 2016-12-12:
Hi Mike,

Thank you for the suggestios, specialy read the wiki, I'm new in github, arduino and processing.

I forget to say the sketch I'm trying to upload is FreeIMU/examples/FreeIMU_serial/FreeIMU_serial.ino

- I've changed the BaudRate to 115200.
- I've read many issues too and I've chaged all the FreeIMU's libraries to the main Arduino's library folder.

- I've commented and uncommented the lines you told me, but haven't worked, the error:
           my3IMU.getBaroTemperature().

Ok, well, later, I fall in the same issue that others and I uncommnet 2 boards:
#define FREEIMU_v035_BMP
#define GY_87 //that's the board I have

I commented the first and now I've only 1 board:
//#define FREEIMU_v035_BMP
#define GY_87
But I have the error...'class FreeIMU' has no member named 'getBaroTemperature'
I think I found the issue...FreeIMU.h, line 339 there are two defined(GY_88), one must be defined(GY_87) et voilà...working.

Now I'm 'fighting' with the calibration...python+libraries+win, may be I'll try in linux.
I'll check the wiki for the calibration.

I can't see the 8 point of wiki, this is a minor issue, I think is processing and I'm interested in.

Sorry for the english...

Regards, Jose.

mjs513, 2016-12-15:
There are two things you need to remember using the python calibration program:  1) remember to change the baud rate so it matches your the sketch baud on the Arduino and 2) if you are using the mega change the value of the "word" variable to 2.

closed 2017-01-24 by mjs513