mjs513/FreeIMU-Updates #71: Curie using Linux Compile Error

dvonhausen, 2016-12-18:
On Ubuntu Linux, I'm have the curienano hooked up to my usb port with 
"#define CurieIMU_Mag" uncommented in FreeIMU.h and I get the errors
below when compiling in the Arduino IDE (1.6.12).

In file included from /home/dave/Arduino/libraries/FreeIMU/examples/FreeIMU_serial/FreeIMU_serial.ino:37:0:
Multiple libraries were found for "I2Cdev.h"
/home/dave/Arduino/libraries/FreeIMU/FreeIMU.h:437:24: fatal error: CurieImu.h: No such file or directory
 Used: /home/dave/Arduino/libraries/I2Cdev
   #include "CurieImu.h"
 Not used: /home/dave/Arduino/libraries/MotionDriver
                        ^
compilation terminated.
exit status 1
Error compiling for board Arduino/Genuino 101.

Am I missing something?
Dave

DarinFlippo, 2016-12-21:
I have this same issue.  To work around it, I included the CurieIMU lib from Sketch->Include library->CurieIMU (under Contributed libraries)

mjs513, 2016-12-21:
Are you running the CurieFreeIMU_serial_mag.ino sketch or the FreeIMU_serial.ino sketch.  Should be using the first.

dvonhausen, 2017-01-21:
Problem resolved when applying last comment.

closed 2017-01-21 by dvonhausen