mjs513/FreeIMU-Updates #76: Teensy 3.2 troubles

barbosafs, 2017-01-23:
I am currently using FreeIMU with Arduino Due and it works perfectly. No drift and the angles vary +-0.02 degs only. That's great for me.
I have tried using my Teensy 3.2 with Teensyduino, and I could not manage to get a similar performance. The angles do not lock at all.
What versions of Teensyduino and Arduino IDE would you suggest using? Would it have this hard influence?
Should I modify the code any differently from Due?

Thanks,
Fernando

mjs513, 2017-01-31:
It shouldn't matter.  Haven't used the lib on a 3.2 in a while but from what I remember i use to put a delay of 100ms in the update loop in the main sketch.  You might also have to tweak the kp and ki's a bit to get better convergence on the teensy.

barbosafs, 2017-02-02:
A delay of 100ms? Wouldn't it limit my update frequency to 10Hz?
Besides, you have written this: "02-05-16 Several additional updates had to be to for the 1.6.7 IDE to work nice with the FreeIMU library."
Does this means that 1.6.7 is the newest version it works propperly? I ask this because Teensyduino does not support this version anymore.
One more question: shouldn't kp and ki's depend on the sensors, and not on the board itself?

mjs513, 2017-02-02:
Change it to 10ms which would give you 100hz.  also the changes made apply to Arduino IDE 1.6.7 and above not just for 1.6.7.  Since you are compiling fine for the teensy there is no problem with the teensyduino so that should not be an issue.

Kp and Ki depend on the board.  However, remember that they essentially control feedback correction to the algorithms so as your processor speed changes you may have to adjust the factors to ensure convergence of the algorithm.  You should read the paper by madgwick.  Also, the library supports 4 versions of imu algorithms try a couple of the other ones.

barbosafs, 2017-02-03:
I will give a try soon again. Later I come back for some more comments.
Thanks for the explanations.

closed 2017-03-05 by mjs513