mjs513/FreeIMU-Updates #78: A way to financially support development?

zrecore, 2017-01-31:
Is there a Patreon account, PayPal account, or other means of financially supporting development of this fork of FreeIMU libraries? (Assuming that would be welcomed, of course)

mjs513, 2017-01-31:
Hi Alex.  No there is really no accounts to support further development.  The only reason there is no further develpment at this point is that I am working on another project.  Also, I was not planning on expending any more on different boards.  I have a whole collection as is :).  I think there may be some interest in making it work with IoT boards like the ESP32 etc but I don't have those in my inventory.  If there is anything specific you are looking for let me know.  I am not adverse to generating some updates.  I do work on it periodically.  Right now I am working on an autonomous rover.

Mike

dvonhausen, 2017-02-03:
Mike,

I, for one, would be happy to ship a ESP32 board to you to get this board integrated with FreeIMU.

Dave

mjs513, 2017-02-04:
Dave

Let me ask you a question.  What do you want to do with the esp32 and this library?  Wifi is going to have limited range and incorporating it into the lib is not really within the scope of the intent of the library which is to provide access and learning on IMUs.  ESP32 you could really consider as a communication protocol.

I started looking at interfacing the Arduino to a ESP8266 but to get it set up is a complicated process just to get data to the something like IBM bluemix etc.  The interface would be difficult.

zrecore, 2017-02-04:
...perhaps just ensuring FreeIMU compiles/runs on different architectures
should suffice

dvonhausen, 2017-02-04:
The ESP32 includes both WiFi and Bluetooth. I'm interested in Bluetooth as my application is short range and requires very low power consumption.
I'm currently looking at the freeimu code to see how to get it working properly with ESP32. I'm thinking that if ESP32 is going to be a popular as I think it will be (or is), then it may be better to have one person (who knows what they are doing) integrate it into freeimu rather than everyone who uses ESP32 try to integrate it. I'm hoping that it will not be that hard to integrate but I just started looking at the code.

dvonhausen, 2017-02-04:
Also, I don't seem to be having any problems download arduino code to the ESP32. It seems to work fine on the micro usb connector.

dvonhausen, 2017-02-04:
BTW, I'm using Sparkfun "ESP32 Thing".

mjs513, 2017-02-04:
Have you tried to load the library on to the EPS32 - did it work?

mjs513, 2017-02-04:
By the way, I just compiled it and the only that you have to change in the sketch is to default to #define HAS_EPROM 0 otherwise it compiles fine.

mjs513, 2017-02-05:
Sorry I missed an earlier post from Dave.  I have tried to use BLE from an Arduino 101 to PC and it is almost an impossible task.  There are ways to do it to an android device but I don't really use that so I am not really going to support BLE.  Note BLE does not have an SPP characteristic so you can not just stream the data in the same way.

In looking at the Thing I noticed this statement:

> The Arduino board definitions for the ESP32 are still a work in progress. There are a handful of peripherals and features that have yet to be implemented, including:
> Bluetooth
> Analog Input (analogRead([pin]))
> Analog Ouptut (analogWrite([pin], [value]))
> WiFi Server and WiFI UDP
> Real-Time Clock
> Touch-controller interface

If you are just going to use BT 2.0 then why go with the ESP32 you can just pick up a BT breakout and attach it to the board of your choice.  BT is real easy to implement all you have to do is to change the Serial port your are writing to, i.e., Serial1 for example instead of Serial if you attached the BT to Serial1. The strength of the ESP32 is really going to be the WiFi.

dvonhausen, 2017-02-05:
Thanks for the update.

Since you ordered the board, let me share a problem that I have with mine.
When you run FreeIMU_cube_Odo and you notice the orientation jumps every few seconds, then it may be that int getQ(...), the adruino micros() is overflowing causing this behaviour. To fix this change in getQ(..)
from:
  now = micros();
  sampleFreq = 1.0 / ((now - lastUpdate) / 1000000.0);
to:
  now = millis();
  sampleFreq = 1.0 / ((now - lastUpdate) / 1000.0);

This seems to stop the jumping although may be at the cost of more jitter.

Dave

mjs513, 2017-02-14:
@dvonhausen and @zrecore 

While I was waiting for the ESP32 I decided to break out my linknode d1 and run FreeIMU on it.  I was using the FreeimuYPR sketch as my test case and it seems to run no problem ( I have test some of the other sketches and the Gui with it though).  Out of curiosity I did manage to connect to IBM Bluemix and send the YPR data to the cloud but it only has an update rate of 1 per second.  I even managed to get some graphs made.

Next, which I think is my preferred way is to use UDP to send data to Processing on the PC.  I did run a test case and process works fine.  Now the to do the rest of the testing.  The ESP32 should work the same way but will have different library calls.  If you are interested in UDP here are a couple of links for you:
[https://thearduinoandme.wordpress.com/tutorials/esp8266-send-receive-binary-data/](url)

[https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/udp-examples.md](url)

 I have another project in work so I need to finish that off first.

mjs513, 2017-02-18:
Ok.  I have been testing the ESP8266 with a version of the Processing Cube sketch for FreeIMU and have no problems with stability or sending and receiving data.  The ESP32 Thing on the other hand is causing me all kinds of grief.  Part of the problem I think is because of the immaturity of the of the library or the thing itself.  Don't know.  The UDP to processing works fine one day and the next have problems.  I ordered a different ESP32 dev board and am going to test it.  I will update the library with the ESP8266 sketches for reference shortly

closed 2017-02-18 by mjs513

mjs513, 2017-03-02:
Here is an update.  I could not get the wifi working on the The Thing to save my life.  I just updated the library with three example sketches that I tested on a LinkNode D1, ESP32 dev board from onehorse on tindie and a Microduino WiFi (ESP) module.  I also included two processing sketches that work with WiFi.

zrecore, 2017-03-03:
Hi Mike, would you mind posting a link to the ESP32 board you're using? I'd
like to purchase the exact same hardware and attempt to help sort out the
issues.

- Alex

mjs513, 2017-03-03:
Heres the link Alex

https://www.tindie.com/products/onehorse/esp32-development-board/

This board uses dfu so you have to hold down the boot and reset buttons to load the sketch.  Hot reset after it loads and it should run.  To make it easier with this board I edited the platform.txt to add another esp32 dev board and changed dio to dfu.  I gave the board a different name.  I am going to see if I can get them to officially add the board.

You will need the esp32 arduino library of course

dvonhausen, 2017-03-09:
Thanks for the update (and getting this to work!!!). I plan to get the Tindie board next and try it out.

mjs513, 2017-03-16:
Let me know how it works out for you.

zrecore, 2017-03-16:
Hi folks,

I've just now ordered​ two (2) ESP32 boards through Tindie, and will report
back shortly.

Also, is there a particular IMU breakout to test with the ESP32 board?

- Alex Albino

mjs513, 2017-03-16:
Hi Alex.  I tested it with the Adafruit lsm9ds1 and a 9250 with baro from onehorse. Both worked with the esp32 from onehorse.

By the way I did some modifications to the serial sketch to continously send data.  Also changed the processing sketch accordingly. Also used multiwii instrument and graph capabilities. Still a little buggy at times.

kriswiner, 2017-03-16:
The board will work with any I2C device, and I have tested it using a few
motion sensors and posted sketches here:

https://github.com/kriswiner/ESP32

I am planning a production run since putting these together by hand is
tedious. You guys could help by spreading the word about your experiences
with the board. Thanks.

zrecore, 2017-03-16:
Hi Kris,

May I suggest a LitePlacer machine?

I use one, and it works well enough for low volume production.

- Alex

kriswiner, 2017-03-16:
I use a fab in China where the cost is low and quality high so for a board
that promises even a couple of dozen sales it is a better use of my time to
pay for a 100-unit batch production.

Besides, I like putting them together by hand (it's an art!) it's just I
have other demands on my time and can't afford to do it for even low volume
production.

mjs513, 2017-03-16:
Just saw the new posts. Trying to recover shoveling from the snow storm. Anyway Kris is absouletly correct. Any sensor will work.

mjs513, 2017-03-17:
@kriswiner ..  just added the link on the YouTube video that I have. Its been pretty popular.  Here's the link: https://youtu.be/lwlOZzNJqJc for your reference.