mjs513/FreeIMU-Updates #79: mpu9250

fereshtemohebbi, 2017-02-02:
Dear I am new to processing and mpu9250. I try to use your freeIMU but I receive this error:error: redeclaration of 'iCompass FreeIMU::maghead' 
would you please guide me what should I do

mjs513, 2017-02-02:
Did you remember to configure the Freeimu.h file for the MPU9250.  In other words uncomment the MPU9250 board and make sure all other boards are commented out.

fereshtemohebbi, 2017-02-02:
[image]
yes

mjs513, 2017-02-02:
No not really.  you need to comment out the #define CurieIMU_Mag.  To start I would comment out the #define DISABLE_MAGJAM as well.

fereshtemohebbi, 2017-02-02:
Arduino: 1.6.7 (Windows 7), Board: "Arduino Mega ADK"

In file included from C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4\Zumo32U4Encoders.cpp:4:0:

C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h:214:2: error: #error FastGPIO does not support this board.
 #error FastGPIO does not support this board.

  ^

C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h: In static member function 'static void FastGPIO::Pin<pin>::setOutputLow()':

C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h:228:21: error: 'pinStructs' was not declared in this scope

             _FG_CBI(pinStructs[pin].portAddr, pinStructs[pin].bit);
                     ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h:52:12: note: in definition of macro '_FG_CBI'
     : "I" (mem_addr - __SFR_OFFSET), "I" (bit))
            ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h: In static member function 'static void FastGPIO::Pin<pin>::setOutputHigh()':

C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h:240:21: error: 'pinStructs' was not declared in this scope

             _FG_SBI(pinStructs[pin].portAddr, pinStructs[pin].bit);

                     ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h:50:12: note: in definition of macro '_FG_SBI'

     : "I" (mem_addr - __SFR_OFFSET), "I" (bit))
            ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h: In static member function 'static void FastGPIO::Pin<pin>::setOutputToggle()':

C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h:249:21: error: 'pinStructs' was not declared in this scope

             _FG_SBI(pinStructs[pin].ddrAddr, pinStructs[pin].bit);

                     ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h:50:12: note: in definition of macro '_FG_SBI'

     : "I" (mem_addr - __SFR_OFFSET), "I" (bit))
            ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h: In static member function 'static void FastGPIO::Pin<pin>::setOutput(bool)':
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h:263:21: error: 'pinStructs' was not declared in this scope
             _FG_SBI(pinStructs[pin].ddrAddr, pinStructs[pin].bit);
                     ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h:50:12: note: in definition of macro '_FG_SBI'

     : "I" (mem_addr - __SFR_OFFSET), "I" (bit))
            ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h: In static member function 'static void FastGPIO::Pin<pin>::setOutputValueLow()':

C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h:276:21: error: 'pinStructs' was not declared in this scope
             _FG_CBI(pinStructs[pin].portAddr, pinStructs[pin].bit);
                     ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h:52:12: note: in definition of macro '_FG_CBI'
     : "I" (mem_addr - __SFR_OFFSET), "I" (bit))

            ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h: In static member function 'static void FastGPIO::Pin<pin>::setOutputValueHigh()':
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h:289:21: error: 'pinStructs' was not declared in this scope
             _FG_SBI(pinStructs[pin].portAddr, pinStructs[pin].bit);
                     ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h:50:12: note: in definition of macro '_FG_SBI'
     : "I" (mem_addr - __SFR_OFFSET), "I" (bit))
            ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h: In static member function 'static void FastGPIO::Pin<pin>::setOutputValueToggle()':

C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4/FastGPIO.h:304:21: error: 'pinStructs' was not declared in this scope

             _FG_SBI(pinStructs[pin].pinAddr, pinStructs[pin].bit);
                     ^
[... 73 lines cut ...]
                    ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4\Zumo32U4Encoders.cpp:62:19: note: in expansion of macro 'LEFT_B'
     FastGPIO::Pin<LEFT_B>::setInputPulledUp();
                   ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4\Zumo32U4Encoders.cpp:62:25: error: template argument 1 is invalid
     FastGPIO::Pin<LEFT_B>::setInputPulledUp();
                         ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4\Zumo32U4Encoders.cpp:62:46: error: invalid type in declaration before ';' token
     FastGPIO::Pin<LEFT_B>::setInputPulledUp();
                                              ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4\Zumo32U4Encoders.cpp:9:20: error: the value of 'IO_E2' is not usable in a constant expression
 #define LEFT_B     IO_E2
                   ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4\Zumo32U4Encoders.cpp:80:31: note: in expansion of macro 'LEFT_B'
     lastLeftB = FastGPIO::Pin<LEFT_B>::isInputHigh();
                               ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4\Zumo32U4Encoders.cpp:9:20: note: 'IO_E2' was not declared 'constexpr'
 #define LEFT_B     IO_E2
                    ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4\Zumo32U4Encoders.cpp:62:19: note: in expansion of macro 'LEFT_B'
     FastGPIO::Pin<LEFT_B>::setInputPulledUp();
                   ^
C:\Users\fereshte\Documents\Arduino\libraries\Zumo32U4\Zumo32U4Encoders.cpp:80:37: error: could not convert template argument 'IO_E2' to 'unsigned char'
     lastLeftB = FastGPIO::Pin<LEFT_B>::isInputHigh();
Sorry for inconvenience, I comment both of them but I receieve these errors could you please let me what should I have to do

fereshtemohebbi, 2017-02-02:
I remove all zumo arduino I receive the other error I try to solve the.just to inform you that please do not waste your time for previous error thanks

fereshtemohebbi, 2017-02-02:
/MPU Address Select 
  //Use following define if MPU60X0 address is set to 0x69
  //otherwise the default address is used = 0x68
  //uncomment following line and comment out successor line
  //#define FIMU_ACCGYRO_ADDR MPU60X0_ADDRESS_AD0_HIGH

i dont understand what should I have to do exactly with this comments,

mjs513, 2017-02-02:
Don't worry about it now.  Typically the default address for the 9250 is set to 0x68 which the library assumes.  Go ahead and run it

fereshtemohebbi, 2017-02-03:
I have question about processing. I have to draw cube for each of accelerometer, gyro and magnetometer .Could you please guide me about the formula for rotation. I have no idea for z axis would please guide me regarding this issue.so many thanks

 int x_rotation = 90; 
  //Show gyro data
  pushMatrix(); 
  translate(width/6, height/2, -50); 
  rotateX(radians(-x_gyr - x_rotation));
  rotateY(radians(-y_gyr));
  draw_rect(249, 250, 50);
  popMatrix(); 
  //Show accel data
  pushMatrix();
  translate(width/2, height/2, -50);
  rotateX(radians(-x_acc - x_rotation));
  rotateY(radians(-y_acc));
  draw_rect(56, 140, 206);
  popMatrix();
  //Show gyro data
  pushMatrix();
  translate(5*width/6, height/2, -50);
  rotateX(radians(-x_gyro - x_rotation));
  rotateY(radians(-y_gyro));
  draw_rect(93, 175, 83);
  popMatrix();

mjs513, 2017-02-04:
Would suggest you read the processing web site - lots of information.  Remember in that the screen z-axis comes out of the screen - y is up and x is the right. If I remember correctly.

fereshtemohebbi, 2017-02-09:
Hi , I run processing V5 from experimental folder but I get this error :Caused by: java.lang.NoClassDefFoundError: processing/core/PGraphicsJava2D  how should I solve that

fereshtemohebbi, 2017-02-14:
Dear 
I try to use the calibration application on windows and exactly I installed the mentioned URL in wiki page of calibration. and also change the word to 2(I use arduino mega) and change both IMU serial and python library to 57600. First i got below error 
:Traceback (most recent call last):
    |    File "C:\Python27\lib\site-packages\pyqtgraph\opengl\GLViewWidget.py", line 180, in drawItemTree
    |      i.paint()
    |    File "C:\Python27\lib\site-packages\pyqtgraph\opengl\items\GLScatterPlotItem.py", line 149, in paint
    |      glDrawArrays(GL_POINTS, 0, int(pos.size / pos.shape[-1]))
    |  AttributeError: 'list' object has no attribute 'size'

Also  H = numpy.array([x, y, z, -y**2, -z**2, numpy.ones([len(x)])])
I change the pos in GLscatteritem to  self.pos = np.array([0,3]) and I dont have error right now but after that when I connect the serial port the application crashes I just receive the connection successful and waitint to rest arduino but gui crashes. What should I have to do ?
I installed every thing in 32 bit.Could you please guide me

mjs513, 2017-02-14:
Hi fereshtemohebbi 

You can ignore the traceback errors the python script will run fine with that particular error.  I have not been able to figure out whats causing it.  Fabio put it together so I have not dug into the guts.

As for the numpy error I thought I fixed it.  The numpy error will call the crash.  It should read:

H = numpy.array([x, y, z, -y**2, -z**2, numpy.ones([len(x)])])

Did you remember to install the numpy and scipy libraries for python?

fereshtemohebbi, 2017-02-14:
Yes I have installed numpy and acipy,I use your GUI which you have modified the code .Moreover I ignore the error but I enter the COM port and I receive the message "Arduino serial port opened correctly
" but the application crashes

mjs513, 2017-02-14:
Can you tell me what does it mean that it crashes. Did you click on start sampling

fereshtemohebbi, 2017-02-14:
The application is not responding and I can do nothing. the application freezes

mjs513, 2017-02-14:
After you close in go into task manager and make sure you do not have any other pythonw.exe files running. Sometimes python does not end cleanly and leaves behind old .exe files running.  End those tasks and then give it a try again

fereshtemohebbi, 2017-02-14:
[image]

I change my system to ubunto, with 57600 it doesnt connect and I chnage both serial 115200. but as you see it doesnt draw any thing

mjs513, 2017-02-14:
Getting closer. I think I mention the following in the wiki. If you put the mouse cursor in any of the windows you can use the scroll wheel to zoom in and out on the frame that it's in. If you hold the the left mouse button down while you are in one of the windows you can drag it left and right, up and down and even rotate the sphere. I will usually just use the zoom out feature.

Remember to stop sampling before you hit calibrate.

fereshtemohebbi, 2017-02-15:
Dear i read that and I also use that to check but it doesnt draw any thing.I decided to just rotate for calculation of off-set and scale. the magnetometer off-set is 0 always and scale in some axes are nan below are the errors
[image]

[image]

mjs513, 2017-03-05:
@fereshtemohebbi I apologize for not getting back to you on this I just saw it.  I hooked up my 9250 and I have no problems reading the data from the IMU.  If you haven't given up let's try to sort this out starting from the beginning.  
1. What microcontroller are you using?  Did you hook everything up correctly?  Easy way to verify is after you uploaded the FreeIMU_serial.ino to the microcontroller open the serial window at the correct baud rate (defaults to 57600).  Type 'v' in the serial window command line and hit the send key.  If the IMU is ready you will see a message returned starting with FreeIMU and info on the board you selected.  Repeat a few times until you see the message.  The type 'r' in the command line and hit send again you should see the raw IMU readings.  The first three values are the acceleration in the x, y, z direction.  The three are the gyro and the next three are the magnetometer.  If the IMU is working correctly the az should be on the order of -16535 (this is the third value) and the values should not be constant.  If they are then it is probably not connected correctly.
2.  If the values look reasonable check the GUI baud rate to make sure its set to 57600 (as long as it matches what the sketch baud rate you set.  Check the word value is 2 if you are using a mega type board or 4 if you are using a teensy or due type board.  Let me know.
3.  Make sure you the com port you selected on the top line of the GUI is same as the com port for your board.
4.  Make sure you hit connect and then start sampling after your board connects.

Let me know.

fereshtemohebbi, 2017-03-09:
dear I was successful to use the gui. I use free IMU cube.My problem is that after calibration, when I press "H", the cube has o much drift, by doing again and again home positioning I see that the cube has less than a degree drift,Do you have any idea whats is the problem? My calibration values are provided below, Do they seem fine to you?Do you have another solution to check my sensor drift and values?
const int acc_off_x = 264;
const int acc_off_y = 74;
const int acc_off_z = -946;
const float acc_scale_x = 16389.884034;
const float acc_scale_y = 16447.537636;
const float acc_scale_z = 16766.444503;

const int magn_off_x = 251;
const int magn_off_y = -183;
const int magn_off_z = -114;
const float magn_scale_x = 225.463511;
const float magn_scale_y = 219.277990;
const float magn_scale_z = 221.115866;

mjs513, 2017-03-09:
The calibration values look reasonable for a 9250.  Would really need to see a screenshot of the raw calibration from the cal gui. 

How much drift you get over a time, i.e. degrees/ minute?

You might have to tweak the kp/ki values? Recalibrate getting more data? Also, which option are you using 0 and 4 seem to work best for a 9250?

You might want to see the discussion going on with this issue:  [https://github.com/mjs513/FreeIMU-Updates/issues/81#issuecomment-285281713](https://github.com/mjs513/FreeIMU-Updates/issues/81#issuecomment-285281713)

fereshtemohebbi, 2017-03-14:
thanks for your advice, I change it to 4 and it seems that it is better than the others. I connect two mpu-9250 I get the raw values and every thing from both of my sensors.But the problem is that after less than a minute I get nan value for q  for both of my sensors,but when i change it to 3 I get the q values'.I have to say that when I get nan values for q< ichecked the raw values and I have them

mjs513, 2017-03-14:
Yeah I know with 4.  Hit the 'r' key and it resets everything and then works fine.  Haven't been able to trace the problem yet

closed 2017-08-15 by mjs513

JulesThuillier, 2017-09-27:
@fereshtemohebbi I am encountering the same issue you did (15 Feb) : magnetometer offset at 0. How did you managed to fix that please ?

Thanks !!!