mjs513/FreeIMU-Updates #80: MPU9250_5611 drifts

allenyin, 2017-03-03:
Hi,

I'm using the your libraries with a MPU9250_5611 sensor to read the Euler angles (as ypr) with the getEuler() function. Before I start taking the measurements, I have gone through the calibration step with the cal_gui utility -- the process went fine and I see a pretty reasonable ellipsoid calibration window.

While testing, with the MARG option in FreeIMU.h set to anything but the "Madgewick original AHRS implementation", the yaw- the first euler angle value, would change briefly when the sensor is moved and return immediately into some value.

With the MARG option set to "Madgewick original AHRS implementation", the yaw euler angle reading would change as expected. However, all three readings would experience a drift when the sensor is not being moved. These drifts are continuous (10min is the longest I have measured), and is significant -- around 5 degrees/minute.

I am not familiar with the actual computation of the update algorithms. But theoretically incorporating the magnetometer in the sensor fusion algorithms is precisely to eliminate the drifts, right? So if the MPU9250 has a magnetometer built in, what can be causing this drift? 

I want to use the IMU to track head direction, so yaw is the most important measurement. Would I be better off to use simply the magnetometer reading?

mjs513, 2017-03-03:
Have you tried Magdwicks Gradient Descent or DCM option?  The other thing is that controls the drift is are the Kp/Ki factors used for the sensor.  While I tied to get them to zero drift you may have to tweak them a bit more.  In the wiki I tried to provide you with how to do that.  As for the drift the gyros play a big part in that as well.  Try using the 'g' command to zero out the gyro values again.  It does it on start up for you.  One of things you need to be sure of is that on plugging the sensor in you hold it still while it zeros itself out.

allenyin, 2017-03-03:
Yes I tried all other options, only the "Madgewick original AHRS" gives me a drift that's not super fast.

The procedures you described is for the calibration procedure right? 

Also, once calibrated, is it necessary to re-calibrate the sensor every time, if the experiment conditions stay the same?

mjs513, 2017-03-04:
That's kind of strange.  After you did the calibration did you remember to save the calibration.h file to the freeimu directory from the gui.  

The procedure is correct, used it many times.

You should only have to calibrate once.  I usually do it once a year.

allenyin, 2017-03-04:
I saved the calibration.h to the folder where my .ino file is and made sure that copy is included during compilation.

I'm going through tweaking the Kp/Ki and beta values for the sensors right now, referring to the wiki. A few questions though -- while tweaking the Kp/Ki values, what do you set the initial beta value to? The wiki suggests that small beta value results in drift. And since the drift is likely a result of integration error, would something like Kp=2, Ki=0.1, and beta=0.1 be a good starting point?

For DCM, there are extra parameters Kp/Ki_ROLLPITCH, Kp/Ki_Yaw...briefly how are they related to the Kp and Ki values?

mjs513, 2017-03-04:
Wrong folder.  You need to put the calibration file in the FreeIMU library folder.  Just copy and paste it there and try again before you start adjusting the Kp/Ki factors.  Check the wiki there is some guidance there.

allenyin, 2017-03-04:
I modified the calibration.h so it's only included once, and it is called before the include statement for FreeIMU.h

Regardless, I used the same calibration.h in the FreeIMU/ folder. I still see the same effects.

Here is my calibration.h file btw, do the values look reasonable?

```
#define CALIBRATION_H

const int acc_off_x = 1653;
const int acc_off_y = 77; 
const int acc_off_z = -2166;
const float acc_scale_x = 11414.752253;
const float acc_scale_y = 18772.019556;
const float acc_scale_z = 17989.572536;

const int magn_off_x = 2;
const int magn_off_y = -100;
const int magn_off_z = -58;
const float magn_scale_x = 204.724280;
const float magn_scale_y = 209.970326;
const float magn_scale_z = 221.706599;
```

mjs513, 2017-03-04:
Actually your calibration seems a bit strange for the acceleration cal.  Using a 9250-5611 this is my cal file:
```
#define CALIBRATION_H

const int acc_off_x = 269;
const int acc_off_y = 99;
const int acc_off_z = 251;
const float acc_scale_x = 16287.221651;
const float acc_scale_y = 16494.574349;
const float acc_scale_z = 16586.496981;

const int magn_off_x = 60;
const int magn_off_y = 62;
const int magn_off_z = -247;
const float magn_scale_x = 239.261959;
const float magn_scale_y = 244.822201;
const float magn_scale_z = 253.262444;
```

Here are a couple of screen shots from the GUI:
[image]
[image]

allenyin, 2017-03-04:
Ok I have re-calibrated my sensors (still have acc_off_x on the order of 1000). Using the Madgewick Gradient Descent. According to the Wiki (and looking at the `MadgeWickAHRSupdate()` function, the only tunable parameter is `betadef`.

With `betadef` greater than 1.5, I am now getting reasonable Euler angle readings.

One question about the behaviour of the gradient descent method -- when I rotate the sensor around its z-axis between two different orientations, I see an overshoot in the yaw-measurement correlated to how fast I'm turning the sensor (see below):

Overshoot when turning sharply
[image]

No overshoot when turning slowly
[image]

This overshoot is less severe for roll and pitch. I'm guessing this is due to the sudden stop creating high accelerometer readings that gets summed into the sensor fusion?

Thanks for all the help!

mjs513, 2017-03-05:
Interesting. I have been testing my 9250 and for option 1 I see a lot of jitter when at rest and nice and smooth when moving. For opt 2 it's rock steady with no jitter and smooth. By the what processor are you using

allenyin, 2017-03-05:
I am using the [femtobeacon](https://github.com/femtoduino/femto-beacon) board, which uses the Atmel SAM R21 (ARM Cortex M0+, SAM D21 with built in AT86RF233).

So with option 1, you are not seeing the overshoot at all?

I will also try option 2. The tuning parameters for that option are just Kp and Ki, right?

mjs513, 2017-03-05:
Option 0 gives me better results that option 1 with the settings that I have.  I am using an Arduino 101 for testing.  I don't have a femtobeacon board, to test it on.

mjs513, 2017-03-06:
Are you using the fork from the fetobeacon website?  or this fork.

allenyin, 2017-03-06:
I'm using the femtobeacon fork. But as far as I can tell, the only changes were the #include directive for the calibration.h files.

mjs513, 2017-03-06:
Ok thanks. Just wanted to know which verdion.  There were other changes but it should nm it affect the algorithm

allenyin, 2017-03-07:
Ok, I've been tweaking with Option 0 -- Mahoney implementation of Madgewick.

Interestingly, when I set Kp and Ki both equal to 0, I get pretty good readings (better than all the other combinations I have tried). But the yaw still has about 5 degrees/min drift.

I'm guessing this is calibration-related?

mjs513, 2017-03-07:
I would think so or there is something wrong with the sensor on the board.  Like I said I don't have the board you are using so I can't tell you for sure.

mjs513, 2017-03-07:
Try option 4 as well as a test.

allenyin, 2017-03-11:
So I tried reading the values with `getYawPitchRoll180()`, the readings do not have any drift for any of the fusion algorithms, and decided on option 1 (gradient descent) since there's only one parameter to tweak.

I'm not sure if it makes sense that the readings from `getEuler()` and other euler angle methods should have a drift..get the `getYawPitchRoll180()` etc seems to give consistent results.

However, the overshoot is problem is still there (as mentioned a few posts ago) during sharp turning of the sensor-board. Another screen shot:

[image]

I wanted to see if it's the IMU's problem, so I plotted the magnetometer's calibrated readings as well (from `getValues()`, val[6], val[7], val[8]).

[image]

The magnetometer readings don't have the overshoots and slow settling time. I figure this probably means it's something in the sensor fusion. I have tried to tweak the `betaDef` value. Increasing that value seem to create the little wiggles in the pitch and roll readings as in the first figure, but doesn't seem to have much effect on the overshoots. Increasing `betaDef` even further results in nonsensical readings..as expected of too much proportional gain.

Looking through the `MadgewickAHRSUpdate()` method, there doesn't seem to be any other tunable parameters...perhaps I will do a post-filtering of the angle readings..?

------------
On another note, would you recommend trying instead the Kalman filter included in FreeIMU-Updates? Have you seen any advantages of the KF over the options in FreeIMU.h?

mjs513, 2017-03-11:
To be honest I haven't done many comparisons to the Kalman filter but you can give it a try.  The overshoot would be more visible with accel and gyro since those readings change as you move the board.  The magnetometer is measuring the magnetic field in your location so I would not expect to see the same type of behavior.

allenyin, 2017-03-12:
> The magnetometer is measuring the magnetic field in your location so I would not expect to see the same type of behavior.

That's what I thought as well, so I tried to confirm that by looking at `vals[6]`, `vals[7]`, `vals[8]` from `getRawValues()`. Turns out the raw magnetometer readings were overshooting as well, and the reason I don't see the overshoots from the outputs of `getValues()` is probably because those outputs are low-passed and compressed into a smaller range.

Not sure if the overshoots in raw magnetometer reading is supposed to happen, but that all makes sense now.

Thanks for all the help!

closed 2017-03-12 by allenyin

mjs513, 2017-03-12:
Interesting.  Haven't really seen that with the mag will have to check.  You might want to check the last GUI I finished putting together.  I uses a multiwii style display and you can see the plots within processing.