mjs513/FreeIMU-Updates #81: FreeIMU_serial and Cube vs FreeIMU_yaw_pitch_roll

s0bes, 2017-03-04:
Hi, thanks for the work you've done.I'm digging into it for 3rd day now. Sorry if the question is silly.
I have MPU6050 + HMC5883L (GY-88 board) with arduino mega 2560. I successfully calibrated my board using cal_gui.py and when I run FreeIMU_cube_Odo.pde everything is smooth and just as it should be. no drift at all.
But when I upload FreeIMU_yaw_pitch_roll.ino and try to get ypr values, they are changing over time a lot. Yaw especially.  
Please show me right direction regarding my problem. 

Appreciated it. thanks

s0bes, 2017-03-04:
update.
seems like drift also happens in cube demo as well. It depends on ... i don't know.. how you move sensor. Sometimes you move it and everything remains steady when you stop moving. and sometimes it will continue drifting. i don't know. right now I'm running for 10 minutes sketch which puts euler angles in serial port. in first 2 minutes yaw was drifting. then i moved sensor a bit and right now for 8 mins it seems to be steady af.
I will try tomorrow better calibration and may be another board (i have 2 gy_88).

But I still will be glad if you could help me with an advice.
and also. please clarify to me one thing. After calibration can I just use my3IMU.getEuler360degAttitude(ypr, q, val);
to get angels? Will these angels be like filtered ones? Like in cube sample? Cause I saw cube code and there are a lot of functions which seemed to me are doing some calculations/calibrations/ to get good data from sensors. 

thank you anyway though

mjs513, 2017-03-04:
Easy question first: yes you can call that function to get the euler angles.  Just remember you will need arrays set up in your sketch fro ypr, q and val.  Should already be set up in one of the sketches.  The Angles will be based on the filter selected.

For some reason the cube code recalculates all the angles even though they are calculated in the library.  The library passes q and the normalized/calibrated sensor values and the cube code does the rest.  Guess its one way to reduce size of code in the library.

On start up you are going to see some drifting until it gets to its rest position - since the filter is iterative the final value is asymptotic.

s0bes, 2017-03-07:
It takes 3 minutes after starting up to get to the final non drifting values
is it normal?

mjs513, 2017-03-07:
3 minutes is way too long.  30secs is about right.  Since you are using a 6050 you need keep it flat during start up so it has time to zero out its registers.  There is also a zero gyro routine that requires you to keep it still during startup.  If you move it during start up you will get some drifting.  Put 'g' in the command line of the serial monitor when you see drifitng it will cause the gyros to zero out again.  If that doesn't improve it you can try a few other things: (1) try a different option - 0 and 4 seem to work best, (2) recalibrate and try again.   Try these two first.

s0bes, 2017-03-09:
thank you once again. I tried your suggestions. and I got it working finally
I tried a lot of commands and different variations of them in a different order. And in my case the best result in yaw drifting was 1deg per 1.5-2.5 min (pitch and roll are always the same.no drift at all). Other combinations led me to 3deg per minute or worse.

so I did reset, zero gyros, zero quanterions, disable temp compensation
1->g->2-t
it doesn't work for me at all if I remove resetting mpu and disabling temp compensation on start up. As I said, the same things but in different order shows me yaw drift 3deg per min.
And now it takes only 20 sec to normalize ypr values.
thanks)

s0bes, 2017-03-09:
thank you once again. I tried your suggestions. and I got it working finally
I tried a lot of commands and different variations of them in a different order. And in my case the best result in yaw drifting was 1deg per 1.5-2.5 min (pitch and roll are always the same.no drift at all). Other combinations led me to 3deg per minute or worse.

so I did reset, zero gyros, zero quanterions, disable temp compensation
1->g->2->t
it doesn't work for me at all if I remove resetting mpu and disabling temp compensation on start up. As I said, the same things but in different order shows me yaw drift 3deg per min.
And now it takes only 20 sec to normalize ypr values.
thanks)

s0bes, 2017-03-09:
and with this code it takes only 5 sec to normalize ypr values
```
void setup() {
  Serial.begin(115200);
  Wire.begin();  
  delay(5);
  my3IMU.init(); 
  delay(5);
  my3IMU.init(true);
  my3IMU.initGyros();
  my3IMU.RESET_Q(); 
  my3IMU.setTempCalib(0);
}
```

mjs513, 2017-03-09:
I have see that but can't remember the magnitude.  Once you pick the option you like you can start tweaking the Kp/Ki or betadef, etc to see if you can get it better.

If you need steadier readings you can always try a complimentary filter on each ypr value.

mjs513, 2017-03-12:
Also, you can set up a complimentary filter on the ypr so the swing is reduced further or you can only print values every 10 iterations of the algorithm.

closed 2017-04-12 by mjs513