mjs513/FreeIMU-Updates #82: Cannot find a class or type named "PSurface"

scottkobe0221, 2017-03-09:
Hi,
First of all thanks for all the effort maintaining this.
I ran into an issue while trying to compile the "FreeIMU_cube_Odo_Exp_v6" code using Processing.
I followed the wiki and changed the serialPort and BaudRate to the correct settings. But when I run the code, the issue "cannot find a class or type named "PSurface"" popped up on this line:
"static final void removeExitEvent(final PSurface surf)"

Can you please help? Appreciate your time.

mjs513, 2017-03-10:
I can not duplicate this one.  I did find this post: https://forum.processing.org/two/discussion/19860/trouble-with-window-size-when-using-secondapplet on the processing forum.  

What version of processing are you using?

scottkobe0221, 2017-03-10:
I am using Processing 2.0B7. 
I also tried other versions of the Experiment folder and found out that the issue exists from v4 to v6. v3 and down do not have this issue. Thanks.

mjs513, 2017-03-11:
That's the problem.  Because of the differences between 3.0 and two there are were major changes to processing code especially with the way the Gwoptics works.  Windows disappeared so the new method won't be compatible with old.  That is why you were getting the error you got with psurface.

closed 2017-03-11 by mjs513