mjs513/FreeIMU-Updates #83: Using 2 MPU9250s

scottkobe0221, 2017-03-21:
Hi,
I am exploring the option of using 2 MPU9250s simultaneously with a Arduino Mega 9260. As far as I understand, this option is possible if I have different I2C addresses for the MPU9250s.
My plan is to use the first MPU9250 with AD0 pin low, this will give me 0x68 as I2C address, and the 2nd MPU9250 with AD0 high, which will give me 0x69. I have the following question:
1. In the FreeIMU.h file, I can select the MPU9250 I2C address based on AD0. Is there a way to enable both 0x68 and 0x69? Here is the portion of the code:
"#elif HAS_MPU9250()
  #include <Wire.h>
  #include "I2Cdev.h"
  #include "MPU60X0.h"
  #include "AK8963.h"
  #include "iCompass.h"
  //MPU Address Select 
  //Use following define if MPU60X0 address is set to 0x69
  //otherwise the default address is used = 0x68
  //uncomment following line and comment out successor line
  //#define FIMU_ACCGYRO_ADDR MPU60X0_ADDRESS_AD0_HIGH
  #define FIMU_ACCGYRO_ADDR MPU60X0_DEFAULT_ADDRESS"

Currently it is at default address, which is 0x68. Is it as simple as having both uncommented out?

My plan for each of the MPU9250:
1st MPU9250: Use accelerometer data only (no mag data)
2nd MPU9250: Use accelerometer and mag data

So I guess my question is essentially if I can use both boards at the same time but pick and choose which board I want the data from. 

Thans a lot for the help!

mjs513, 2017-03-21:
Right now the library doesn't support multiple IMUs.  It's a little more complicated since the i2cdevlib is only really configured for one IMU.  This was on Fabio's todo list and I guess I will put it on mine.  I can't tell you when it would be done thought.

You could used the original code for the i2cdevlib:  [https://github.com/jrowberg/i2cdevlib/issues/63](https://github.com/jrowberg/i2cdevlib/issues/63)

You can check this post out as well. [https://www.i2cdevlib.com/forums/topic/15-wire-vs-wire1/](https://www.i2cdevlib.com/forums/topic/15-wire-vs-wire1/)

mjs513, 2017-03-22:
I think there may be a way to do this.  I just ordered a second 9250 and will give it a try.  If it works I will post an update.  I will leave this open for now.

scottkobe0221, 2017-03-22:
Thank you very much for giving a try!

fereshtemohebbi, 2017-03-24:
Hi I am using two mpu9250 the thing is that I can read acc and gyroscope correctly.But by testing I understand that I am reading the first sensor (0x68) data. Could you please guide me how can I change the library to switch off the first magnetometr and activate second on.as far as I know I have to use INT pin cfg 
but I couldnt 
thanks

mjs513, 2017-03-24:
You would have to create two calls to init like this:
mpu92501 = init(int accgyro_addr, false);
mpu92502 = init(int accgyro_addr, false);

This does not take into account the mags which is why I said that I have some testing to do.

fereshtemohebbi, 2017-03-24:
I have done some thing like this :   
FreeIMU my3IMU = FreeIMU(0x68);
FreeIMU my3IMU1 = FreeIMU(0x69);

my3IMU.init(0x68,false);
 my3IMU1.init(0x69,false);
But the thing is that, when I physically disconnect the second sensor, The value for acc and gyr are zero but the I see the value for mag:Below result is related to raw value of second sensor when it is physically disconnected 
0,0,0,0,0,0,50,-242,-246,0,10872,
0,0,0,0,0,0,38,-232,-254,0,10886,
The other thing That I have checked is that I change the code for processing to support two sensor and I see that when I move one sensor the other sensor rotates too.Can I ask when it is possible to check that sorry it is so important for me

mjs513, 2017-03-25:
That's because the mag is read separately from the acc and gyro so you still get the data.  Please try you init this way:

FreeIMU my3IMU;
FreeIMU my3IMU1;

then in the setup keep what you have.  Still waiting for the second 9250 so it all depends when I receive it.  Like I said there is some changes that have to made but I won't start that until I get the second sensor.  So why is this so important to get it done so fast?  Is

fereshtemohebbi, 2017-03-25:
Be cause mine is mpu9250 I change a bit in both freeimu.cpp and MPU60x0:
in FreeIMU:
FreeIMU::FreeIMU(uint8_t address) {
***
elif HAS_MPU9250()
    accgyro = MPU60X0(address);
    mag = AK8963();
    maghead = iCompass(MAG_DEC, WINDOW_SIZE, SAMPLE_SIZE);
}
and in MPU60x0 file :  
MPU60X0::MPU60X0(uint8_t address) {
        devAddr = address;
    }
also base on kriswiner Each MPU9250 has an internal switch to allow mag acces from the host. Just
turn this switch off for the MPU9250 with I2C address 0x68 and you can read
the mag from the MPU9250 with address 0x69.
// Configure Interrupts and Bypass Enable
// Set interrupt pin active high, push-pull, hold interrupt pin level HIGH
until interrupt cleared,
// clear on read of INT_STATUS, and enable I2C_BYPASS_EN so additional chips
// can join the I2C bus and all can be controlled by the Arduino as master
writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x22);
set INT_PIN_CFG to 0x20 to turn off switch.
I do this in MPU9250 initialize function but it doesnt work. 
I have deadline for my project and I put lots of time on it and gather data and then found that the data for magnetometer is not correct
thanks

mjs513, 2017-03-25:
AD0 is not a software switch.  If the boards AD0 pin is tied to ground the MPU9250 would have a address of 0x68, it tied to 3.3v its considered high and would set the address to 0x69.  As far as I can see there is no way to change the address of the AK8963 (magnetometer).  I believe what the int_pin_cfg register is doing is enabling or disabling the pass through mode of the magnetometer so you and either read it or not read it directly.  It is not changing the address of the magnetometer.  Suggest that you read the MPU9250 register map and Spec pdf's to get a better understanding.

mjs513, 2017-03-25:
You may want to read this thread over at Kris's site: [https://github.com/kriswiner/MPU-9250/issues/68](https://github.com/kriswiner/MPU-9250/issues/68)

fereshtemohebbi, 2017-03-25:
> kriswiner/MPU-9250#68 <https://github.com/kriswiner/MPU-9250/issues/68>

Yes i read them but i wasn't able to solve my problem with one i2c bus i am
going to read again register data sheet but if it is possible please let me
know when you do it or find a solution

mjs513, 2017-03-25:
Looks like you would have to change from using pass through mode to the MPU9250s i2c master mode.  Check this out [http://stackoverflow.com/questions/24677234/issues-accessing-magnetometer-data-from-mpu9150](http://stackoverflow.com/questions/24677234/issues-accessing-magnetometer-data-from-mpu9150)

mjs513, 2017-03-25:
This is the solution but need to incorporate it [https://developer.mbed.org/questions/6482/Configuring-MPU9250-magnometer/](https://developer.mbed.org/questions/6482/Configuring-MPU9250-magnometer/)

fereshtemohebbi, 2017-03-27:
I tried two different way, the one with   http://stackoverflow.com/questions/24677234/issues-accessing-magnetometer-data-from-mpu9150 and the other one with master method and the other one with the kries wiener solution but I had no success. some times it works some times again it reads from the mag data.Please let me know your idea or when you start it.

mjs513, 2017-03-27:
Been playing around with it but without much luck.  Think I am missing something.  Anyway you might want to check this out [https://forum.pjrc.com/threads/37891-MPU-9250-Teensy-Library](https://forum.pjrc.com/threads/37891-MPU-9250-Teensy-Library)

mjs513, 2017-03-31:
Just by way of an update.  After a few days of trying different things got master mode working with Brian's help.  I will post the changes with a sketch in a couple days.  Had a devil of time getting it work.  Tried each option but finally works.  Looks promising for multiple 9250s.  Only left to solve is the issue of calibration.

scottkobe0221, 2017-04-01:
that's awesome! Thank you so much for spending the time on this!

mjs513, 2017-04-01:
Thank Bryan Taylor over at the Teensy forum who developed the Teensy 9250 library.  I think eventually I may just convert over to using it.

Anyway, I updated the library files FreeIMU.h and FreeIMU.cpp to use master mode for the 9250 only.  There is no support for multiple buses or multiple calibration files yet.  But I want to get this out to you all to try out with two MPUs.

In any sketch that you use you need:
1. Establish two instances of the library:
// Set the FreeIMU object
FreeIMU my3IMU0 = FreeIMU();
FreeIMU my2IMU1 = FreeIMU();

2. in the setup initialize the 9250s with the correct addresses
```
my3IMU0.init(0x68, true);
my3IMY1.init(0x69, true);
```
When you raw data you will need to make sure you identify the data to the correct IMU object:
```
my3IMU0.getRawValues(raw_values0);
my3IMU1.getRawValues(raw_values1);
```
You will need to setup two sets of raw values.

Let me know how it works out.

fereshtemohebbi, 2017-04-01:
Dear I think you forget to upload MPU60X0 new file which I think have the configuration of master support
tnx

mjs513, 2017-04-01:
You are correct - forgot those two.  Just updated so they should be there.

mjs513, 2017-04-09:
Ok.  Officially released the lib with master mode support.  If you all have a example of visualization using two MPU's let me know. 

Thanks
Mike

closed 2017-04-09 by mjs513

scottkobe0221, 2017-04-13:
Hi Mike,

I tried your codes. And after a little tweaking, they work for my purpose
perfectly. I just want to express my appreciation to your time and effort
on this. Thank you very much!

Scott Lee

mjs513, 2017-04-13:
Hey Scott

No problem.  Glad its working for you.  Any tweaks you want me to incorporate let me know.  Good luck with your project.

Mike