mjs513/FreeIMU-Updates #85: Wrong use of Wire.requestFrom

Koepel, 2017-04-18:
The Wire.beginTransmission and Wire.endTransmission should not be used with Wire.requestFrom, they are only used when writing data.

The Wire.requestFrom executes a complete I2C transaction on its own. When the Wire.requestFrom returns, the I2C transaction has completely finished and the received data is in a buffer in the Wire library. It is therefor also not needed to wait for something.

Unnecessary waiting is in most cases not harmful, but in case of a bus collision or a hardware bus problem it can be.
An unnecessary Wire.beginTransmission before Wire.requestFrom has no effect for AVR chips.
An unnecessary Wire.endTransmission after Wire.requestFrom generates an extra I2C bus transaction for AVR chips because the address of the sensor is put on the I2C bus and the sensor acknowledges that.

This means that the I2Cdev library is slower for no reason, and the bma180 functions could return a wrong return value for non-AVR chips.

The wrong use of Wire.requestFrom is in:
- HMC58X3.cpp The beginTransmission and endTransmission around requestFrom should be removed (two times).
- I2Cdev.cpp Only where the Arduino Wire library is used. The beginTransmission, the timeout and the endTransmission around the requestFrom should be removed (two times).
- ADXL345.cpp The beginTransmission and endTransmission around requestFrom should be removed (once).
- ADXL345.cpp
- bma180.cpp The endTransmission after requestFrom should be removed (three times). The return value is unpredictable.
- BMP085.cpp The beginTransmission and endTransmission around the requestFrom should be removed (once).
- ITG3200.cpp The beginTransmission and endTransmission around the requestFrom should be removed (once).
-

mjs513, 2017-04-19:
Hi Koepel.   I saw your comment to zrecore as well.  Thanks for the information.  The libraries that you are referencing were downloaded from other sources so there may be other challenges as well.  I will take a look at and incorporate for the set of revisions.

mjs513, 2017-04-26:
Just pushed the updates.  Thanks for pointing these changes out.

closed 2017-04-26 by mjs513