mjs513/FreeIMU-Updates #86: Problem with compiling the example sketches

Gabande, 2017-04-21:
```
C:\Users\Benedikt\Documents\Arduino\libraries\AP_Math_freeimu/AP_Math_freeimu.h:32:0: warning: "RAD_TO_DEG" redefined

 #define RAD_TO_DEG 57.295779513082320876798154814105f

 ^

In file included from sketch\FreeIMU_yaw_pitch_roll.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:51:0: note: this is the location of the previous definition

 #define RAD_TO_DEG 57.295779513082320876798154814105

 ^

C:\Users\Benedikt\Documents\Arduino\libraries\MS561101BA\MS561101BA.cpp: In member function 'float MS561101BA::getPressure(uint8_t)':

C:\Users\Benedikt\Documents\Arduino\libraries\MS561101BA\MS561101BA.cpp:72:18: warning: NULL used in arithmetic [-Wpointer-arith]

   if(rawPress == NULL) {

                  ^

C:\Users\Benedikt\Documents\Arduino\libraries\MS561101BA\MS561101BA.cpp:73:12: warning: converting to non-pointer type 'float' from NULL [-Wconversion-null]

     return NULL;

            ^

C:\Users\Benedikt\Documents\Arduino\libraries\MS561101BA\MS561101BA.cpp:77:12: warning: NULL used in arithmetic [-Wpointer-arith]

   if(dT == NULL) {

            ^

C:\Users\Benedikt\Documents\Arduino\libraries\MS561101BA\MS561101BA.cpp:78:12: warning: converting to non-pointer type 'float' from NULL [-Wconversion-null]

     return NULL;

            ^

C:\Users\Benedikt\Documents\Arduino\libraries\MS561101BA\MS561101BA.cpp: In member function 'float MS561101BA::getTemperature(uint8_t)':

C:\Users\Benedikt\Documents\Arduino\libraries\MS561101BA\MS561101BA.cpp:90:12: warning: NULL used in arithmetic [-Wpointer-arith]

   if(dT != NULL) {

            ^

C:\Users\Benedikt\Documents\Arduino\libraries\MS561101BA\MS561101BA.cpp:94:12: warning: converting to non-pointer type 'float' from NULL [-Wconversion-null]

     return NULL;

            ^

C:\Users\Benedikt\Documents\Arduino\libraries\MS561101BA\MS561101BA.cpp: In member function 'int32_t MS561101BA::getDeltaTemp(uint8_t)':

C:\Users\Benedikt\Documents\Arduino\libraries\MS561101BA\MS561101BA.cpp:100:17: warning: NULL used in arithmetic [-Wpointer-arith]

   if(rawTemp != NULL) {

                 ^

C:\Users\Benedikt\Documents\Arduino\libraries\MS561101BA\MS561101BA.cpp:104:12: warning: converting to non-pointer type 'int32_t {aka long int}' from NULL [-Wconversion-null]

     return NULL;

            ^

C:\Users\Benedikt\Documents\Arduino\libraries\MPU60X0\MPU60X0.cpp: In member function 'void MPU60X0::initialize9250MasterMode()':

C:\Users\Benedikt\Documents\Arduino\libraries\MPU60X0\MPU60X0.cpp:140:20: error: call of overloaded 'begin(long int)' is ambiguous

   Wire.begin(400000);

                    ^

C:\Users\Benedikt\Documents\Arduino\libraries\MPU60X0\MPU60X0.cpp:140:20: note: candidates are:

In file included from C:\Users\Benedikt\Documents\Arduino\libraries\I2Cdev/I2Cdev.h:70:0,

                 from C:\Users\Benedikt\Documents\Arduino\libraries\MPU60X0\MPU60X0.h:40,

                 from C:\Users\Benedikt\Documents\Arduino\libraries\MPU60X0\MPU60X0.cpp:37:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src/Wire.h:53:10: note: void TwoWire::begin(uint8_t)

     void begin(uint8_t);

          ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src/Wire.h:54:10: note: void TwoWire::begin(int)

     void begin(int);

          ^

exit status 1

```
Hello i tried to compile the example sketches but none of them worked. I copied the libraries in the correct folder and uncommented only the #define GEN_MPU9250  in the FreeIMU.h.

mjs513, 2017-04-21:
The warnings are not an issue.  The issue is that for some reason it is call twowire.  I just compiled for the Teensy 3.5 and there were no issues using the serial2 sketch, it compiled fine.  Can you tell what board you are using?

Gabande, 2017-04-21:
mega 2560. I also tried out an older IDE (1.5.8) and reinstalled my current one but it didn't worked too. In my library folder are just the libraries wich are included in your libraries folder.

mjs513, 2017-04-21:
Ok.  Just compiled for the Mega 2560.  Found the problem.  Until I update the library, edit the MPU60X0.cpp file in the MPU60X0 library so that line 140 is changed to:
Wire.begin();

and add right after that:
Wire.setClock(400000L);

Let me know if that works

mjs513, 2017-04-21:
Just updated the library with that correction.

Gabande, 2017-04-21:
works now thanks

closed 2017-04-21 by Gabande