mjs513/FreeIMU-Updates #87: Stop Sampling ---> Hangs

reycreator17, 2017-06-02:
Hi @mjs513 

Thanks for your wonderful work. 

I've followed your instructions. 

commented out mpu9250 and etc. Tried it in serial monitor sending "v" and "r" and its working. 
Also, I'm using the Free_IMUserial2.ino since the Free_IMUserial.ino do have some error saying:

"cmd1" is not defined. I think it should be cmd, anyway, I just used  Free_IMUserial2.ino.

My problem is I got no luck when using the gui. 

In linux OS, I'm having "segmentation fault" using Python 2.7.13. 

while in windows 7, I'm having the "list"  no "size" error. I've checked the cal_lib.py and its 

 H = numpy.array([x, y, z, -y**2, -z**2, numpy.ones([len(x)])])

After that I typed COM6, hit connect, hit Start sample, and it would print : serial connected, but the GUI would not draw anything. 

And when I click Stop Sample, it hangs. I don't know if the problem is in the arduino sketch or in the Python. 

[image]

Hope you can help

Thanks

Rey

reycreator17, 2017-06-02:
Another question, should I really be using Arduino 1.5.7 ? or it should be 1.6.7? 

BTW, I'm using Python 2.7.13

mjs513, 2017-08-15:
Should run on the latest version of the IDE.  Also, I just added a stand alone version of the calibration routine that gets rid of all the erros.

closed 2017-10-31 by mjs513