mjs513/FreeIMU-Updates #88: Developing on uno R3

april4s008, 2017-07-10:
Hi mjs513,

Saw your video and was amazed how stable your implementation could be. I'm using Arduino uno R3 to process a MPU 9250 chip. Trying to follow your solution but there are just too many files , could you please instruct on which are the files for Arduino development for 9250? I'm not planning on creating GUI, just want to draw the yaw, pitch and roll.

Many thanks!
Jon

mjs513, 2017-07-11:
Hi Jon

Unfortunately I don't think the sketch will fit on an Arduino Uno R3.  You will need at least a mega.  You can try Kris Winer's 9250 sketch.  It should fit.

Mike

april4s008, 2017-07-11:
Hi Mike,

May I ask which sketch is for mega then? I can try modify it to uno. I've
been using Kris Winer's sketch and implemented already, the problem is yaw
pitch and roll kept fluctuating within ~5 degrees. I saw the video by
cybermerlin https://www.youtube.com/watch?v=lwlOZzNJqJc   which shows very
good stability. That directs me to this site. So I just want to learn how
to stablize the data.

Many thanks!

best regards,
Jon

-- 
Jiannan Li
Master of Science
Electrical and Computer Engineering,
University of California, Davis
CA, USA

Bachelor of Engineering
Department of Nanoelectronic and Microelectronic
Tsinghua University, Beijing 100084, P.R.China
Phone: +1-5306013732

mjs513, 2017-08-15:
Use the serial2 version of the sketch but you will probably have to make a few other modifications

closed 2017-10-31 by mjs513