mjs513/FreeIMU-Updates #91: unit of accelerometer and magnetometer and gyroscope

fe-mo, 2017-10-31:
HI, Thanks for your nice job, I've used your library for calibration but my question is that what is the unit of accelerometer , magnetometer and gyroscope  because I was a bit confused a why after calibration the range of accelerometer changed too much ? Is after calibration you change the the unit from mg to g ? I attched the pdf of accelerometer< and his would happen for magnetomer and gyrsocope would you please explain it
thanks
[accelerometercalibrated(1).pdf](https://github.com/mjs513/FreeIMU-Updates/files/1430655/accelerometercalibrated.1.pdf)
[accelerometerrawdata.pdf](https://github.com/mjs513/FreeIMU-Updates/files/1430656/accelerometerrawdata.pdf)

mjs513, 2017-10-31:
hi fe-mo.  There is absolutely nothing wrong.  In you plot of calibrated data the accelerometer axes are normalized based on the calibrated values so that z-axis is approximately 1 (pointing down).  Y and x are close to zero.  Your raw data plot is just that, its the data that is coming out of the imu in counts.  

Mike

fe-mo, 2017-10-31:
Sorry the unit of the accelerometer is g or mG and also the magnetometer it is micro tesla? and why the accelerometer is not -9.8 and it is 1 ?

mjs513, 2017-10-31:
Normalization.   When at rest it will be 1.  To get g's multiply the calibrated values by 1g (9.81 meter/sec^2).   Raw data is in counts and if you are using a MPU9250 for instance for a setting of 2g's the resolution for each tick is 2.0/32768.0 (the 32768 is the total range, you need to look at the register map or spec sheet).  to get g's from the raw data multiply the accelerometer value by resolution (2/32768).

fe-mo, 2017-12-07:
Hi thanks for your explanation about calculating g for accelerometer . But what about magnetometer how it normalize the magnetomer ? what is the unit of the magnetometer after and before calibration sorry i ask it because i get confused
[image]
[image]

I mean how the magnetometer data is normalised thanks for your answer

ambidexi, 2018-07-20:
Hi! I'm not a dev, just a user.

The units of raw accelerometer and magnetometer data depend on the scaling set on the sensors.

When normalizing the data, accelerometer is normalized to percieved gravity and magnetometer to the percieved magnetic field in a way that accel=1 represents 1g = 8.91m/s2 and magneto=1 is the local strength of the magnetic field, which varies greatly. (According to the Wikipedia, the intensity of the Earth's magnetic field ranges between approximately 25,000 and 65,000 nT.)

Normalization has no way of knowing whether you are measuring on the Mars or the sensor has linearity errors. The raw data has a specific unit but is exposed to bias and linearity errors. The calibrated values are in the aforementioned relative units.