mjs513/FreeIMU-Updates #92: Not able to compile in Arduino IDE, issues with cstring

ashaposhnikov, 2018-01-02:
Hi guys,

I'm trying to compile freeimu for atmega328p using modified FreeIMU v0.4 definition. Nonetheless even with original I'm not able to compile cstring related part of stlport and getting the following error:

Arduino/libraries/stlport/cstring:32:28: fatal error: cstring: No such file or directory
 #    include_next <cstring>

What I'm missing ? 

Thanks,
Alex

mjs513, 2018-01-02:
Hi Alex

What sketch are you trying to run?  

Mike

ashaposhnikov, 2018-01-02:
The simple one: FreeIMU Yaw_Pitch_Roll

ashaposhnikov, 2018-01-02:
I've modified it with full path to the StandardCplusplus library and it worked fine, so it looks like Arduino IDE not including all the libraries or not able to find it by some reason. Now I'm stuck on templates: 
Arduino/libraries/AP_Math_freeimu/AP_Math_freeimu.h:34:15: error: 'enable_if' in namespace 'std' does not name a template type
 typename std::enable_if<std::is_integral<typename std::common_type<Arithmetic1, Arithmetic2>::type>::value ,bool>::type 
               ^
Arduino/libraries/AP_Math_freeimu/AP_Math_freeimu.h:34:24: error: expected unqualified-id before '<' token
 typename std::enable_if<std::is_integral<typename std::common_type<Arithmetic1, Arithmetic2>::type>::value ,bool>::type

mjs513, 2018-01-02:
Ok.  First, I just compiled it on my machine and there were no errors.  What version IDE are you using and what board?

ashaposhnikov, 2018-01-02:
I'm using 1.8.5 IDE with custom schematics. I'm using 328P with MPU6050 + HMC5883. It's basically have the same HW as FreeIMU v0.4 but without BARO sensor so I've disabled it(baro) everywhere.
Could it be because I'm not copied all the directories in Library folder to the Arduino library ? I could do so, but trying to get less mess with all of them there :)
It basically unclear

ashaposhnikov, 2018-01-02:
Initially it calls stlport library which makes a call to the cstring using next_import cstring which I assumed should be from standardcpluplus and I pointed to it. So, your suggestion is just to remove this next import ?

mjs513, 2018-01-02:
Ok.  First, found that i do use it in the Math library.  No you don't need all the libraries.  for your configuration you will need:
```
 AP_Filter  
AP_Math_freeimu 
DCM 
Eigen325 
EigenArduino-Eigen30 
ExtendedKalman 
FreeIMU 
FreeIMU60XX 
FreeIMUUtils 
HMC5883L 
HMC58X3 
I2Cdev 
Kalman 
MPU60X0 
iCompass 
stlport 
```
Just leave these in your libraries folder.  Also if you are just using the 6050 and 5883 without a barometer use the  delete the freeimuv04 board from the defines for  has_press and has_ms5611.  You will be better off in the log run.

Give it a try once you clean up your library.  Something might be missing.

mjs513, 2018-01-02:
> So, your suggestion is just to remove this next import ?
No, don;t do that yet.  Was going to say comment it out of the freeimu_math.h file but that may cause a couple of other minor problems.  Like i said i can not duplicate the problem.

ashaposhnikov, 2018-01-02:
I've just moved out my original libraries folder from Arduino and replaced it with all FreeIMU libraries and still getting issues with cstring:

`"/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   "-I/Library/Arduino15/packages/arduino/hardware/avr/1.6.20/cores/arduino" "-I/Library/Arduino15/packages/arduino/hardware/avr/1.6.20/variants/eightanaloginputs" "-I/Library/Arduino15/packages/arduino/hardware/avr/1.6.20/libraries/Wire/src" "-I/Library/Arduino15/packages/arduino/hardware/avr/1.6.20/libraries/SPI/src" "-I/Documents/Arduino/libraries/I2Cdev" "-I/Documents/Arduino/libraries/AP_Math_freeimu" "-I/Documents/Arduino/libraries/stlport" "/var/folders/gl/gn_rgpsn1xl96cn_w2ncgnjw0000gp/T/arduino_build_168957/sketch/head-tracker-mpu-hcl.ino.cpp" -o "/var/folders/gl/gn_rgpsn1xl96cn_w2ncgnjw0000gp/T/arduino_build_168957/preproc/ctags_target_for_gcc_minus_e.cpp"
In file included from /Documents/Arduino/libraries/stlport/type_traits:33:0,
                 from /Documents/Arduino/libraries/AP_Math_freeimu/AP_Math_freeimu.h:5,
                 from /Documents/Arduino/head-tracker-mpu-hcl/head-tracker-mpu-hcl.ino:28:
/Documents/Arduino/libraries/stlport/cstring:32:28: fatal error: cstring: No such file or directory
 #    include_next <cstring>
                            ^
compilation terminated.
Using library Wire at version 1.0 in folder: /Library/Arduino15/packages/arduino/hardware/avr/1.6.20/libraries/Wire 
Using library SPI at version 1.0 in folder: /Library/Arduino15/packages/arduino/hardware/avr/1.6.20/libraries/SPI 
Using library I2Cdev in folder: /Documents/Arduino/libraries/I2Cdev (legacy)
Using library AP_Math_freeimu in folder: /Documents/Arduino/libraries/AP_Math_freeimu (legacy)
Using library stlport in folder: /Documents/Arduino/libraries/stlport (legacy)`

mjs513, 2018-01-02:
Are you using 
[stlport-avr-STLport-master-arduino-installed.zip](https://github.com/mjs513/FreeIMU-Updates/files/1598708/stlport-avr-STLport-master-arduino-installed.zip)
an Arduino uno?  Substitute this version of stlport with your version:

ashaposhnikov, 2018-01-02:
Wow. That one worked just perfectly for me!!! Thanks a lot. I've used one which you have in repo in stlport folder.

Thanks a lot for a quick help and solution, I really appreciate that!

mjs513, 2018-01-02:
Not a problem.  My fault.  Never pushed the button to upload the new stlport library in GitHub.  Thanks for pointing it out.

Good luck.  If you have any other issues please feel free.

Mike

closed 2018-01-02 by mjs513