mjs513/FreeIMU-Updates #93: Not able to use MPU6050 with HMC58X3 instead MPU9250(with AK8963)

ashaposhnikov, 2018-01-02:
So, I've successfully build example with FreeIMU_v4 but not able to decouple few things and becaus of that not fitting in 32k of 328p. Things which would be included during compilation but not needed: SPI, AK8963, MPU9250-related functions. Any recommendations about how to fix that or may be I chose incorrect way of doing it ?

What I've tried so far:
1. Use modified FreeIMU_v04 define without baro/pressure sensor.
2. Import separately MPU60X0 and HMC58X3. 

In both cases I'm getting errors about SPI import (which was easy to remove btw) but much worse is procedures of initialization of AK which is inside of MPU9250. 

Thanks!

mjs513, 2018-01-02:
Make sure you commented out the define for the MPU9250 and only have the define for freeimuv4 selected.

mjs513, 2018-01-02:
PS.  When I compiled the ypr example it just fit on a Arduino Uno.

ashaposhnikov, 2018-01-03:
Sketch uses 33296 bytes (108%) of program storage space. Maximum is 30720 bytes.
Global variables use 1549 bytes (75%) of dynamic memory, leaving 499 bytes for local variables. Maximum is 2048 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board Arduino Nano.

Anyway, I also found that it doesn't work fine with newer I2Cdev library I had(11/1/2011 vs 2013-06-05) and that caused mostly all the issues I had with compiling of just simple example.

mjs513, 2018-01-03:
Didn't know you were using the newer version of i2cdev.  Using the old version as it allowed flexibility to use SPI as well and everything was already geared to that version.  One thing I noticed is depending on the board the size of the sketch can vary.  If you want a better processor check out the Teensy 3.2.  You won't have a problem with memory and its a lot faster.

That aside I think I have a bare boards version laying around somewhere - I will have to see if I can find it.

ashaposhnikov, 2018-01-03:
Once I removed/using of pressure sensor/barometer it fit in 25k which should be fine so we could close that.

Thanks a lot!

mjs513, 2018-01-03:
Ok found it- no guarantees on the final size or if it going to work.

[FreeIMU Barebones.zip](https://github.com/mjs513/FreeIMU-Updates/files/1599090/FreeIMU.Barebones.zip)

yep the pressure sensor uses a lot of memory.

mjs513, 2018-01-03:
Great. I will go ahead and close this issue.

closed 2018-01-03 by mjs513